raomin/ESPAltherma #3: Smart Grid functions

misko903, 2020-12-30:
are any Smart Grid functions planned? I have bought BRP069A61 and its crap. Using excessive energy from Photovoltaics works only when heating/cooling is off (not in standby), just because of missing room thermostat. It would be easy to switch heating just for 1/2 hour and then switch it back...

raomin, 2020-12-30:
Hi! not sure what you expect by "Smart Grid functions" but I would imagine that the heating/cooling should be turned On when producing excessive energy. 
If you have this information on Home Assistant then it would be easy to trigger the heat pump through the _external on/off thermostat_ acted by ESPAltherma. This would not trigger DHW if you have it.

misko903, 2020-12-30:
Hi, not really, because with Smart Grid Functions you are able to regulate the consumption of the heat pump. See this:
https://www.daikin.eu/content/dam/document-library/Installer-reference-guide/ctrl/BRP069A61,%20BRP069A62_4PEN464229-1_2017_08_Installer%20reference%20guide_English.pdf
Search for Smart Grid

raomin, 2020-12-31:
OK I now understand that this the LAN adapter is able to limit the consumption of the heat pump to only the excessive energy. ESPAltherma cannot do this for now as it's only capable to act as a OnOff thermostat.
So you could turn it On when you have excessive production but it might consume more than what you produce... 

As BRP069A61 communicates on P1P2, the power limitation can probably be achieved with [Arnold's P1P2Serial](https://github.com/Arnold-n/P1P2Serial)

Also if you have the _Demand PCB_ option then it seems that you can easily set a specific power consumption limit. A 4 relay module and some additional code would then allow ESPAltherma to limit the power. 
Now there should also be a way to interface directly with the port X85A (to which the _Demand PCB_ is interfaced with) but I couldn't find any information on this port. Maybe someone knows about this port...?

misko903, 2021-01-02:
The LAN adapter itself is not capable of using excessive energy. For that I
need an extra electricity meter, which measures Outgoing energy, sending
the signal to LAN adapter.
Behaviour of the adapter is like:
- listen on X1A/N+L contacts, if there is 230V. If yes, then:
- listen on X1A/1+2 (SG0) and X2A/3+4 (SG1) if there is dry contact. Then:
- if* SG0 is ON*, unit is in recommended ON mode. But *only if
heating/cooling is powered OFF* (using Electricity meter signal, or static
power limitation set on adapter's webpage)
- if *SG0 and SG1 is ON*, the unit is in forced ON mode. No matter the
heating, no matter the electricity meter, it is consuming as much as
possible to stabilise the network (I am simulating this with simple manual
switch, as I don't have real SG1 signal from my electricity provider)

It would be enough to make it this way:
1. when floor heating back water temp is on SET value, and SG0 is ON, it
will switch OFF the heating (to take LAN Adaptor SG0 into account)
2. then the rest will do the LAN adaptor - heat water in boiler to its max
value - 60deg (already programmed in LAN Adaptor, which is using S0 signal
from electricity meter, or static SET value through web interface)
3. Once in 15 or 30 min is checked if the floor heating temp real value is
corresponding to SET value
4. If real=SET, continue with with step 1

Or should I simply buy a stupid Thermostat which will switch off the
heating, or I will wait until they improve the software on the LAN adapter?
I am disappointed with the LAN adapter which I consider as an unfinished
product.

miso

andreapier, 2021-02-04:
Hi, I am currently exploring how to integrate my PV interver with Altherma. I think it would be nice to manage it with ESPAltherma, so I only have 1 device that controls all aspects of HP.  
My idea is to use a Shelly EM (or any other similar EM) and plug 2 clamps. 1st clamp measures PV production. 2nd clamp measures total building consumption.  
I send those values to appropriate ESPAltherma's channels and use them to command 2 pins. Those 2 pins will in turn command 2 relays that will open/close SG0/SG1 dry contacts.  
### Parameters
I came up with this data and algorithm to set SG mode, please give advise if you think it could be improved:  
| Parameter         | Description                                | Direction                             | Channel                                           |
|-------------------|--------------------------------------------|---------------------------------------|---------------------------------------------------|
| Delta_SG_3        | Power consumption of unit when in SG3 mode | Fixed constant, known at compile time | Source code                                       |
| Produced_Power    | Power produced by PV                       | Input                                 | /ESPAltherma/em/Production                        |
| Consumed_Power    | Power consumed by building                 | Input                                 | /ESPAltherma/em/Consumption                       |
| SG_Mode           | Current SG working mode                    | Output                                | /ESPAltherma/em/SG_Mode                           |
| Consumed_Power_HP | Power consumed by HP                       | Internal                              | Calculated ad V*A from already available channels |
| T_DHW             | Current DHW temperature                    | Internal                              | From already available channels                   |
| S_DHW_SG0         | DHW setpoint in SG0 operation              | Internal                              | From already available channels                   |
| S_DHW_SG3         | DHW setpoint in SG3 operation              | Internal                              | From already available channels                   |
  
### Algorithm
```
IF Produced_Power - Consumed_Power >= Delta_SG_3 - Consumed_Power_HP THEN //This IF tries to prevent to consume more energy than what is being produced.
  IF T_DHW <= S_DHW_SG3 THEN  // Only switch to SG3 mode if set temperature is not reached yet
    SG_Mode = 3
    PIN_SG1 = HIGH
    PIN_SG2 = HIGH
  ELSE
    SG_Mode = 2      // Otherwise, switch to SG2 mode, to try to maximize floor heating
    PIN_SG1 = HIGH
    PIN_SG2 = LOW
ELSE IF Produced_Power - Consumed_Power >= 0 THEN // Only switch to SG2 mode if there is enough energy (this could be improved using a Delta_SG_2
  SG_Mode = 2
  PIN_SG1 = HIGH
  PIN_SG2 = LOW
ELSE  // If not power is left, set normal operation mode SG0
  SG_Mode = 0
```

### Possible improvements
- Set a threshold for SG 2 mode, eg. `Delta_SG_2` (same to SG 3 mode). I didn't currently considered this only for the sake of simplicity, and also because I don't know how to find a reasonable value.  
- Add a `SG_Working_Mode_Auto` parameter: if `true`, ESPAltherma works as described. Else, go in "manual mode" where `SG_Mode` is commanded from an external device. This could help users that already have some kind of SG Ready controller and only want ESPAltherma to control relays. It is doable, but I think it's better to wait for feedback to be sure it is something actually useful
- I did not consider SG 1 mode (forced OFF). This is by design as I think there is no valid reason to risk a user's comfort. If a user really wants to switch Altherma OFF, and we agree to implement manual mode, he could explicitly command an `SG_Mode = 0` through some other control system

### Questions
1st question is of course: @raomin  do you think this is a valuable feature for ESPAltherma? I would be pleased to implement it and upstream to your repo, but only if you like the idea. I will gladly accept a "no" and still respect and admire your work.
2nd question is for feedback. Please tell me what you think and if you see any problem with my proposal

raomin, 2021-02-04:
Hi @andreapier, thanks for your feedback and constructive proposal!

In my designs for home automation I always tend to consider the ESP as actuators or sensors, the thinking part (balancing, sensor based decision, etc...) should come from the home automation hub (HA, jeedom, node-red etc...).
That means ESPAltherma would manage 2 extra output GPIOs for SG0/SG1 relays and would monitor MQTT for that (as it does currently for the external thermostat). And the decision would be based on the values your home automation hub is receiving from the different sensors (ESPAlthetrma, Shelly EM etc...). That would ensure an easy monitoring, adaptability of the solution to other cases, etc...

Obviously, adding 2 MQTT switches on ESPAltherma would not be an issue. I was considering adding 4 and hooking to the X85A for those (like me) who do not have BRP069A61.

Tell me what you think.

misko903, 2021-02-04:
BRP069A61 is crap. I have it at home and it is not working correctly since
weeks. They promised to work it with electricity meter, but forget it. Even
SG0 processing is not working reliable. Using it with SolarEdge Zigbee Dry
Contact switch.

miso

KalaNagHTD, 2021-02-05:
https://www.daikin.eu/en_us/products/BRP069A78.html

andreapier, 2021-02-07:
> Tell me what you think.
  
Well, your considerations make perfect sense.  
Having an integrated solution could be easier for users, but at the end it would probably be better for them to separate logic and IO as you suggest.

raomin, 2021-02-26:
Thanks @andreapier for this. I believe it's a great addition to ESPAltherma for every owners of compatible HP.

I still need to figure out how to port this to Altherma3 which has nowhere SG1/SG2 ports, only through the lan adapter, or through the _ondemand pcb_ which looks like just a bunch of optocoupler ending on the X85A port. I'll start playing with this port to see how the HP behaves.

raomin, 2021-03-20:
Now that @andreapier's code is merged, I'll close this issue.
I'll reopen one if I manage to hack the X85A to directly connect with 2 extra relays.

closed 2021-03-20 by raomin