raomin/ESPAltherma #4: PlatformIO problem

pengemetal, 2021-01-02:
Hi, 

I am very happy to fond this project! I am new in PlatformIO. I can't flash ESP32-devkit v4, I get a lot of problems! What do i do wrong? 

[image]

raomin, 2021-01-03:
Hi! Glad you like the project. 
It's a bit difficult to see from the screenshot.
Could you run a build (the ☑️) and paste the compilation logs?

raomin, 2021-01-03:
Make sure you selected ONE file definition in `setup.h`.
Also, download the latest revision, I just fixed a bug that would prevent you from compiling.

pengemetal, 2021-01-03:
Thank you for your effort! I get the same errors:

> Executing task in folder ESPAltherma-main: C:\Users\Admin\.platformio\penv\Scripts\pio.exe run <

Processing default (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (2.1.0) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10004.201016 (1.0.4)
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <PubSubClient> 2.8.0
|-- <ArduinoOTA> 1.0
|   |-- <Update> 1.0
|   |-- <WiFi> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|-- <WiFi> 1.0
Building in release mode
Compiling .pio\build\default\src\main.cpp.o
Generating partitions .pio\build\default\partitions.bin
Compiling .pio\build\default\lib8eb\PubSubClient\PubSubClient.cpp.o
Compiling .pio\build\default\libc79\Update\Updater.cpp.o
Compiling .pio\build\default\lib13e\WiFi\ETH.cpp.o
In file included from src\main.cpp:7:0:
src\setup.h:2:1: error: 'WIFI_SSID' does not name a type
 WIFI_SSID "Emelet"//**Your SSID here**
 ^
In file included from src\setup.h:29:0,
                 from src\main.cpp:7:
include/def/ALTHERMA(LT_CA_CB_11-16KW).h:5:1: error: 'LabelDef' does not name a type
 LabelDef PROGMEM labelDefs[] = {
 ^
In file included from src\main.cpp:9:0:
include/converters.h:8:37: error: 'LabelDef' has not been declared
     void getLabels(char registryID, LabelDef* ret[], int &num)
                                     ^
include/converters.h:59:18: error: 'LabelDef' has not been declared
     void convert(LabelDef* def, char *data)
                  ^
include/converters.h: In member function 'void Converter::getLabels(char, int**, int&)':
include/converters.h:11:29: error: 'labelDefs' was not declared in this scope
         for (auto &&label : labelDefs)
                             ^
include/converters.h: In member function 'void Converter::readRegistryValues(char, char*)':
include/converters.h:39:9: error: 'LabelDef' was not declared in this scope
         LabelDef* labels[128];
         ^
include/converters.h:39:19: error: 'labels' was not declared in this scope
         LabelDef* labels[128];
                   ^
include/converters.h: In member function 'void Converter::convert(int*, char*)':
include/converters.h:61:14: error: request for member 'asString' in '* def', which is of non-class type 'int'
         def->asString[0] = {0};
              ^
include/converters.h:62:27: error: request for member 'convid' in '* def', which is of non-class type 'int'
         int convId = def->convid;
                           ^
include/converters.h:63:24: error: request for member 'dataSize' in '* def', which is of non-class type 'int'
         int num = def->dataSize;
                        ^
include/converters.h:74:26: error: request for member 'asString' in '* def', which is of non-class type 'int'
[... 85 lines cut ...]
                                    ^
src\main.cpp:67:44: error: 'TX_PIN' was not declared in this scope
   MySerial.begin(9600, SERIAL_8E1, RX_PIN, TX_PIN);
                                            ^
src\main.cpp:72:20: error: 'MQTT_SERVER' was not declared in this scope
   client.setServer(MQTT_SERVER, MQTT_PORT);
                    ^
src\main.cpp:72:33: error: 'MQTT_PORT' was not declared in this scope
   client.setServer(MQTT_SERVER, MQTT_PORT);
                                 ^
src\main.cpp:73:24: error: 'MAX_MSG_SIZE' was not declared in this scope
   client.setBufferSize(MAX_MSG_SIZE); //to support large json message
                        ^
src\main.cpp:87:23: error: 'labelDefs' was not declared in this scope
   for (auto &&label : labelDefs)
                       ^
src\main.cpp: In function 'void loop()':
src\main.cpp:137:46: error: 'FREQUENCY' was not declared in this scope
   Serial.printf("Done. Waiting %d sec...\n", FREQUENCY / 1000);
                                              ^
*** [.pio\build\default\src\main.cpp.o] Error 1
================================================================================================== [FAILED] Took 7.41 seconds ==================================================================================================
The terminal process "C:\Users\Admin\.platformio\penv\Scripts\pio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

raomin, 2021-01-03:
There seems to be a problem with your setup.h.

Line 2 should read
`#define WIFI_SSID "Emelet"//Your SSID here`

Can you paste your setup.h here? (mark **** for your WiFi pwd)

pengemetal, 2021-01-03:
//Setup your credentials and mqtt info here:
WIFI_SSID "Emelet"//**Your SSID here**
WIFI_PWD "********"//**Your password here** leave empty if open (bad!)

MQTT_SERVER "192.168.1.114"//**Your IP address here**
MQTT_USERNAME "szunyog"//leave empty if not set (bad!)
MQTT_PASSWORD "********"//leave empty if not set (bad!)
MQTT_PORT 1883

FREQUENCY 30000 //query values every 30 sec
RX_PIN    3// Pin connected to the TX pin of X10A 
TX_PIN    1// Pin connected to the RX pin of X10A
PIN_THERM 0// Pin connected to the thermosta relay (normally open)

MAX_MSG_SIZE 4096//max size of the json message sent in mqtt 

//Uncomment *ONE* of the following according to your installation.
//Then, open and edit the selected file in the include folder and uncomment each values you are interested in.

//#include "def/ALTHERMA(BIZONE_CB_04-08KW).h"
//#include "def/ALTHERMA(BIZONE_CB_11-16KW).h"
//#include "def/ALTHERMA(GSHP).h"
//#include "def/ALTHERMA(GSHP2).h"
//#include "def/ALTHERMA(HPSU6_ULTRA).h"
//#include "def/ALTHERMA(HYBRID).h"
//#include "def/ALTHERMA(LT-D7_E_BML).h"
//#include "def/ALTHERMA(LT_11-16KW_HYDROSPLIT_HYDRO_UNIT).h"
//#include "def/ALTHERMA(LT_CA_CB_04-08KW).h"
#include "def/ALTHERMA(LT_CA_CB_11-16KW).h"
//#include "def/ALTHERMA(LT_DA_04-08KW).h"
//#include "def/ALTHERMA(LT_DA_PAIR_BML).h"
//#include "def/ALTHERMA(LT_GAS_INJ)20200702.h"
//#include "def/ALTHERMA(LT_MULTI_DHWHP).h"
//#include "def/ALTHERMA(LT_MULTI_HYBRID).h"
//#include "def/ALTHERMA(MONOBLOC_CA_05-07KW).h"
//#include "def/ALTHERMA(TOP-GRADE).h"
//#include "def/DAIKIN_MINI_INVERTER_CHILLER04-08KW.h"
//#include "def/DEFAULT.h"

raomin, 2021-01-03:
Ah, you should NOT remove the `#define` from any line. This is how is define a constant value in C.

So, get back the original file https://github.com/raomin/ESPAltherma/blob/main/src/setup.h and replace what is between the `"  "`.

Your first lines should look like this: 
```c
//Setup your credentials and mqtt info here:
#define WIFI_SSID "Emelet"//Your SSID here
...
```

pengemetal, 2021-01-03:
OMG! Thank you, I dont know why delete, the whole rows, including #define! (C is new for me) I managed to compile! Tomorrow I will start wiring :)

raomin, 2021-01-03:
Great, that's good news. Happy wiring!

closed 2021-01-03 by raomin