raomin/ESPAltherma #5: Sensor without attributes

pengemetal, 2021-01-04:
Hi, 
I am here again! I wired to the heat pump. I can see sensor.althermasensors in HA, but i can't see any attributes. I changed the TX, RX pins in setup.h according to: https://live.staticflickr.com/4764/40089095211_ec1fee0087_b.jpg (TX 1, RX 3)

#define FREQUENCY 30000 //query values every 30 sec
#define RX_PIN    3// Pin connected to the TX pin of X10A 
#define TX_PIN    1// Pin connected to the RX pin of X10A

Do i need change main.cpp?
Or maybe I can set something in my heatpump to send data on X10A port?

Thanks in advance!

raomin, 2021-01-04:
Hi @pengemetal,
The ESP32 has 3 serial ports. The first one, Serial0 is reserved for ESP<-USB->PC communication and ESP Altherma uses the Serial0 for logging (as any other project would do). So if you open the serial monitor on your PC, you'll see some debug from ESPAltherma. 

This also mean that ESPAltherma cannot use Serial0 to communicate with Altherma. You need to use another Serial port to communicate with the Altherma.
ESP32 can map any GPIO to the serial ports. If you want to stick with the default of your board that would be:
[image]

So:
```c
#define RX_PIN 9
#define TX_PIN 10
```

pengemetal, 2021-01-04:
I wrote the new config, but the result is the same :( No attributes in HA only the friendly name.
Do I need restart the heat pump? (turn off the electricity, not only the on/off button)

raomin, 2021-01-04:
Check your wiring: 
- Your port GPIO9 (D2) should be connected to X10A port 2.
- Your port GPIO10 (D3) should be connected to X10A port 3.

If it still doesn't work, plug in your USB cable to your ESP32 (while connected to the X10A) and open the port monitor: 
[image]

Debug logs should appear. Paste them here.

pengemetal, 2021-01-04:
Wiring is good,
Here is the log:
> Executing task in folder ESPAltherma-main: C:\Users\Admin\.platformio\penv\Scripts\pio.exe device monitor <

--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at http://bit.ly/pio-monitor-filters
--- Miniterm on COM4  9600,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
Connecting to Emelet
.....................................................................................................                                                                                                                           1���␗␘�V�@���    M�␌�Connecting to Emelet
..Connected. IP Address: 192.168.1.166
Connecting to Emelet
Connected. IP Address: 192.168.1.166
Attempting MQTT connection...[E][WiFiClient.cpp:232] connect(): connect on fd 55, errno: 118, "Host is unreachable"
Attempting MQTT connection...connected!
Querying register 0x10... Time out! Check connection
Querying register 0x10... Querying register 0x10... Time out! Check connection
Querying register 0x10... Querying register 0x20... Time out! Check connection
Querying register 0x20... Time out! Check connection
Querying register 0x20... Time out! Check connection
Querying register 0x20... Querying register 0x21... Time out! Check connection
Querying register 0x21... Time out! Check connection
Querying register 0x21... Time out! Check connection
Querying register 0x21... Time out! Check connection
Querying register 0x30... Time out! Check connection
Querying register 0x30... Time out! Check connection
Querying register 0x30... Time out! Check connection
Querying register 0x30... Time out! Check connection
Querying register 0x60... Time out! Check connection
Querying register 0x60... Time out! Check connection
Querying register 0x60... Querying register 0x60... Time out! Check connection
Querying register 0x61... Querying register 0x61... Time out! Check connection
Querying register 0x61... Querying register 0x61... Time out! Check connection
Querying register 0x62... Time out! Check connection
Querying register 0x62... Time out! Check connection
Querying register 0x62... Querying register 0x62... Time out! Check connection
Querying register 0x64... Time out! Check connection
Querying register 0x64... Time out! Check connection
Querying register 0x64... Querying register 0x64... Time out! Check connection
Sending values in MQTT.
Done. Waiting 30 sec...

raomin, 2021-01-04:
Eh... that's looks like a connection issue. What is the model of your heat pump?

pengemetal, 2021-01-04:
ERLQ011CV3 outdoor unit
EHBX11CB3V indoor unit

raomin, 2021-01-04:
Hmm. this should work. Try:
- checking again all wirings and connections. 
- turning it off and on at circuit breaker.
- changing the RX TX pins to 25 and 26 in setup.h and wire accordingly.

raomin, 2021-01-05:
Ah you might want to increase timeout in setup.h maybe...

pengemetal, 2021-01-05:
I changed the board to a Devkit 1, I wired RX2 and TX2 port, and it WORKS! :) My dev4 board was the problem! Thak you for your help!

raomin, 2021-01-05:
That's great news! I'm glad it worked.
That would still be interesting to know why it did not for the devkit v4. Maybe you could try on ports 16 (RX2) and 17 (TX2) of your v4?

pengemetal, 2021-01-05:
Good news! Devkit v4 is also working on 16 and 17 ports!

closed 2021-01-05 by raomin

pengemetal, 2021-01-05:
I thank you for helping a lot! Would you like to send photos of wiring or my setup.h config?

raomin, 2021-01-05:
;) Photos of your wiring

pengemetal, 2021-01-05:
[image]
[image]
It will get some plastic box :)

raomin, 2021-01-05:
We cannot see your picts. Try copy/paste.

pengemetal, 2021-01-05:
[image]
[image]