raomin/ESPAltherma #6: Home assistant missing celsius value

pengemetal, 2021-01-06:
Hi,

Just one more question :) In HA is created sensors, but C is missing (and Amper too) everywhere. Is it possible heat pump dont give these values? It is important for grahps. I can see your sensors have these.
[image]

raomin, 2021-01-06:
Hi @pengemetal,

As explained in the Readme, you'll have to extract your Attributes as Entities using Template.
When you declare your Template you can specify a unit.
eg:
```yaml
  - platform: template
    sensors:
      espaltherma_dhw:
        friendly_name: "DHW Temp"
        value_template: "{{ state_attr('sensor.altherma','DHW tank temp. (R5T)') }}"
        unit_of_measurement: '°C'
```
Once you restart HA, your templated attributes will be available for your charts.
Happy dashboarding :)

--
If you like this project, please consider supporting it, [buy me a beer!](https://www.buymeacoffee.com/raomin)

closed 2021-01-06 by raomin

pengemetal, 2021-01-06:
Oh, so sorry, I forgot to add unit of measurment, only the value template. :)