raomin/ESPAltherma #7: Frequent disconnects

giulup, 2021-01-10:
Hi Raomin,
thanks for that great possibility to monitor our Heatpumps. 
I have that issue that the M5Stick disconnects and shuts off after a while. I thought it was due to the insufficient voltage and low battery level. I charged the battery but still get the disconnects. Is this a known issue or something wrong at my side?

raomin, 2021-01-10:
Hi @giulup! 

My first option would be to check the wiring and make sure it is getting 5v in VIN. From what you described it seem the M5stickc is running on its own battery, which should normally not be needed. 
But, I would need more information to help:

- Before it 'disconnects' is ESPAltherma working properly? 
- when you say it shuts off, does it means it is not responding and depleted its own battery? 
- what logs do you see on your mqtt broker on espaltherma/log?
- is it "shutting down" quickly? Always the same duration?
- what is the model of you heat pump? If is normally has a blinking green Led on the PCB is it still blinking when the M5stickc shut off? 
- could you try to add a USB charger to power the M5stickc and see if it stays on? 

--
If you like this project, please consider supporting it, [buy me a beer!](https://www.buymeacoffee.com/raomin)

raomin, 2021-01-10:
Also, are you sure you connected to 5V in ( **5V ⬅️**) and not 5V Out (**5V ➡️**)? 

--
If you like this project, please consider supporting it, [buy me a beer!](https://www.buymeacoffee.com/raomin)

giulup, 2021-01-10:
The longest run was approximately 2 hrs before it stopped. During that time there were no errors. 
my first guess was also the wiring. This is all right. 
I measured the output on the X10a VCC and got 4.8V. Might it be too low to power the Stick?

I have a Rotex HPSU with the 8-pin connector at X10a. Does it have more than one 5V output? I will make a try with a usb charger as I have a plug nearby. If this works I will have to look for a workaround to power it.

raomin, 2021-01-10:
I don't have a Rotex, So it might be possible that the 5v is too weak to supply the M5... 
I've just added (for you :) ) a feature to report data on voltage and consumption of the M5 VIN and its battery.
You should see easily if power is drawn from the battery at all. For information, my M5 draws 70mA from VIN and 0 from the battery:
[image]

Download the newest code, set your environment to M5StickC [image] and you should see new values poping in the attributes returned.

--
If you like this project, please consider supporting it, [buy me a beer!](https://www.buymeacoffee.com/raomin)

giulup, 2021-01-11:
Ok... seems clear that the Voltage supply of my X10a Connector is insufficient or not connected properly. I plugged the Stick via USB and it was working like a charm for the last 10 hrs. Now i wanted to upload your newest Version, but something seems to be wrong with it. I have problems compiling it due to errors with bad environments. Could you give it a look?

raomin, 2021-01-11:
Hi @giulup,
Thanks for this feedback. I realized I introduced an additional environment on platformio.ini.
That is fixed now. You should be able to get the new code, compile and upload OTA.

--
If you like this project, please consider supporting it, [buy me a beer!](https://www.buymeacoffee.com/raomin)

giulup, 2021-01-11:
@raomin Thanks for that fast fix. It is working now and as i assumed the voltage supply on my connector is not stable enough to power the stick.

This is with installed usb-plug:
[image]

That is without the usb-plug:
[image]

Thank you for your commitment.

giulup, 2021-01-11:
Thanks a lot. After all you had new ideas to implement in your project ;)

closed 2021-01-11 by giulup

raomin, 2021-01-11:
Too bad for not getting proper 5v.
Did you check other ports or elsewhere on the board? 
On mine there is a 7805 with a massive heat sink.