raomin/ESPAltherma #8: Implementation of Rotex support

giulup, 2021-01-11:
Hi @raomin 
its again me. ;)
In the german forum there are lots of users owning a Rotex HPSU. The Rotex is a branch of Daikin so that the protocoll is related.
The corresponding values for the Rotex-units are spread among the Altherma models.
Would it be difficult to collect the right values into one or more new models for a Rotex if we tell you which value of which Altherma is needed for the Rotex?

raomin, 2021-01-11:
Hi again!
If it's the same protocol, I would imagine that the registers and values are also similar.
So we would need to clone one of the definition file and compile all interesting values for Rotex.
You can do it, paste it here and I'll add it to the repository :)

giulup, 2021-01-13:
Everythin fine. Seems like only less interesting values are missing or wrong. So no need to adjust anything.

crow187, 2021-01-13:
I have a Rotex hpsu compact ultra 508
I used the ALTHERMA(HPSU6_ULTRA).h and can confirm following registers contain reasonable values:

"Operation Mode": 
"Defrost Operation":
"Low noise control":
"I/U operation mode": current operation state
"Circulation pump operation":  
"Alarm output":                
"Target Evap. Temp.":        
"Flow sensor (l/min)":       
"Outdoor air temp.":         
"Discharge pipe temp.":     
"O/U Heat Exch. Mid-temp.":  //sensor in the middle of the outdoor unit
"INV fin temp.":             // What is this?
"Voltage (V)":               
"INV primary current (A)":   //What are primary and secondary current?
"INV secondary current (A)": //
"Outlet Water Heat Exch. Temp. (R1T)": //t-V
"Outlet Water BUH Temp. (R2T)":        //t-V,BH
"Refrig. Temp. liquid side (R3T)":     //t-liq
"Inlet water temp.(R4T)":              //T-R
"DHW tank temp. (R5T)":                //t-DHW2
"RT Temp.":                            //What is this?
"LW setpoint (add)":               
"RT setpoint":                         
"Flow sensor (l/min)":                 //water flow
"Water pump signal (0:max-100:stop)":  //set pump speed
"Pressure sensor":                     //what unit? Is it water pressure?
"Boiler Heating Target Temp.":         //
"2nd Domestic hot water temperature":  //t-DHW1
"Target delta T heating":              
"Target delta T cooling":              

Some questions:
 - How do I read the current additional heat source (electic booster)? 
 - Where can I read the values of the 3-way valves. Have not found any useful data...

giulup, 2021-01-13:
I am coming back with this issue. Seems like some Rotex-Registers are not found in your profiles. Could you explain how to get to the registers. The expansion valve for example is not working. As crow187 mentioned, these are the only compatible registers at the moment.

raomin, 2021-01-13:
The registry definition comes from D-Checker. So if it's not in D-Checker it is not in ESPAltherma. Similarly, if you know where to find it in D-checker, you'll find it in ESPAltherma.

Now, you can also have a look in other definition files to see if you can find the missing information you are looking for and try to get it from the Rotex HPSU.

RE @crow187 question and missing values, I'll paste back what I wrote in #10 here:
- How do I read the current additional heat source (electic booster)?
BUH is the BackUp Heater. I don't know about its consumption value.
When you are unsure about a value, you can look for the register from other pumps. They some time have better names:
eg.
0x62,15, "Pressure sensor" => {0x62,15,152,2,2,"**Refrigerant pressure sensor**"}. But I don't know about the unit.

- Where can I read the values of the 3-way valves. Have not found any useful data...

I'm using `{0x60,12,306,1,-1,"3way valve(On:DHW_Off:Space)"}` -- Mine has only one valve.
Your other valve could be //{0x64,2,301,1,-1,"Bypass Valve Output"},??

For your other comments, some insights: 
- RT Temp should be "Room thermostat temp"
- I'm also wondering about INV Primare vs Secondary currents. If someone have an idea...

giulup, 2021-01-13:
Unfortunately I get all values using the corresponding D-Checker profile. With ESPAltherma the returned values are zero although not possible to be zero.  

Like crow187 said no other values than these are transmitted.

raomin, 2021-01-13:
Which profile do you use in D Checker?

giulup, 2021-01-13:
In D-checker I can use 
ALTHERMA(LT_CA_CB_04-08KW)
As well as
ALTHERMA(LT_DA_04-08KW)

They are giving reasonable values for all data. 
in ESPAltherma I will have to recheck every value to see which one is missing.

raomin, 2021-01-13:
Then, why not use /include/def/ALTHERMA(LT_CA_CB_04-08KW).h ! 😉

raomin, 2021-01-13:
Now if you are still missing some values with ALTHERMA(LT_CA_CB_04-08KW).h selected can you please check if you are correct values for registry 0x30? (ie lines starting with `{0x30,`). There could potentially be a problem of address for this one.

raomin, 2021-01-13:
Ah, just so you know, I've just added translated versions of the definition. German is supported 😉

giulup, 2021-01-14:
That’s the problem. I am using include/def/ALTHERMA(LT_CA_CB_04-08KW).h and the functionality is not the same as in D checker. 

I will take a look on the registers “0x30”

giulup, 2021-01-14:
Took some time but here is a list of values that are not supported within ESPAltherma. Most of them are not even necessary for me. At least the expansion valve would be the most interesting value at the moment.

`{0x00,0,801,0,-1,"*Kältemittel Typ"},
{0x00,1,996,1,-1,"Override CMD:0x00 N_INV=1"},
{0x00,2,996,0,-1,"Override CMD:0x00 N_STD=0"},
{0x00,3,996,2,-1,"Override CMD:0x00 N_FAN=2"},
{0x00,4,996,2,-1,"Override CMD:0x00 N_EV=2"},
{0x00,5,996,1,-1,"Override CMD:0x00 N_20S=1"},
{0x00,6,996,1,-1,"Override CMD:0x00 N_CH=1"},
{0x00,7,996,3,-1,"Override CMD:0x00 N_SV=3"},
{0x00,0,152,1,-1,"Anzahl - Sensoren"},
{0x00,1,152,1,-1,"Anzahl - Inverterverdichter"},
{0x00,2,152,1,-1,"Anzahl – Standardverdichter"},
{0x00,3,152,1,-1,"Anzahl - Außengerätelüfter"},
{0x00,4,152,1,-1,"Anzahl - Expansionsventile"},
{0x00,5,152,1,-1,"Anzahl - 4 Wege Ventile"},
{0x00,6,152,1,-1,"Anzahl - Ölsumpfheizungen"},
{0x00,7,152,1,-1,"Anzahl – Magnetventile"},
{0x00,8,152,1,-1,"Maximal anschließbare Innengerätezahl"},
{0x00,9,152,1,-1,"Erkannte Innengerätezahl (max. 59 anzeigbar)"},
{0x00,10,152,1,-1,"O/U MPU ID (xx)"},
{0x00,11,152,1,-1,"O/U MPU ID (yy)"},
{0x00,12,105,1,-1,"Außengerät Leistung (kW)"},
{0x10,1,307,1,-1,"Thermostat EIN/AUS"},
{0x10,4,203,1,-1,"Betrieb / Störung"},
{0x10,5,204,1,-1,"Störcode"},
{0x10,8,114,2,1,"Zielverflüssigung in °C"},
{0x10,10,310,1,-1,"Verdichtungsendtemp.Schutz retry"},
{0x10,10,311,1,-1,"Verdichterüberstromschutz retry"},
{0x10,11,310,1,-1,"Hochdruckschutz retry"},
{0x10,11,311,1,-1,"Niederdruckschutz retry"},
{0x10,12,310,1,-1,"Inverterplatinentemp.Schutz retry"},
{0x10,12,311,1,-1,"Nicht im Einsatz"},
{0x11,0,215,1,-1,"O/U EEPROM (1. Ziffer)"},
{0x11,1,215,1,-1,"O/U EEPROM (3. 4. Ziffer)"},
{0x11,2,215,1,-1,"O/U EEPROM (5. 6. Ziffer)"},
{0x11,3,215,1,-1,"O/U EEPROM (7. 8. Ziffer)"},
{0x11,4,215,1,-1,"O/U EEPROM (10. Ziffer)"},
{0x11,5,214,1,-1,"O/U EEPROM (11. Ziffer)"},
{0x00,0,995,1,-1,"NextDataGrid"},
{0x20,2,105,2,1,"O/U Wärmetauscher Temp."},
{0x20,6,105,2,1,"Ansaugleitungstemperatur"},
{0x20,8,105,2,1,"Wärmetauscher Mitteltemp."},
{0x20,10,105,2,1,"Flüssigkeitsleitungstemp. (R6T)"},
{0x20,14,405,2,1,"Druck(T)"},
{0x21,7,105,2,1,"Frostschutz Einlasstemp."},
{0x21,9,105,2,1,"Frostschutz Auslasstemp."},
{0x00,0,995,1,-1,"NextDataGrid"},
{0x30,0,151,2,-1,"Expansionsventils (pls)"},
{0x30,2,151,2,-1,"Y1E-Expansionsventil Mitteldruck_Unterkühler"},
{0x30,4,151,2,-1,"Expansionsventils3 (pls)"},
{0x30,6,151,2,-1,"Expansionsventils4 (pls)"},
{0x30,8,151,2,-1,"Expansionsventils5 (pls)"},
{0x00,0,998,1,-1,"In-Out separator"},
{0x60,1,152,1,-1,"Adresse des Innengeräts"},
{0x60,3,204,1,-1,"Störcode"},
{0x60,5,203,1,-1,"Betrieb / Störung"},
{0x60,6,219,1,-1,"I/U Kapazitätscode"},
{0x60,15,215,1,-1,"Software Hydrobox (xx)"},
{0x60,14,215,1,-1,"Software Hydrobox (yy)"},
{0x61,1,152,1,-1,"Adresse des Innengeräts"},
{0x61,12,105,2,1,"Innenraum Umgebungstemp. (R1T)"},
{0x62,1,152,1,-1,"Adresse des Innengeräts"},
{0x62,2,307,1,-1,"Nachheizen ein/aus"},
{0x62,2,306,1,-1,"Speicher Eco ein/aus"},
{0x62,2,305,1,-1,"Prerisc. serbatoio ON/OFF"},
{0x62,2,304,1,-1,"Hochleistungs-DHW-Betrieb. EIN/AUS"},
{0x62,2,303,1,-1,"Raumheizbetrieb EIN/AUS"},
{0x62,2,302,1,-1,"System AUS"},
{0x62,2,300,1,-1,"Notbetrieb aktiviert/nicht aktiviert"},
{0x62,3,105,2,1,"LW Einstellpunkt (Zusatz)"},
{0x62,5,105,2,1,"RT Einstellpunkt"},
{0x62,7,307,1,-1,"Add. Ext. RT Eingang Kühlen"},
{0x62,7,306,1,-1,"Add. Ext. RT Eingang Heizen"},
{0x62,7,305,1,-1,"Haupt-RT-Kühlung"},
{0x62,7,304,1,-1,"Haupt-RT-Heizung"},
{0x62,7,303,1,-1,"Stromverbrauchsgrenze 4"},
{0x62,7,302,1,-1,"Stromverbrauchsgrenze 3"},
{0x62,7,301,1,-1,"Stromverbrauchsgrenze 2"},
{0x62,7,300,1,-1,"Stromverbrauchsgrenze 1"},
{0x62,8,307,1,-1,"Keine"},
{0x62,8,303,1,-1,"Speicher vorheizen ein/aus"},
{0x62,8,302,1,-1,"Umwälzpumpenbetrieb"},
{0x62,8,301,1,-1,"Alarmausgabe"},
{0x62,8,300,1,-1,"Raumheizung Betriebsleistung"},
{0x62,9,105,2,-1,"Durchflussmenge (l/min)"},
{0x62,11,105,1,2,"Wasserdruck"},
{0x62,12,152,1,-1,"Pumpengeschwindigkeit (0: max-100: Stopp)"},
{0x62,13,152,1,-1,"[Zukunft] 3-Wege-Ventil Mischung 1"},
{0x62,14,152,1,-1,"[Zukunft] 3-Wege-Ventil Mischung 2"},
{0x63,1,152,1,-1,"Adresse des Innengeräts"},
{0x63,2,215,1,-1,"I/U EEPROM (3. Ziffer)"},
{0x63,3,215,1,-1,"I/U EEPROM (4. 5. Ziffer)"},
{0x63,4,215,1,-1,"I/U EEPROM (6. 7. Ziffer)"},
{0x63,5,215,1,-1,"I/U EEPROM (8. 9. Ziffer)"},
{0x63,6,215,1,-1,"I/U EEPROM (11. Ziffer)"},
{0x63,7,214,1,-1,"I/U EEPROM (12. Ziffer)(rev.)"},
{0x64,1,152,1,-1,"Adresse des Innengeräts"},
{0x64,2,316,1,-1,"Hybridbetrieb Modus"},
{0x64,2,303,1,-1,"Boiler Betrieb Anforderung"},
{0x64,2,302,1,-1,"Boiler DHW Anforderung"},
{0x64,2,301,1,-1,"Bypassventil Ausgang"},
{0x64,3,105,2,-1,"BE_COP"},
{0x64,5,105,2,1,"Hybrid Heizung Solltemp."},
{0x64,7,105,2,1,"Boiler Heizung Solltemp."},
{0x64,9,302,1,-1,"Add pump"},
{0x64,9,301,1,-1,"Main pump"},
{0x64,10,118,2,1,"Mixed water temp."},
{0x00,0,996,0,-1,"Override All Clear"}
};`

raomin, 2021-01-14:
Thanks @giulup for reporting this. 
How do you determine that the information is missing. You see this value to be 0 or Off while you have a different value in D-Checker?
In my situation I have a Daikin Altherma and some values are reported 0 or Off (eg BE_COP -  break even coefficient of performance) but I do not recall finding this value in D-Checker either...

raomin, 2021-01-14:
Ah, I read this [here](https://m5stack.com/products/fire-iot-development-kit)
_The GPIO 16 / 17 in Fire is connected to the PSRAM by default, so when you connect or stack other function modules, **you need to avoid conflicts with these two pins to prevent the device from working improperly and causing instability**._
So, if it's indeed a M5Stack, that could explain why... I'll put a warning in the Readme for M5Stack users.
Thanks!

--
_If this project has any value for you, please consider [buying me a beer](https://www.buymeacoffee.com/raomin). I don't do this for money but it feels good to get some support! Thanks :)_

raomin, 2021-01-14:
@giulup, the Expansion valve should be {0x30,2,151,2,-1,"Expansion valve (pls)"} but I did not implement this conversion (151) that's why it was probably returning _conv151 not avail._ 
As it's useful to you and possibly to other, I've just added it.

If you update now, you should have it appearing. On mine it now returns 51203 PLS. 

Tell me if there are some other values that you would need which returns _convXXX not avail._. I just did not want to complexify too much the code for things that I would not use... but if it can be useful for others I would add it.

raomin, 2021-01-14:
As there are few comments around about the values, I started a Wiki page [here](https://github.com/raomin/ESPAltherma/wiki/Information-about-Values) where everyone could add their comments on the value definition. :)

giulup, 2021-01-16:
@raomin thank you for your fast and great work.
The value for the expansion valve does not seem plausible. As well as the "INV frequency". Could you recheck them. I think the range for the expansion valve in D-Checker is in percentage. So 0-100 %. Maybe D-Checker has an internal calculation to display the value.

KalaNagHTD, 2021-01-16:
Inv-frequecy is o.k., fan speed is wrong (should be around 70)! Exp. valve steps should be in the range of 400-900.

KalaNagHTD, 2021-01-16:
„Fan 2 (step)“ delivers the right value for fan speed!

Kala Nag (Karl)

raomin, 2021-01-16:
For Fan1 you see `Fan 1 (10 rpm)` in the Daikin name. I would understand that the unit is  x10 RPM. So I have 8 now, I understand it as 8x10 => 80 RPM.
Could it make sense?

For the expansion valve, PLS refers to [a number of revolution per resolution](http://smartservo.org/en/pls-unit/). So I guess it can differ depending on the model. I'll also have a look at `{0x00,4,152,1,-1,"Expansion Valve Data Qty"},` I believe it's only telling how many values there is... but it's not returning anything for now, as ESPAltherma does not query registry 0x00... yet!

raomin, 2021-01-17:
I've added support for registry 0x00 (and also removing the trailing zeros in the floats).
"Expansion Valve Data Qty" returns.... 0 🤦‍♂️

In DChecker it is also (at least for mine) in PLS: 
[image]

Does anyone has a clue as to INV primary vs secondary?
```c
{0x21,0,105,2,-1,"INV primary current (A)"},
{0x21,2,105,2,-1,"INV secondary current (A)"},
```
Should I sum them to get the consumption of my heat pump? 😟

KalaNagHTD, 2021-01-17:
No, it doesn‘t! It should dispay around 70 which is 700rpm. The item „Fan 2 (step)“ shows the right value.

KalaNagHTD, 2021-01-17:
The primary current is drawn from the mains (230V AC in Germany). Multiplied with the voltage you get the electrical power consumption of the heat pump. The secondary current is the inverter output.

KalaNagHTD, 2021-01-17:
I´ve been overlooking the fact that the registers are different between the Rotex HPSU generations! So it´s very important that we allway consider the unit type when speaking about adresses.

Speaking about the Altherma HPSU6, Ultra I can confirm that 
- the fan speed (rpm) register is o.k. (not matching the register in Altherma LT).
- the Exp. valve (pls) can be found in 0x30, 3, 151. 3. -1
- the Condensing Pressure is "Pressure Sensor" 0x62, 15, 105, 2, 2
- the Water Pressure address 0x62, 11, 105, 1, 2 doesn´t show reasonable values ("0")

crow187, 2021-01-18:
> The primary current is drawn from the mains (230V AC in Germany). Multiplied with the voltage you get the electrical power consumption of the heat pump. The secondary current is the inverter output.

Do you know that for a fact? Otherwise I would try to measure the current input to the outdoor unit and see if it does correlate to the Inv. primary current.

KalaNagHTD, 2021-01-18:
I measured the current and compared! Unfortunately the increments are just 0,5 A. So if you f. i. read 3,5A it might be between 3,26A and 3,74A.

raomin, 2021-01-19:
`The secondary current is the inverter output.` You mean *input* right?
And I also understand that the primary is the total current including the inverter (secondary).
Which is coherent with the results I got:
[image]

Now, if I have the inverter at 15 A (3.7KW) what is taking 12 more Amps? (I didn't graph the BUH but I doubt it is it as it was in Heating only, not in DHW)

KalaNagHTD, 2021-01-19:
How I understand it: The Inverter receives current from the mains (AC 50/60Hz) which is the primary current input). The inverter delivers AC current with variable frequency to the electrical motors which is the secondary current (outpu). While the secondary voltage is lower then the primary voltage, the secondary amperage is higher.

m-reuter, 2021-01-22:
Hi, for me 0x30,0,211,1,-1, shows the right value for the fan speed (at least it is correlated with the noise outside). Also 
0x30,0,152,1,-1,"Drehzahl Verdichter (U/Sek)" = INV frequency (rps) does not show rotation per second, but is the identical value to the fan speed (rotations per minute / 10).

raomin, 2021-02-06:
Hello @maxman01, welcome!
I don't have a Rotex so helping you will be hard for me. Which definition file are you using? 

@KalaNagHTD is a Rotex HPSU expert 😉, maybe he can help.

KalaNagHTD, 2021-02-06:
We have the registers for the HPSU Ultra (2018 up) but not for the earlier versions yet :-(