raomin/ESPAltherma #10: EHS status and 3-way valves

crow187, 2021-01-13:
I have a Rotex hpsu compact ultra 508

I can not find some status information:
 - How do I read the current additional heat source (electic booster)?
 - Where can I read the values of the 3-way valves for DHW and bypass? (3UV DHW and 3UVB1)
 - 

I used the ALTHERMA(HPSU6_ULTRA).h and can confirm following registers contain reasonable values:
"Operation Mode":
"Defrost Operation":
"Low noise control":
"I/U operation mode": current operation state
"Circulation pump operation":
"Alarm output":
"Target Evap. Temp.":
"Flow sensor (l/min)":
"Outdoor air temp.":
"Discharge pipe temp.":
"O/U Heat Exch. Mid-temp.": //sensor in the middle of the outdoor unit
"INV fin temp.": // What is this?
"Voltage (V)":
"INV primary current (A)": //What are primary and secondary current?
"INV secondary current (A)": //
"Outlet Water Heat Exch. Temp. (R1T)": //t-V
"Outlet Water BUH Temp. (R2T)": //t-V,BH
"Refrig. Temp. liquid side (R3T)": //t-liq
"Inlet water temp.(R4T)": //T-R
"DHW tank temp. (R5T)": //t-DHW2
"RT Temp.": //What is this?
"LW setpoint (add)":
"RT setpoint":
"Flow sensor (l/min)": //water flow
"Water pump signal (0:max-100:stop)": //set pump speed
"Pressure sensor": //what unit? Is it water pressure?
"Boiler Heating Target Temp.": //
"2nd Domestic hot water temperature": //t-DHW1
"Target delta T heating":
"Target delta T cooling":

raomin, 2021-01-13:
BUH is the BackUp Heater. I don't know about its consumption value.
When you are unsure about a value, you can look for the register from other pumps. They some time have better names:
eg.
0x62,15, "Pressure sensor" => {0x62,15,152,2,2,"Refrigerant pressure sensor"}. But I don't know about the unit.

I'm also wondering about INV Primare vs Secondary currents. If someone have an idea...

raomin, 2021-01-13:
I'll close this one and continue on #8.

closed 2021-01-13 by raomin