raomin/ESPAltherma #9: upload on M5Stack fails

jockele79, 2021-01-13:
Hello,

I hope somebody can help me here. I tried to prepare the M5 according to the instructions. During upload platformio connects to com 4 port but doesn't continue. Com 4 is detected automatically.

I tried the "easyuploader" from m5stack.com , this works on port com 34.

Unfortunately I'm a layman concerning programming, is it possible to adapt to com 34 or is it possible to change windows7-configuration to com 4?

Another question: Is there a possibility to use the tool by LAN and not WLAN?

Best regards

raomin, 2021-01-13:
Hi @jockele79,

Normally, Platformio detects the right COM port and, if it has a doubt, will ask you to decide.
You can force a specific COM port by adding this line to platformio.ini:
```ini
upload_port = COM34
```
add this under the right `env` section.

The tool is developed to take advantage of the Wifi connectivity of ESP32. The code could be adapted to run on another board with LAN capabilities.

jockele79, 2021-01-13:
ok, thanks. I tried the following:

`[env:m5stickc]
platform = espressif32
board = m5stick-c
framework = arduino
monitor_speed = 9600
upload_speed = 215200
upload_port = COM34`

result:

`Looking for upload port...
Auto-detected: COM4
Uploading .pio\build\esp32\firmware.bin
esptool.py v3.0
Serial port COM4
Connecting.....`

next try:

`[env:esp32]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
monitor_speed = 9600
upload_speed = 215200
upload_port = COM34`

result:
`Looking for upload port...
Use manually specified: COM34
Uploading .pio\build\esp32\firmware.bin
esptool.py v3.0
Serial port COM34
Traceback (most recent call last):
  File "C:\Users\wendelj\.platformio\packages\tool-esptoolpy\esptool.py", line 3969, in <module>
    _main()
  File "C:\Users\wendelj\.platformio\packages\tool-esptoolpy\esptool.py", line 3962, in _main
    main()
  File "C:\Users\wendelj\.platformio\packages\tool-esptoolpy\esptool.py", line 3551, in main
    esp = chip_class(each_port, initial_baud, args.trace)
  File "C:\Users\wendelj\.platformio\packages\tool-esptoolpy\esptool.py", line 271, in __init__
    self._port = serial.serial_for_url(port)
  File "c:\users\wendelj\.platformio\penv\lib\site-packages\serial\__init__.py", line 90, in serial_for_url
    instance.open()
  File "c:\users\wendelj\.platformio\penv\lib\site-packages\serial\serialwin32.py", line 64, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port 'COM34': FileNotFoundError(2, 'Das System kann die angegebene Datei nicht finden.', None, 2)
*** [upload] Error 1
========================================================== [FAILED] Took 34.99 seconds ==========================================================

Environment    Status    Duration
-------------  --------  ------------
esp32          FAILED    00:00:34.987`

raomin, 2021-01-13:
FileNotFoundError
There seems to be no COM34. Only the detected COM4.
Remove the COM34 line and lower the upload speed to 115200.
Also, paste all logs from the upload of COM4.

jockele79, 2021-01-13:
`> Executing task: C:\Users\wendelj\.platformio\penv\Scripts\pio.exe run --target upload <

Processing esp32 (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (2.1.0) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10004.201016 (1.0.4)
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - tool-mkspiffs 2.230.0 (2.30)
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <PubSubClient> 2.8.0
|-- <ArduinoOTA> 1.0
|   |-- <Update> 1.0
|   |-- <WiFi> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|-- <WiFi> 1.0
|-- <EEPROM> 1.0.3
Building in release mode
Compiling .pio\build\esp32\src\main.cpp.o
Generating partitions .pio\build\esp32\partitions.bin
Compiling .pio\build\esp32\libc72\PubSubClient\PubSubClient.cpp.o
Compiling .pio\build\esp32\lib020\Update\Updater.cpp.o
Compiling .pio\build\esp32\libac2\WiFi\ETH.cpp.o
Archiving .pio\build\esp32\lib020\libUpdate.a
Compiling .pio\build\esp32\libac2\WiFi\WiFi.cpp.o
Compiling .pio\build\esp32\libac2\WiFi\WiFiAP.cpp.o
Archiving .pio\build\esp32\libc72\libPubSubClient.a
Compiling .pio\build\esp32\libac2\WiFi\WiFiClient.cpp.o
Compiling .pio\build\esp32\libac2\WiFi\WiFiGeneric.cpp.o
Compiling .pio\build\esp32\libac2\WiFi\WiFiMulti.cpp.o
Compiling .pio\build\esp32\libac2\WiFi\WiFiSTA.cpp.o
Compiling .pio\build\esp32\libac2\WiFi\WiFiScan.cpp.o
Compiling .pio\build\esp32\libac2\WiFi\WiFiServer.cpp.o
Compiling .pio\build\esp32\libac2\WiFi\WiFiUdp.cpp.o
Compiling .pio\build\esp32\liba3b\ESPmDNS\ESPmDNS.cpp.o
Compiling .pio\build\esp32\libbad\ArduinoOTA\ArduinoOTA.cpp.o
Compiling .pio\build\esp32\lib5af\EEPROM\EEPROM.cpp.o
Archiving .pio\build\esp32\libac2\libWiFi.a
Archiving .pio\build\esp32\libFrameworkArduinoVariant.a
Compiling .pio\build\esp32\FrameworkArduino\Esp.cpp.o
Archiving .pio\build\esp32\liba3b\libESPmDNS.a
Compiling .pio\build\esp32\FrameworkArduino\FunctionalInterrupt.cpp.o
Archiving .pio\build\esp32\libbad\libArduinoOTA.a
Archiving .pio\build\esp32\lib5af\libEEPROM.a
Compiling .pio\build\esp32\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\esp32\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\esp32\FrameworkArduino\IPv6Address.cpp.o
Compiling .pio\build\esp32\FrameworkArduino\MD5Builder.cpp.o
Compiling .pio\build\esp32\FrameworkArduino\Print.cpp.o
Compiling .pio\build\esp32\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\esp32\FrameworkArduino\StreamString.cpp.o
Compiling .pio\build\esp32\FrameworkArduino\WMath.cpp.o
Compiling .pio\build\esp32\FrameworkArduino\WString.cpp.o
Compiling .pio\build\esp32\FrameworkArduino\base64.cpp.o
Compiling .pio\build\esp32\FrameworkArduino\cbuf.cpp.o
Compiling .pio\build\esp32\FrameworkArduino\esp32-hal-adc.c.o
Compiling .pio\build\esp32\FrameworkArduino\esp32-hal-bt.c.o
Compiling .pio\build\esp32\FrameworkArduino\esp32-hal-cpu.c.o
Compiling .pio\build\esp32\FrameworkArduino\esp32-hal-dac.c.o
Compiling .pio\build\esp32\FrameworkArduino\esp32-hal-gpio.c.o
Compiling .pio\build\esp32\FrameworkArduino\esp32-hal-i2c.c.o
Compiling .pio\build\esp32\FrameworkArduino\esp32-hal-ledc.c.o
Compiling .pio\build\esp32\FrameworkArduino\esp32-hal-matrix.c.o
Compiling .pio\build\esp32\FrameworkArduino\esp32-hal-misc.c.o
Compiling .pio\build\esp32\FrameworkArduino\esp32-hal-psram.c.o
Compiling .pio\build\esp32\FrameworkArduino\esp32-hal-rmt.c.o
Compiling .pio\build\esp32\FrameworkArduino\esp32-hal-sigmadelta.c.o
Compiling .pio\build\esp32\FrameworkArduino\esp32-hal-spi.c.o
Compiling .pio\build\esp32\FrameworkArduino\esp32-hal-time.c.o
Compiling .pio\build\esp32\FrameworkArduino\esp32-hal-timer.c.o
Compiling .pio\build\esp32\FrameworkArduino\esp32-hal-touch.c.o
Compiling .pio\build\esp32\FrameworkArduino\esp32-hal-uart.c.o
Compiling .pio\build\esp32\FrameworkArduino\libb64\cdecode.c.o
Compiling .pio\build\esp32\FrameworkArduino\libb64\cencode.c.o
Compiling .pio\build\esp32\FrameworkArduino\main.cpp.o
Compiling .pio\build\esp32\FrameworkArduino\stdlib_noniso.c.o
Compiling .pio\build\esp32\FrameworkArduino\wiring_pulse.c.o
Compiling .pio\build\esp32\FrameworkArduino\wiring_shift.c.o
Archiving .pio\build\esp32\libFrameworkArduino.a
Linking .pio\build\esp32\firmware.elf
Building .pio\build\esp32\firmware.bin
Retrieving maximum program size .pio\build\esp32\firmware.elf
Checking size .pio\build\esp32\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  14.6% (used 47816 bytes from 327680 bytes)
Flash: [======    ]  58.1% (used 760942 bytes from 1310720 bytes)
esptool.py v3.0
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM4
Uploading .pio\build\esp32\firmware.bin
esptool.py v3.0
Serial port COM4
Connecting.....`

raomin, 2021-01-13:
That's what you have after ```Connecting.....``` that I'm interested in... :)

jockele79, 2021-01-13:
nothing more, that's where it ends.

Upload started ~1h ago

raomin, 2021-01-13:
?? That's super weird. Try restarting the computer, then if it still fails, reinstall the CP2104 drivers https://m5stack.com/pages/download

raomin, 2021-01-13:
Also you can close other potential disturbing COM applications (eg Cura) and other COM devices (eg deactivate Bleutooth)

jockele79, 2021-01-13:
so I managed to upload the project by using "upload_port = COM34".

When upload is finished it seems the stick is frozen. If i turn of power (6 seconds on power button) I cannot turn it on again.
Only flashing with easyuploader turns on the stick, but after uploading the project I have the same issue :-(

raomin, 2021-01-13:
ESPAltherma is probably correctly working. Don't expect the screen to turn On. ESPAltherma do not use the screen.
If you want to "see" some things happening, you can open the serial monitor [image] while your USB cable is still connected to the M5. You should see the debug message of ESPAltherma activity on your M5.

jockele79, 2021-01-14:
that's all that happens:

`--- Enter port index or full name: COM34
--- Miniterm on COM34  9600,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
`

raomin, 2021-01-14:
Hmm, again, this is very strange behavior. Normally, when the upload finishes, the ESP32 should start running. 
Do not touch the power button. Just open the port monitoring.
Would you have another ESP32 to test on?

jockele79, 2021-01-14:
no reaction after another upload :-(

Could this one work:
https://www.amazon.de/AZDelivery-ESP32-NodeMCU-gratis-eBook/dp/B07Z83MF5W/ref=sr_1_4?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&dchild=1&keywords=esp32&qid=1610621844&sr=8-4

jockele79, 2021-01-14:
it seems to work, I did not change the pin connections, after doing this connection works.
I can also see MQTT topics arriving without values because M5 is not connected yet.

Sorry for the trouble and thank you very much for your patience .
Perhaps it might be useful to point out this issue stronger,  laymen like me will have less problems

raomin, 2021-01-14:
Glad to read it's working!
Can you clarify what is that you did to make it work? Changing RX_PIN and TX_PIN in setup.h??

jockele79, 2021-01-14:
In setup.h I uncommented line 14+15, and activated line 18+19

raomin, 2021-01-14:
That is weird, can you try to undo your changes and try the upload again?

jockele79, 2021-01-14:
ok, I made 3 tests:

1. (original setup.h)
`#define RX_PIN    16// Pin connected to the TX pin of X10A 
#define TX_PIN    17// Pin connected to the RX pin of X10A

//Use these for M5StickC
//#define RX_PIN    36// Pin connected to the TX pin of X10A 
//#define TX_PIN    26// Pin connected to the RX pin of X10A`

**does not work**

2. pin-numbers changed
`#define RX_PIN    36// Pin connected to the TX pin of X10A 
#define TX_PIN    26// Pin connected to the RX pin of X10A

//Use these for M5StickC
//#define RX_PIN    36// Pin connected to the TX pin of X10A 
//#define TX_PIN    26// Pin connected to the RX pin of X10A`

**works**

3. lines 14-15/18-19 changed active/non-active
`//#define RX_PIN    36// Pin connected to the TX pin of X10A 
//#define TX_PIN    26// Pin connected to the RX pin of X10A

//Use these for M5StickC
#define RX_PIN    36// Pin connected to the TX pin of X10A 
#define TX_PIN    26// Pin connected to the RX pin of X10A`

**works**

raomin, 2021-01-14:
Ok, that's bizarre. What is the reference of the board you are using?
M5Stack or M5StickC?

raomin, 2021-01-14:
Ah, I read this [here](https://m5stack.com/products/fire-iot-development-kit):

_The GPIO 16 / 17 in Fire is connected to the PSRAM by default, so when you connect or stack other function modules, you need to **avoid conflicts with these two pins to prevent the device from working improperly and causing instability**._

So, if you are indeed using a M5Stack that explains your issue. I'll put up a warning on the Readme page.

Thanks for testing!

--
_If this project has value for you, please consider [buying me a beer](https://www.buymeacoffee.com/raomin). I don't do this for money but it feels good to get some support! Thanks :)_

jockele79, 2021-01-14:
it's M5StickC

raomin, 2021-01-15:
Hi @jockele79,
I have reproduced the issue with one M5Stick-C I had. M5Stick does need full access to GPIO16/17 to operate properly.
I made it clear (hopefully) in the readme and dedicated a section for M5Stick-C in setup.h

Thanks for reporting this!

closed 2021-01-15 by raomin