raomin/ESPAltherma #11: Altherma EBHQ008BBV3

guillermomolina, 2021-01-14:
Hi, first of all, great work, for this project.
I've successfully connected ESPAltherma  with my EBHQ008BBV3 via the EKCBX008BBV3 control box. It's X10A connector is 8 pins size. I've figured out the pinout successfully. Start receiving messages by MQTT with information via try and error, I have some temperatures and a few states so far. But I was thinking how can I figure out all the rest of the values available? With this Information I could make a contribution to the project.
Here it is a picture of the little beast
[image]

thanks in advance
Guillermo

raomin, 2021-01-14:
Hi @guillermomolina, bravo for your setup! 

It's hard to guess the link between a model and a definition file. Have a look at the content and see for instance if, in the definition file, there is a value for a feature that you don't have (eg, here, you probably shouldn't have DHW information in your definition file). If unsure go with default.h but you might miss some values.

Once you have selected your definition file, you can start by un-commenting all lines. Unsupported values will end up as 0 or OFF. It's hard to tell whether it actually OFF/0 or unsupported but ESPAltherma has no possibility to tell the difference.

I'll put up a page with pictures if different user setups, can I add yours?

guillermomolina, 2021-01-14:
Hi @raomin thank for the info. It is more or less what I end up doing. Some values give me errors. Some give me unavailable transformation. Some give me 0.00000 and others negative values. I don't have the D checker software, if you have it please confirm that my model is not available. 
Sure you can use that. But I will end up making a proper cable and some meaningful info on the device screen. It would end up being prettier. I will upload it then. I am also struggling accessing the MQTT data with openhab.
I will let you know if I end up with something worthy of being published here
Thanks, Guillermo

raomin, 2021-01-14:
All the profiles for Altherma series from D-Checker has been ported to ESPAltherma.
I would say that yours would be most probably the ALTHERMA(MONOBLOC_CA_05-07KW).h

I did not implement all conversion, if you are missing an important one, tell me, I'll try to implement it.

Can't wait to see your final setup! :)

--
_If this project brings value for you, please consider [buying me a beer](https://www.buymeacoffee.com/raomin). I don't do this for money but it feels good to get some support! Thanks :)_

closed 2021-01-14 by raomin