raomin/ESPAltherma #12: JSON data with ioBroker

KalaNagHTD, 2021-01-15:
I use ioBroker as MQTT server and want to use VIS for visualisation. My problem is, that ioBroker doesn‘t recognize the data string coming from ESPAltherma as a valid JSON table and doesn‘t assign a JSON ID. So I can not use the JSON table widget. Any idea is highly appreciated!

raomin, 2021-01-15:
Hi @KalaNagHTD,
I'll have a look at ioBroker requirements but, I've put the JSON produced by ESPAltherma on a JSON validator and it returns as compliant with RFC 8259.
Can you also run a check on your side from the JSON generated?
Do you have any specific logs from ioBroker? Maybe it's just too big...?

KalaNagHTD, 2021-01-15:
I found that the JSON data string is formatted perfectly. I‘m recently tested with just one variable, so the string is as short as possible. What I don‘t know is why ioBroker doesn,t recognize these data as „JSON table“ and therefore doesn‘t assign a JSON ID. Does anyone else use ioBroker with ESPAltherma?

raomin, 2021-01-15:
I don't know ioBroker and I don't speak/read German which makes it a bit difficult to me to understand how it works.
But from what I can understand from [this page](https://forum.iobroker.net/topic/15965/wie-wird-das-widget-json-table-eingestellt-formatiert/28) it seems that ioBroker VIS expects a JSON table with multiple values.

A JSON table is formatted between [], and allow to specify a list of multiple json objects.
To me, this makes little sense to format the json message as a table in this use-case as we are sending only one object (plus this would break HomeAssistant compatibility).

Now if you want to test it on your side, this should be as simple as changing in `include/mqtt.h`:
Line 11: 
```c
char jsonbuff[MAX_MSG_SIZE] = "[{\0";
```
and then, starting at line 25:
```c
  jsonbuff[strlen(jsonbuff) - 1] = '}';
  strcat(jsonbuff,"]");
  client.publish(MQTT_attr, jsonbuff);
  strcpy(jsonbuff, "[{\0");
}
```
This is not fantastic code, but that should do the trick.

KalaNagHTD, 2021-01-16:
Roamin, you are the greatest!
I updated the code and it works as expected!
Thanks alot! Enjoy your fresh beer! :-)

raomin, 2021-01-16:
Thanks @KalaNagHTD! I'll see to make it an option in setup.h

closed 2021-01-16 by raomin

KalaNagHTD, 2021-01-16:
I‘m so happy! :-)

raomin, 2021-01-16:
Hi Karl!
I added the code for ioBroker in the repo. It is deactivated by default and will be activated with this `#define` in `setup.h`:
```c
//Uncomment this line if the JSON message should be in a Json Table. Only for IOBroker Vis: 
//#define JSONTABLE 
```
... so you don't have to do the modification every time :)