raomin/ESPAltherma #13: No data from heatpump

m-reuter, 2021-01-16:
Hi, and thanks for all this amazing work! 

I am trying to get my m5 stickC connected to my heatpump. 
Double checked cabling and all seems fine, also I used your newest code with
#include "def/German/ALTHERMA(LT_CA_CB_04-08KW).h"
pio worked also, but the screen on the stickC looks scrambled and is switched on (is this the way it is supposed to look, also would pull much more power from the heatpump board with the screen on). 

The StickC is getting into Wifi and connects to the MQTT Broker (as I can see in the Home Assistant Mosquitto log). However, 
this is all I get from the sensor.althermasensors (by the way, the name "sensor.althermasensors" is different from what is in your readme): 

M5VIN: 4.949V
M5AmpIn: 107.500mA
M5BatV: 4.165V
M5BatCur: 0.000mA
M5BatPwr: 0.000mW
friendly_name: AlthermaSensors

How can I debug this? Thanks

raomin, 2021-01-17:
Hi @m-reuter, thanks and welcome!

The very latest code (few minutes ago) handles the screen properly: it displays the logs at the beginning and shuts off until you press the main button. 

You can look at the logs from the warmth of your mqtt broker also: 
```bash
~$ mosquitto_sub -v -t "espaltherma/#"
```

Now, if you get nothing from the heatpump, cross check again the RX TX Pins from `setup.h`. Remember they are crossed: the RX of ESP goes to TX of the heatpump X10A.

What is the model of your heat pump?

m-reuter, 2021-01-17:
Thanks for the fast reply. 

here a pic of the cabling:

https://up.picr.de/40339297na.jpg

5v red: 1 (pin on the right in the photo)
G26 orange 2
G36/G25 green 3
GND brown 5
(blue and yellow are not connected at the moment). 

How can I run the mosquitto_sub on my raspi  (my home assistant runs in a docker and so does the mosquitto)? mosquitto_sub is not available in the OS. 

I just checked the log and am getting these:
1610842006: New connection from 192.168.178.61 on port 1883.
[INFO] found mqtt on local database
1610842006: New client connected from 192.168.178.61 as ESPAltherma (p2, c1, k15, u'mqtt').
1610843106: Socket error on client ESPAltherma, disconnecting.
1610843180: New connection from 192.168.178.61 on port 1883.
...

And currently no data at all in sensor.althermasensors

Could it be that I need to power the StickC via USB? Although I still see it in Wifi Clients on my Network. 

My heat pump is the Rotex HPSU Compact Ultra RHSX08P50D  (aka: DAIKIN Altherma 3 R ECH2O )

raomin, 2021-01-17:
Your RX and TX are reversed. You should have this:
[image]

m-reuter, 2021-01-17:
Thanks, will try tomorrow. But then your drawing in the readme is incorrect.

raomin, 2021-01-17:
Damn, yes, my drawing is wrong, I forgot to update it!

Your Orange wire should go to G36
and your Green one to G26.

Note: you have an M5Stickc Plus, never had one before but you might have to put the G25 as floating as it shares G36:
   gpio_pulldown_dis(GPIO_NUM_25);
   gpio_pullup_dis(GPIO_NUM_25);

m-reuter, 2021-01-17:
Where would I need to put those lines?

raomin, 2021-01-17:
In main.cpp line 147
```c
  digitalWrite(PIN_THERM, HIGH);
  gpio_pulldown_dis(GPIO_NUM_25);
  gpio_pullup_dis(GPIO_NUM_25);
  EEPROM.begin(10);
```

raomin, 2021-01-17:
Ok, I updated the schema. Thanks for noticing!

raomin, 2021-01-17:
Working now?

m-reuter, 2021-01-17:
yes, just tried it and it works! Thanks. I did not need to change the lines in main.cpp . 

I noticed on my StickC Plus that your screen output only covers 1/4 of the screen the rest is filled with noise intensities, probably the screen on the older stickc is much smaller.

raomin, 2021-01-17:
Great I'm glad it works!
Maybe I'll make another environnement for the M5stickc plus, but I don't have any at hand to test at the moment. 

You can try, it should be as easy as changing the library from m5stickc.h to `#include <M5StickCPlus.h>` and changing also the lib-dep in platformio.ini.
This should be enough to have a proper screen display! 
Not this is not really a critical feature for ESPAltherma 😉

m-reuter, 2021-01-17:
Thanks, it is not critical indeed. I won't open the heatpump again until Tuesday or Wednesday (to attach a permanent USB cable so I have faster access in the future). Will let you know then. 

Also do you know where I would attach the cables for switching on and off (via the relay?). On the Daikini board or on the Rotex board?

raomin, 2021-01-17:
Can't wait to see your final setup.

Regarding controlling your heatpump, I checked what could be the technical drawings for it and if I'm not mistaken, I would say the first 2 pins of J8.

[image]
But better to let @KalaNagHTD confirm indeed!

KalaNagHTD, 2021-01-17:
You should connect to the RT („room thermostat“) connector J16, pin 1 and 2!

closed 2021-01-17 by raomin