raomin/ESPAltherma #14: ESP8266 support

andreapier, 2021-01-17:
Hi, thanks for this great repository!  
Can this work with an ESP8266?  
If yes, can you provide some hint on what needs to be changed please?  
Thanks again,
Andrea

raomin, 2021-01-17:
Hi @andreapier,
ESPAltherma needs 1 dedicated serial port for communication With the heat pump. 
The main issue is that esp8266 has only 1.5 serial port (one for communication with the pc, the other is tx only).
There could be a possibility to use a software serial library for that, but really, the price difference now between the esp8266 and the esp32 is not worth the hassle. Go with an esp32! 😋

andreapier, 2021-01-17:
Sure, but I already have a spare ESP8266 so just wanted to double check if it could be of any use.  
Will get an ESP32 then, thank you

closed 2021-01-17 by andreapier