raomin/ESPAltherma #15: Q: possible integration alongside official equipment

misko903, 2021-01-19:
Question: Is it possible to use it alongside with **Altherma LAN Adapter** and **Madoka wired thermostat**?
Thanks,
Michal

raomin, 2021-01-20:
ESPAltherma connects to the internal diagnostic port that is not the P1/P2 can bus used by Altherma LAN Adapter and Madoka wired thermostat. So yes, the 3 can coexist.

Note that if you want to use the optional relay as external thermostat, then your Madoka thermostat will still work but only to adjust LWT.

closed 2021-01-20 by raomin

misko903, 2021-01-20:
Thanks for letting me know. Currently, my setup with Smart Grid forces me to use Madoka. Does not matter, It will be nice to see everything in OpenHAB (hopefully 😃) for few bucks. Looks like really nice project 🥇