raomin/ESPAltherma #16: How to identify heat pump model?

andreapier, 2021-01-20:
Hi again,
I my question might sound obvious to you but I don't know how to check my heat pump model...  
The documentation I have does not match any of the definition in `include/def` folder, so I am using the `DEFAULT` profile, but I'd like to use the most appropriate one if possible.  
My internal unit is item code `EHSX08P30D2`, external unit is `ERGA06DAV3`.  
Can you please help me identity the most correct configuration please?

KalaNagHTD, 2021-01-20:
It is a „DA“, so Altherma (LT_DA_04-08_KW) should fit best.

andreapier, 2021-01-20:
Thanks!  
I think a wiki page with details on how to match the best device definition might be a good idea.  
I will try to change definition and let you know if it is all right.  
Again, thanks

closed 2021-01-21 by raomin

andreapier, 2021-01-21:
> Tell me if it's not the case.

It is!  
However, I am still not sure that Altherma (LT_DA_04-08_KW) is the correct option... I get many 0s on channels that I expected to have a value.  
I will double check and report back once I am back home.

andreapier, 2021-01-21:
So, here is my current setup.  
I need values from following channels:
```
{
  "Operation Mode":"Fan Only",         => OK
  "Thermostat ON/OFF":"OFF",           => OK
  "Defrost Operation":"OFF",              => OK
  "Error type":"Conv 203 not avail.",   => KO, not implemented yet
  "Error Code":"Conv 204 not avail.",  => KO, not implemented yet
  "Outdoor air temp.(R1T)":"10",        => OK
  "INV primary current (A)":"0",          => Need to check, machine is idle now
  "INV secondary current (A)":"0",      => Need to check, machine is idle now
  "INV frequency (rps)":"0",                => Need to check, machine is idle now
  "INV frequency 2 (rps)":"0",             => Need to check, machine is idle now
  "Fan 1 (10 rpm)":"0",                       => Need to check, machine is idle now
  "Error Code 2":"Conv 204 not avail.", => KO, not implemented yet. Probably not needed
  "Error detailed code 2":"0",                => KO, not implemented yet. Probably not needed
  "Error type 2":"Conv 203 not avail.", => KO, not implemented yet. Probably not needed
  "Leaving water temp. before BUH (R1T)":"50.7", => OK
  "Leaving water temp. after BUH (R2T)":"46.1",    => OK
  "Refrig. Temp. liquid side (R3T)":"39.7",               => OK
  "Inlet water temp.(R4T)":"46.3",                          => OK
  "DHW tank temp. (R5T)":"47.6",                         => OK
  "Flow sensor (l/min)":"-0.2",                               => KO... it is not supposed to be negative??
  "Water pressure":"0"                                          => KO
}
```
So, it looks like most values are OK, but some are not. It could be an error in the protocol or profile `Altherma (LT_DA_04-08_KW)` is not the correct one.  
What do you guys think?

KalaNagHTD, 2021-01-22:
I just found that you have the internal unit „ECH2O“ with the 300l tank, right? I have the same unit with 500l which makes no difference here. So the even better setup is „HPSU 6, Ultra“! But your current setup is what I started with too and it works with slight corrections. The most zeros you found are due to the idle situation of your unit. Wait until you see it running and go from there!

andreapier, 2021-01-22:
OK, Ill use `HPSU_6_ULTRA` profile then.  
I did some quick testing and it seems to work fine, except for the water pressure value as you said.  
I added my I/U code to the wiki page.  
Thanks!