raomin/ESPAltherma #17: Show your ESPAltherma!

raomin, 2021-01-21:
# ESPAltherma Community post!

> ESPAltherma is used all over the world! well... mostly in Europe... mainly in Germany ;)
> Let's show each other what we did with ESPAltherma!
> This will be nice to see, and will helpnew users can see what can be done with it!

Please post in this _issue_ a small description of your installation with ESPAltherma.
Each post should contain the following information:

### Model and option
- Daikin / ROTEX model 
- ESP32 board type
- if you use the relay or not, the Daikin LAN controller, a custom device on P1P2 etc...

### View of your PCB 
_just copy a picture and paste it_

### Monitoring / control 
_just copy a picture and paste it_
- Describe the solution components (Home Assistant, ioBroker, Graphana etc.)

### Any other info
- specific hacks, tips/tricks, future project etc.
- location, contact information (optional)
- ...

raomin, 2021-01-21:
### Model and option
- Daikin Altherma 3 EAVX-D6V + EPGA016DV. Underfloor heating + cooling (maybe) + DHW
- M5StickC
- Relay as external thermostat

### PCB / ESP32
[image]

### Monitoring & Control
[image]

- Home Assistant
- Home made MQTT ESP thermometer (DS18b20) to feed the climate entity

### Any other info
- First user of ESPAltherma :)
- Future plans: integrate energy limiting modes to ESPAltherma to better leverage peak/off-peak hours (could be used for smart grid also), read about other optimizations - eg avoiding defrost.
- FR, raomin [at] protonmail.com

KalaNagHTD, 2021-01-22:
**Model and option**

- Rotex HPSU Ultra = Daikin Altherma ECH2O = R(E)HSX08P50D2 + R(E)RGA08DAV3
- M5StickC
- ioBroker, Windows 10

**PCB / ESP32**

[img]https://up.picr.de/40380077zg.jpeg[/img]

[img]https://up.picr.de/40380078cz.jpeg[/img]

**Monitoring and control**

[img]https://up.picr.de/40377369tx.jpeg[/img]

IoBroker, Windows 10

KalaNagHTD, 2021-01-23:
With support from the ioBroker community I built some scripts to get data objects from the JSON string and calculate power in/out and COP.

raomin, 2021-01-23:
Good idea! We have indeed everything we need for the COP.
I've just added it to mine.
[image]

Do you confirm the formula?
```yml
  ((state_attr('sensor.altherma','Flow sensor (l/min)')| float * 0.06 * 1.16 * (state_attr('sensor.altherma','Leaving water temp. after BUH (R2T)') | float - state_attr('sensor.altherma','Inlet water temp.(R4T)')|float) )
    /
  (state_attr('sensor.altherma','INV primary current (A)') | float * 250 / 1000))
```
I'll add it to the readme.

KalaNagHTD, 2021-01-23:
All fine except you should use the „Leaving water temp. before BUH(R1T)“

misko903, 2021-01-25:
Guys, this is awesome. I hope I will get so far one day... Mine want to be OpenHAB with Meteostick. Waiting for the parts now.

stosoorok, 2021-01-25:
My Altherma setup:

[image]

More pictures later.

KalaNagHTD, 2021-01-26:
Fine!
From which register have you got the water pressure [bar]?

KalaNagHTD, 2021-01-26:
Fine!
I see with a Daikin I/U (not HPSU) you can get the water pressure.

pengemetal, 2021-01-27:
> All fine except you should use the „Leaving water temp. before BUH(R1T)“. So you can monitor the COP while heating DHW too. And: The Voltage 250V is a bit high! Best is you take the „Voltage (N-phase) (V)“ for calculation (about 230V).
> […](#)

Hi,

My „Voltage (N-phase) (V)“ is always 0.00000 I dont know why? I rewrite the COP sensor fix 228 value but sensor is still unavailable, log said: TemplateError('ZeroDivisionError: float division by zero') while processing template 'Template("{{ ((state_attr('sensor.althermasensors','Flow sensor (l/min)')| float * 0.06 * 1.16 * (state_attr('sensor.althermasensors','Leaving water temp. before BUH (R1T)') | float - state_attr('sensor.althermasensors','Inlet water temp.(R4T)')|float) ) / (state_attr('sensor.altherma','INV primary current (A)') | float * 228 / 1000)) |round(2) }}")' for attribute '_state' in entity 'sensor.espaltherma_cop'
One more question: how do you calculate "Elektr. - und Thermische Leistung"? Thx in advance

raomin, 2021-01-27:
> My „Voltage (N-phase) (V)“ is always 0.00000 I dont know why? 
First, my sensor is called `sensor.altherma`. Yours is probably `sensor.altherma`**sensor**.  You changed most of them but the Voltage one is still sensor.altherma in your formula.
If it's not this, there are 2 possible definitions for „Voltage (N-phase) (V)“ 0x21,2,101,2 and 0x21,4,101,2. You can try with the one you are not using now.

Second this value is 0 when the heat pump is not turning. To avoid ZeroDivisionError when the heat pump is off you can extend the formula like this:
```yaml
{% if is_state_attr('sensor.althermasensors','Operation Mode', 'Heating') and is_state_attr('sensor.althermasensors','Freeze Protection', 'OFF')  %} 
{{ 
  ((state_attr('sensor.althermasensors','Flow sensor (l/min)')| float * 0.06 * 1.16 * (state_attr('sensor.althermasensors','Leaving water temp. before BUH (R1T)') | float - state_attr('sensor.althermasensors','Inlet water temp.(R4T)')|float) )
    /
  (state_attr('sensor.althermasensors','INV primary current (A)') | float * state_attr('sensor.althermasensors','Voltage (N-phase) (V)')|float / 1000))
  |round(2)
}}
{% else %} 0 {%endif%}
```
Like this, the formula will be calculated only when `state_attr('sensor.althermasensors','Operation Mode')` is `Heating`

> One more question: how do you calculate "Elektr. - und Thermische Leistung"? Thx in advance
I'm not sure what you are after but Google translated it as electric and thermal performance. 
This is the ratio done in the formula.
Thermal performance is the Delta temperature x flow (l/min x  0.06 is m3/h) x thermal capacity of water which is 1.16

--
_If this project has value for you, please consider [supporting it](https://www.buymeacoffee.com/raomin). I don't do this for money but it feels good to get some support! Thanks :)_

m-reuter, 2021-01-27:
Hi @pengemetal 
(I used the german names, but you can lookup the english easily in the files):

```
      espaltherma_electrial:
        friendly_name: "Electrical"
        value_template: "{{ state_attr('sensor.althermasensors','INV Primärstrom (A)')  | float * 0.23 }}"
        unit_of_measurement: 'kW'
      espaltherma_rl:
        friendly_name: "Rücklauf Temp"
        value_template: "{{ state_attr('sensor.althermasensors','R4T-Wasser Rücklauftemp. vor dem Plattenwärmetauscher') }}"
        unit_of_measurement: '°C'
      espaltherma_vl1:
        friendly_name: "Vorlauf Temp1"
        value_template: "{{ state_attr('sensor.althermasensors','Auslass Wasser Wärmetauscher Temp. (R1T)') }}"
        unit_of_measurement: '°C'
      espaltherma_durchfluss:
        friendly_name: "Durchfluss"
        value_template: "{{ state_attr('sensor.althermasensors','Durchflussmenge (l/min)') | int * 60 }}"
        unit_of_measurement: 'l/h'
      espaltherma_wl:
        friendly_name: "Wärmeleistung"
        value_template: "{{ ( states('sensor.espaltherma_vl1') | float - states('sensor.espaltherma_rl') | float ) * 0.001163 * states('sensor.espaltherma_durchfluss')  | float}}"
        unit_of_measurement: 'kW'
      espaltherma_cop:
        friendly_name: "COP"
        value_template: "{{ (  ( states('sensor.espaltherma_wl') | float *  states('sensor.espaltherma_invstrom') | float ) / ( states('sensor.espaltherma_invstrom') | float  * states('sensor.espaltherma_invstrom') | float + 0.0001 ) ) | round(2) }}"
```

COP is just: thermal / electrical 
here I did a trick to multiply both numerator and denominator by electrical and add an epsilon to the denominator to avoid division by zero. And to have zero COP if electrical usage is 0.

stosoorok, 2021-01-27:
Yesterday I saw COP -19.43 :) Something fishy about the formula?

[image]

pengemetal, 2021-01-27:
> ```yaml
> {% if is_state_attr('sensor.althermasensors','Operation Mode', 'Heating') and is_state_attr('sensor.althermasensors','Freeze Protection', 'OFF')  %}
> {{ [...]

Hi Guys,

Thank you for all of your help. I changed Voltage to 0x21,2,101,2 and it is working! But it is interesting values are between 110-117 V. I have not rewritten the if parameter yet, I used the original code.

KalaNagHTD, 2021-01-27:
> I changed Voltage to 0x21,2,101,2 and it is working! But it is interesting values are between 110-117 V.
> ```

In depends on the country you live!?

raomin, 2021-01-27:
> Yesterday I saw COP -19.43 :) Something fishy about the formula?

You'll have this negative value when your heat pump is defrosting. It warms up the exchanger, hence giving a very negative heating performance...!

pengemetal, 2021-01-27:
I live in Hungary, 220V thats why it is interesting :) (I wrote 2 * in the code)

raomin, 2021-01-27:
> Thank you for all of your help. I changed Voltage to 0x21,2,101,2 and it is working! But it is interesting values are between 110-117 V. I have not rewritten the if parameter yet, I used the original code.

If this value is changing overtime it's probably not the right register, you are looking at the `INV secondary current (A)` value which is generally at 0x21,2. 

In your case (like in every other except one), Voltage (N-phase) (V) should be in 0x21,4,101,2. Now if you have 0 maybe you have no or broken sensor (what is your model?). Anyway, if your electricity is of decent quality, it should be fairly stable and you can replace it in the formula as you did.

--
_If this project has value for you, please consider [supporting it](https://www.buymeacoffee.com/raomin). I don't do this for money but it feels good to get some support! Thanks :)_

pengemetal, 2021-01-27:
ERLQ011CV3 outdoor   EHBX11CB3V indoor

TingBrouwer, 2021-01-31:
Model and option
Daikin 08AV3 + 08CV3 (heating/cooling)

board type: M5StickC ESP32 PICO
no use off the relay function

[image]

[image]

Monitoring / control
(for now NodeRed and InfluxDB, later Grafana)

[image]

Describe the solution components (Home Assistant, ioBroker, Graphana etc.)
-Roomtemp. data sending to Domoticz, which also controls the 2e input off the unit. (seperate relay)

Any other info
-before used websocket 

location, contact information (optional)
-Netherlands (middle)

Request: the option for a fixed IP address

raomin, 2021-01-31:
Nice work @TingBrouwer. Impressive nodered graph...!
You seem to have an M5StickC-Plus, @m-reuter added a platformio environment for it. Once selected the screen should work properly and use the whole resolution.
Also glad to see ESPAltherma reached Netherlands!

rtenklooster, 2021-02-05:
Hi, first of all, thanx for your great effort! So far it's working great. 
However is someone willing to share their HA configuration?
I can't get it to work. The MQTT autodiscovery isn't picking the data up. Tried to add the sensor manual using:
`- platform: mqtt
    state_topic: espaltherma/ATTR
    name: espaltherma
    `
However the attributes aren't listed. When listening to the topic in HA manual I see all data.

raomin, 2021-02-05:
Hi @rtenklooster,

Welcome in the community!

For HA, do not declare manually, it's better to get mqtt discovery to work :)
Make sure you have it activated in your `configuration.yaml` with this line somewhere: 
```yaml
discovery:
```
And check HA logs, it should tell about the `discovery` module discovering ESPAltherma.

The disconnection is a weird behaviour. I doubt that restarting the ESP would have been enough...
You can try with an OTA update, this shall reboot the ESP remotely.

--
_If this project has value for you, please consider [supporting it](https://www.buymeacoffee.com/raomin). I don't do this for money but it feels good to get some support! Thanks :)_

eldi0s, 2021-02-22:
First of all GREAT work raomin. Thank you very much.

For those who don't want any autodiscovery messing up the HA config here is how you configure the 
sensor manually. As state_topic I gave it the LWT state (Online/Offline) for the main sensor which is
holding all selected values as attributes. For me this was the only reasonable value I wanted to show in the headline. 
But feel free to change with a value template as in my second example.

Sensor showing LWT state as state_topic:

```
sensor:
  - platform: mqtt
    name: "althermasensors"
    state_topic: "espaltherma/LWT"
    json_attributes_topic: "espaltherma/ATTR"
```

Sensor showing whatever attribute you want as state_topic.

```
sensor:
  - platform: mqtt
    name: "althermasensors"
    state_topic: "espaltherma/ATTR"
    value_template: "{{ value_json._whatever_you_want_to_show_ }}"
    json_attributes_topic: "espaltherma/ATTR"
```

I don't use the relay function as I combined ESPAltherma with a Zennio Klic_DA to control the unit, but
it must be similar to this: (untested)

```
switch:
  - platform: mqtt
    name: "altherma"
    state_topic: "espaltherma/STATE"
    command_topic: "espaltherma/POWER"
    availability_topic: "espaltherma/LWT"
    payload_on: "ON"
    payload_off: "OFF"
    payload_available: "Online"
    payload_not_available: "Offline"
```

eldi0s, 2021-02-25:
I also forgot to mention that I'm using this board and had to lower Baudrate to I think 115200 or even lower (don't remember correctly) to get a connection established and the board accepting the firmware upload.

> AZDelivery ESP32 NodeMCU Module WLAN WiFi Dev Kit C Development Board mit CP2102

FlexxFR, 2021-03-02:
Again, great work @raomin  !!
With some help I managed to get it working.
I have a Altherma EHVX11S18CB3V and using a az-delivery-devkit-v4 / WROOM32D ESP32

Connected all 5 pins with dupont wire
[image]

Used several wires to extend and to place ESP outside the case
[image]

Temporary connection, need to fix it somewhere.
I'm using an external power source because it was not working directly connected to the X10A port.
So I only connected the TX/RX and the ground wire.
[image]

No fancy lovelace dashboards yet :)

qris, 2021-03-21:
I haven't installed ESPAltherma on a board. However I am working on a [project](https://github.com/qris/pytherma) to create a Python library and monitoring tools for Daikin heat pumps, and having started to painstakingly reverse-engineer the locations and types of variables in the serial port messages, I discovered that you have already decoded them all. So I wrote some [Python code](https://github.com/qris/pytherma/blob/master/pytherma/espaltherma.py) to parse the ESPAltherma C definition files and output Python code. I also plan to distribute some of those definition files, unmodified (attribution intact), as [part of that project](https://github.com/qris/pytherma/tree/master/data), as permitted by the MIT License.

So many thanks for your massive reverse-engineering effort @raomin !

raomin, 2021-03-21:
Hi @qris, nice to see my work on this project can help others :) 
Good luck with your project! Let me know if you are missing some information!

qris, 2021-03-22:
@raomin thank you! The one thing that would be nice to have, but definitely not essential, is the variable number given to each variable in D-Checker. You can see them in the column headings of the [D-Checker CSV export](https://github.com/qris/pytherma/blob/master/contrib/1_20200601-224524.csv). For example, *O/U MPU ID (xx)* is number *19*.

I have been using these numbers as array indexes, and because the ESPAltherma definitions don't have them, I've had to create a synthetic ID from the definition files instead (page.offset.convid).

I don't mean to ask you to do any significant work, but if you had these IDs available and simply didn't put them into the header file because espaltherma doesn't need them, then I would find them useful, thank you :)

thehijacker, 2021-03-23:
No way. It works :). I have EHVH08S26C9W. Connected one of my spare ESP32 to the serial port and I am getting readings. First thing I done is to buy you a beer. Well deserved!

I ordered M5StickC ESP32 and will replace it with that one and put inside the unit. Currently I put dupont connector to serial pins and extended it with UTP cable outside the unit where I put my ESP32.

Got some additional questions on how to proceed if you cen help me:

- How to figure out what readings my unit can report and what are they? I am interested in power usage (in W), current outside temperature, current set temperature, temperature of in and out water that goes into my floor, etc...
- I wish to separate readings for floor heating pump (turned off in summer) and for sanitary water (always on). Which reading is which? I am not native English speaker and all this fancy variables are confusing :).
- How to get that fancy display in HA? That graphic diagram with readings. That looks so cool.

Thank you for support and this great project. Never knew I have all parts already at home that will allow me to read from my Daikin.

raomin, 2021-03-23:
Hi @thehijacker, thanks for your support :)
Start with ALTHERMA(LT_DA_04-08KW).h definition file.
Basically, you can start by uncommenting all values.

To get Watt you need to multiply `INV primary current (A)` with your voltage `Voltage (N-phase) (V)`
outside temp is ... `Outdoor air temp.(R1T)` !
set temp for your hot water is `DHW setpoint`
set temp for your floor heating is `LW setpoint (main)`
etc...
DHW is Domestic Hot Water
and LW is Leaving Water (that goes to your heating circuit)
All values are [listed here](https://github.com/raomin/ESPAltherma/wiki/Information-about-Values)
I'm sure you'll manage :)

There are also translations available for German, French and Spanish (see the main page).

The black and white engineering-style diagram is from @KalaNagHTD, he is not using HA.
Mine comes from illustrations I found that I turned as `picture-elements` in HA to add the sensors elements on  it.

Have fun with your setup!

thehijacker, 2021-03-23:
Thanks! Used proposed .h file and enabled all but got no json result. Probably was too big. So I removed what I thought was not important.

I now get all this values: https://paste.ubuntu.com/p/WH9h7HJ5Qh/

Thank you for the tips. I will dig into them and try to use what could be important to monitor. Can you look at what I get and you think is also interesting to monitor?

Watt usage is puzzling me. I get values:

```
  "INV primary current (A)": "0.5",
  "INV secondary current (A)": "0",
  "Voltage (N-phase) (V)": "231",
```

To get usage in W that means 0.5 * 231 = 115,5 W? Value seems low to me. What do you think?

Great tip about custom picture diagram. I will "draw" my own and put values on it once I figure them out.

Thank you again!

raomin, 2021-03-23:
.5 A is very low. It should be more around 5. Now this fluctuates a lot, if you looked when it was just getting heating that might be normal, but I doubt it... Now, not sure you need a lot of heating when you have 29.1C indoor! ;)
This is what I get with mine over 24h:
[image]

Overall your values are ok :)

thehijacker, 2021-03-23:
29.1 degree is in the room where I have my inside unit. It is located under stairs where I also I have my communication devices (router, switch, ups, ...). Now that I left it for a while I can see that the values can also go higher than 0.5. On primary and secondary.

[image]

[image]

Why primary and secondary values? Should I use both to calculate power usage?

thehijacker, 2021-03-24:
Sorry to spam this topic. But I really wish to finish this "project" with nice lovelace card in which I would have overview on how my heat pump is working. Few more questions:

- What tool to use to draw schematic of the wiring?
- There are two current properties. Should I sum both and multiply with voltage to get correct power usage?
- Any other tips what would be interesting to monitor?

Thanks.

qris, 2021-03-25:
Hi @raomin, actually there is one other area where I could use your help. In the [LT CA/CB definitions file](https://github.com/raomin/ESPAltherma/blob/main/include/def/ALTHERMA(LT_CA_CB_04-08KW).h), there are some convids used that don't have converters implemented (214, 215, 219, 310, 311, 405, 801, 995, 996, 998). Do you have any information about these convids?

raomin, 2021-03-25:
@thehijacker: The secondary current is the one after the inverter. So it is no representative of your consumption. Just get the primary one and multiply by the voltage to get Watts.
For me getting _a nice lovelace card_ was an iterative process. You start with few sensor values that you are interested in and then look at what else could be missing and you add it...

@qris: I could not find this ID in DChecker. But I'll give it another try. RE the missing converters... best would be that you send me an email: raomin [at] protonmail.com

therealmuaddib, 2021-04-02:
Great project and big thanks to @raomin for ESPAltherma! 👍 I bought you a couple of beers. 🍻😁

I’ve managed to connect the M5StickC, but it’s disconnecting from WiFi after some random time. I have connected only TX, RX and ground. Power for M5StickC is via USB-C. First I thought that this is happening because of weak WiFi signal when I close the unit, but I am having the same issue even if I put M5StickC outside of the unit. I have also noticed that sometimes M5StickC is stuck at connecting to the WiFi regardless that the WiFi is strong enough if the unit is turned on. If I turn it off it immediately connect to the WiFi. So it looks like that if TX, RX and ground are connected then M5StickC is stuck at connecting to WiFi. If I try without these I don’t have this issue. What could be the issue of this problem?

THK56, 2021-04-02:
Many thanks for the huge work. I was "blue-eye" trying to reengineer the protocol of my Altherma interface and finally found the excellent site of @raomin 👍. I'll show my NodeMCU-Altherma implementation as soon that it runs 😊

What turns me actually around is: I can't get a reasonable reply from Altherma. Altherma always replies 0x15, 0xEA when sending a command (like 0x03, 0x40, ...). I tried all sorts of delays aso. Does someone knows what this means? Timeout issue?
I compared the protocol flow of a correctly working DChecker*) application where it seems that DChecker sends the commands for all registers in a quasi "burst" not waiting for a reply after each command.

*) Btw: Dchecker is an application running on a Windows PC and displays all register values for analysing purposes. Field service used it on our installation. The used protocol is exactly what is mentioned in ESPAltherma doc.

raomin, 2021-04-02:
@therealmuaddib: Hi thanks for your support! Much appreciated. I would imagine that the heat pump is creating some sort of noise on the wifi signal. If it is a bit weak this is possible, Now are you also saying that just by disconnecting GND, TX and RX (and leaving the HP running) you get the connection? That would seems weird to me. If you have a spare network cable you can use it to extend the RX TX GND line so your ESP32 can be closer to your Wifi AP.

raomin, 2021-04-02:
@THK56 : It looks like a _misaligned ground_... Are you sure the GND line is properly connected between the ESP32 and the board.

qris, 2021-04-02:
@THK56 I get this same effect when I communicate using serial settings 9600 8n1. When I use 8e1 (even parity), as the ESPAltherma source code uses, then I get answers to requests like the ones sent by ESPAltherma (e.g. 03 40 .. ...) (but not $$$ or a few others sent by D-Checker).

therealmuaddib, 2021-04-03:
@raomin: I have already tried UTP cable which is 5m long to get M5StickC out of the unit and I get the same issue. When I disconnect all cables (RX, TX and ground) the M5StickC is working fine and connects to WiFi flawlessly. Yesterday I first turned on the unit and then M5StickC and it was stuck at connecting to WiFi. And then I removed the cables (RX, TX and ground) and immediately connected. Then I first turned on M5StickC and it connected to the WiFi immediately. After I have turned on the unit it got stuck. Again I have removed the cables and it started working. It’s interesting that now it working for several hours.

THK56, 2021-04-03:
Thanks for your great help @qris : it was indeed the even parity that I 've overseen. My blind spot was the DChecker config without any parity. Now I get reasonable answers back. 
Thanks as well to @raomin and @therealmuaddib: I once more checked your feedback. Wifi and MQTT connections were working fine. I was already on a deep dive on the serial connection with my scope. But once again missed the wrong parity.
That's now a great step forward 👍 next I have to clean my code from all unneeded slopes I wrote to circle in the issue 😊.

raomin, 2021-04-03:
@THK56 I didn't understand you were trying to communicate with the serial port on a project of yours. Funny you report this. I remember the hours I spent on the scope trying to understand why my sequence wasn't understood by the Altherma. Until I found out it was missing one bit... the parity one!

THK56, 2021-04-05:
@raomin Don't worry, I wasn't as clear as I could be. Here my now working prototype (with optocouplers decouple and level adaption). I took the oc idea from Daikin's interface adapter.
[image]

remycrochon, 2021-04-10:
Bonjour,
Bravo @raomin pour ce développement, je cherchai depuis longtemps une solution pour monitorer ma PAC sur Home Assistant et c'est exactement ce que je cherchais. Simple et Economique à mettre en oeuvre. Ma PAC est une EHBX11CB9KW sans ECS.
J'ai utilisé le fichier LT_DA_04-08kw. J'ai également fabriqué mon CI avec Eagle et une CNC (gravure du CI à l'anglaise).

remycrochon, 2021-04-10:
[image]

remycrochon, 2021-04-10:
[image]

remycrochon, 2021-04-10:
[image]

remycrochon, 2021-04-10:
[image]

remycrochon, 2021-04-10:
[image]

remycrochon, 2021-04-10:
[image]

remycrochon, 2021-04-10:
[image]

remycrochon, 2021-04-10:
J'ai également modifier le fichier mqtt.h afin d'inverser la sortie de pilotage du thermostat

remycrochon, 2021-04-10:
[image]

remycrochon, 2021-04-10:
Autre question @raomin, je publie sur un blog des réalisation HA, est-ce que tu m'autorises à communiquer sur le sujet en faisant référence à ta réalisation?
https://domo.rem81.com

remycrochon, 2021-04-10:
Ci après le code pour calculer le COP:
`  - platform: template
    sensors:
      espaltherma_cop :
        friendly_name : " COP "
        unit_of_measurement : ' COP '
        value_template: >-
          {% if is_state("sensor.espaltherma_mode_fonctionnement", "Heating") and (states('sensor.espaltherma_courant_primaire')|float > 0) %}
            {% set a = states('sensor.espaltherma_capteur_debit') | float %}
            {% set b = states('sensor.espaltherma_temp_eau_depart_echangeur_plaque') | float %}
            {% set c = states('sensor.espaltherma_temp_eau_retour_rt4') | float %}
            {% set d = states('sensor.espaltherma_courant_primaire') | float %}
            {% set e = states('sensor.pzem_totale_tension') | float %}
            {{ ((a*0.06*1.16)*(b-c)/(d*e/1000)) | round(2) }}
          {%else%}
            0
          {%endif%}`

remycrochon, 2021-04-10:
Ci après le code pour calculer la température de départ en fonction de la loi d'eau et des paramètres saisies dans la PAC, c'est uniquement pour me permettre de comprendre comment la PAC calcule la température de départ:

  - platform: template
    sensors:
      espaltherma_temp_dep_calculee :
        friendly_name : "Temp dep calc"
        unit_of_measurement : '°C'
        value_template: >-
          {% set p102=37 %}
          {% set p103=25 %}
          {% set p100=-10 %}
          {% set p101=20 %}
          {% set a=(p102-p103)/(p100-p101) %}
          {% set b=p103-a*p101 %}
          {% set t = states('sensor.espaltherma_temp_air_ext') | float %}
          {% if t<= p100 %}
            {{ p100 }}
          {%else%}
            {% if t>= p101 %}
              {{ p101 }}
            {%else%}
              {{ (-0.4*t)+33 | round(2)}}
            {%endif%}
          {%endif%}

remycrochon, 2021-04-10:
[image]

[207 more replies]