raomin/ESPAltherma #24: Daikin Altherma 3 WiFi adapter SD Card

misko903, 2021-02-04:
Hi,
just to inspire someone, there is a SD Card slot in Altherma 3 (I can confirm it on mine - located on the back of the black user interface plate) which can be used for WiFi SD Card adaptor.
[https://www.daikin.eu/en_us/products/brp069a78.html](https://www.daikin.eu/en_us/products/brp069a78.html).
Maybe something like [SD-WIFI with Card-Reader Module](https://www.aliexpress.com/item/4001095717223.html?spm=a2g0o.productlist.0.0.656229ab8J7W5g&algo_pvid=d41254d6-bef1-4dee-a8a8-e98bd79d3839&algo_expid=d41254d6-bef1-4dee-a8a8-e98bd79d3839-0&btsid=0b0a050116124352460731582e0cf9&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_)
can work there? I am only guessing...
Sorry for posting it here, but this looks like the most relevant place on the whole internet :)

raomin, 2021-02-04:
I guessed the SD card could be used for firmware update, I never thought about using it to add Wifi... What a strange design choice, knowing there is a usb port next to it!

Now, I don't have an SD wifi and the only similar thing I can think of is a SD <-> Mass storage <-> Wifi adapter which seems also to be the functioning of the module you suggest (in addition to a storage <-> esp8266).
I'd love to have a look inside the Daikin "SD card"...
This gave me an idea, I'll plug in a wifi usb dongle on the board. Maybe it could work the same way....

raomin, 2021-02-09:
So I checked but plugging in a WiFi usb stick does not work... I really wonder how this 'sd card' behaves on a computer....

misko903, 2021-02-09:
OK, I am not a PC guy really, but what if it needs to have a specific USB
WiFi - driver inside is different than for the one you plugged in?
We need to find someone with the Daikin SD Card and someone to dismantle it
:)

miso

Gollam, 2021-11-05:
I have that card as well, we just got our heat pump installed 2 days ago. The API is based on websockets :/ I am digging through the internet to find some documentation about it. So far I found ideas like this one: https://community.openenergymonitor.org/t/hack-my-heat-pump-and-publish-data-onto-emoncms/2551/35

Kepro, 2022-10-06:
hey @Gollam did you find anything? :-)

Gollam, 2022-10-06:
Hi, with the wifi card, nothing. I added an ESPAltherma as the others. But! Last week, I found a HA integration, which can scrape the cloud data: https://github.com/speleolontra/daikin_residential_altherma/
So now I have the energy consumption values as well.

pwwukpw, 2022-10-29:
I'm also using both .. System control in HA via WiFi integration and getting more granular data Inc I hope COP from this  ESP integration.

raomin, 2026-10-23:
It's been a while! Nothing to build here in the end, and the FAQ answers "why not the Daikin adapter". Closing with the 2.0 release.

closed 2026-10-23 by raomin