raomin/ESPAltherma #26: ESPAltherma loosing connection

rtenklooster, 2021-02-10:
Hi,
Today I experienced the ESPAltherma lost it's connection to the 8Kw Hybrid Altherma.
[image]
This happened to me before. Last time I had to unplug the ESP from the main board. At started working without issues when I plugged it in.

It worked since last friday without problems. This morning the serial connection was lost.
I don't know what I can do to give you more information that might help you trace the cause of this error?

This are the attributes I am querying.
`{0x10,1,304,1,-1,"Defrost Operation"},
{0x10,6,114,2,1,"Target Evap. Temp."},
{0x20,0,105,2,1,"Outdoor air temp."},
{0x20,4,105,2,1,"Discharge pipe temp."},
{0x20,8,105,2,1,"O/U Heat Exch. Mid-temp."},
{0x20,12,105,2,1,"INV fin temp."},
{0x20,14,105,2,2,"Pressure"},
{0x21,0,105,2,-1,"INV primary current (A)"},
{0x21,2,105,2,-1,"INV secondary current (A)"},
{0x21,4,101,2,-1,"Voltage (V)"},
{0x30,0,152,1,-1,"INV frequency (rps)"},
{0x60,9,105,2,1,"LW setpoint (main)"},
{0x61,2,105,2,1,"Outlet Water Heat Exch. Temp. (R1T)"},
{0x61,4,105,2,1,"Outlet Water BUH Temp. (R2T)"},
{0x61,6,105,2,1,"Refrig. Temp. liquid side (R3T)"},
{0x61,8,105,2,1,"Inlet water temp.(R4T)"},
{0x61,12,105,2,1,"RT Temp."},
{0x62,5,105,2,1,"RT setpoint"},
{0x62,9,105,2,-1,"Flow sensor (l/min)"},
{0x64,5,105,2,1,"Hybrid Heating Target Temp."},
{0x64,7,105,2,1,"Boiler Heating Target Temp."},
{0x10,0,217,1,-1,"Operation Mode"},
{0x64,2,302,1,-1,"Boiler DHW Demand"}`

The only thing I have changed was setting the query interval to 5 seconds instead of 30. This might cause the issue, so I just turned it back to 30 seconds. After uploading the firmware, the connection still isn't working.
The firmware upload did succeed because I can see the interval changed to 30 seconds.

raomin, 2021-02-11:
This is strange. 5s pull interval should work fine.
What are the type of connector your are using? In my case the dupont wire had a tendency to come loose and caused disconnection once. I went for a female pins header which was more stable.
I don't explain what else could be the issue...

rtenklooster, 2021-02-12:
Thanks for your reply. 
I haven't touched it but it started working after two hours. I will keep an eye on it, to see if it happens again.

closed 2021-02-12 by rtenklooster