raomin/ESPAltherma #27: Error codes not available

vasarat1, 2021-02-11:
Hi,

I'm trying to implement automatic notification and logging on heat pump failures or warnings via Home Assistant. 

All of the below return 'Conv xxx not avail.". In my understanding, this means they're not implemented yet - could this be done?

```
{0x60,3,204,1,-1,"Error Code"},
{0x60,5,203,1,-1,"Error type"},
..
{0x10,4,203,1,-1,"Error type"},
{0x10,5,204,1,-1,"Error Code"},
```
In my testing `{0x60,4,152,1,-1,"Error detailed code"}` works for me, but that returns only the latter part of the error code (e.g. 6 in the case of failure 7H-06). In normal working state this parameter returns 0, but the latter part of the error code is 0 in some real failures as well (like EA-00). Therefore, it would be nice to have that first part of the failure code available for unambiguous failure detection.

raomin, 2021-02-11:
Hi @vasarat1, good idea. I'll see what I can do for implementing conversion 203 and 204.

raomin, 2021-02-14:
Hey @vasarat1,

Good news! I managed to implement conversion code for the Error codes!

Upgrade, upload and tell me if it works! :)

Raomin

--
_If this project brings value to you, please consider [supporting it](https://www.buymeacoffee.com/raomin). I don't do this for money but it feels good to get some support! Thanks :)_

vasarat1, 2021-02-14:
Now tested, seems to work very nicely! Thanks a lot, this is a great new feature :)

closed 2021-02-14 by vasarat1

raomin, 2021-02-14:
Thanks for reporting @vasarat1! Glad it's working!