raomin/ESPAltherma #28: No transmission of "Outdoor air temp.(R1T)" or "Heat exchanger mid-temp." when heating is off

eldi0s, 2021-02-22:
I'm experiencing an issue with not getting temperature readings for the above mentioned sensors when Klic-DA (Acts like a Thermostat) turns the heating off due to no demand or the unit itself switches off at outside temperatures above the defined temperature limit. Other values like leaving water or inlet water temp still get udpadet. Does ESPAltherma just not poll them when idle or is the unit not updating those values?

Model Template: LT_DA_04-08KW

raomin, 2021-02-25:
Hello @eldi0s, welcome!
ESPAltherma keeps pooling the same values regardless whether the heatpump is running or not. 
There is no obvious reason for it not to return any value. 
When you say "no transmission" do you see some timeout for the registers in the logs or just the value 0? 
Maybe somehow the connection with the compressor is shutdown by klick-da....?

KalaNagHTD, 2021-02-25:
When the HPSU is eqquipped with an optional outdoor temp sensor and the thermostate switches into standby mode the outdoor unit shuts off completely. This includes the temperature sensors of the outdoor unit. Their output freezes until the thermostate switches on again.

KalaNagHTD, 2021-02-25:
The HPSU/ECH2O shuts the O/U off completely when an optional ext. temp. sensor is installed. This helps saving energy. Therefore the output of the sensors placed in the O/U freezes. You can use the output of the optional sensor. The register is 0x61, 14, 105, 2, 1

eldi0s, 2021-02-25:
My heat pump is a Altherma 3 R W with no additional outdoor temp sensor. Klick DA switches the heating operation mode on or off while having the unit set to LWT (Leaving Water Temperature) mode. No thermostat mode selected. The indoor unit is dispIaying the correct outdoor temperature on its dislpay all the time. Also having no timeouts or errors in the MQTT log while querying the registers.

Listening to topic espaltherma/log
```

Querying register 0x62... 
Querying register 0x61... 
........
Done. Waiting 30 sec...
```

Nevertheless, I will try to poll the external ambient temp sensor register and see what happens. Will keep you posted.

eldi0s, 2021-02-25:
R6T Sensor reports 0

Indoor unit still reports correct value of outside air temp.

Any idea?

eldi0s, 2021-02-28:
I did some double checking today, and ESPaltherma is really freezing on any readings related to the outdoor unit the minute when heating demand is cut-off by the themostat. However, the indoor unit still displays and updates the correct outdoor temperature, There is really 100% no other external ambient sensor installed. So the value must still be sent to the indoor unit. Any idea how to fix this, as I wanted to use the temperature for some automations?

raomin, 2021-03-11:
Hi @eldi0s do you have some update on your issue? Is it still happening?
I thought about it, did you try to power your ESP with an external psu?

eldi0s, 2021-03-11:
The ESP is working flawlessly as it‘s just the values from the outdoor unit which are not getting updated when idle.

Yes problem still there :-(

raomin, 2021-03-11:
That's weird indeed. If the display updates the temp value, it should indeed be transmitted to the indoor unit. I fail to see why it would not place it in the register, and I also never heard of such behaviour...  I don't know either how Klick DA works but could it be  messing with the registers?

eldi0s, 2021-03-12:
Klic-DA is a KNX interface connecting to the Daikin P1/P2 Bus. I don‘t think it is interfering as all other registers work.

raomin, 2021-03-20:
Hey @eldi0s, I'm doing a bit of cleanup in the open issues. Will you be able to test with the Klic-DA disconnected? I still believe it would be worth trying...

eldi0s, 2021-03-20:
Sure, will disconnect next week when the weather gets warmer and heating should go into stand-by

eldi0s, 2021-03-28:
Same issue without Klic-Da installed

raomin, 2021-03-28:
That's weird... You have 0 on 0x20,0,105,2 when the heatpump is not operating... I have no clue...

eldi0s, 2021-04-01:
0x61, 14, 105, 2, 1 is reporting zero. 0x20,0,105,2 seemed to just not update after switching off the unit. I now did some trending and found out it does update the value, but with an interval of 3 hours, which is useless to me. The indoor unit however updates the temperature continuously. Very strange.

[image]

raomin, 2021-04-01:
This is indeed very strange. You could have look to see if the whole outdoor unit is turned off and only power on every 3h as a kind of heart beat. Then check which part is turned on at this specific moment, and try to bridge power to it...
Now this is significant work and risks compared to the price of an extra esp32 and a BMP280 ;)
I believe only a Daikin expert could help here.

eldi0s, 2021-07-18:
Hi, you were right. It‘s an energy saving function that gets activated and cuts power to the outdoor unit. A heartbeat is used every 3h to check if temperature is below specified value to switch power back on. So you can close here. Nothing to do for us :-(