raomin/ESPAltherma #29: Relay switch on Jeedom

inakifeliu, 2021-02-23:
First of all thanks for such great project!

I'm running HA and Jeedom, and while in HA the switch is created, in jeedom no, and I can't find the way to...

Any help would be apreciated!!

Iñaki

eldi0s, 2021-02-23:
I have no experience with Jeedom. Is the sensor working in Jeedmo? 

If you haven't tried you may take a look at issue https://github.com/raomin/ESPAltherma/issues/12
Maybe Jeedmo needs the same JSON table as ioBroker

inakifeliu, 2021-02-23:
Hi @eldi0s,

The sensor is working properly. I attach the screenshot.

[image]

Thanks for your time!

eldi0s, 2021-02-23:
I can see some users reporting issues that MQTT plugin for Jeedom doesn't automatically detect their switches, but adding them manually woks like a charm. Have you tried this?

eldi0s, 2021-02-23:
Another user mentioned when he first issued a manual command to the switch, it got picked up by the auto detection of Jeedom.
Just a guess syntax unknown and untested

In your Jeedom MQTT plugin, go to commands. Add a command action. Give it a name
As topic enter 
`espaltherma/POWER`
As value enter 
`{"POWER":"ON"}`

inakifeliu, 2021-02-24:
Thank you @eldi0s, it worked!

I attach the configuration of the MQTT plugin in case it can help to anyone else...

[image]

closed 2021-02-24 by inakifeliu