raomin/ESPAltherma #31: Definition file for Altherma 3 LT EHVH-D6V

rufik, 2021-03-03:
I have new Altherma 3 LT model EHVH08S18DA6V and I'd like to try ESPAltherma with HA :)
So I haven't found any information which definition file is most suitable for my model?

raomin, 2021-03-03:
Hi @rufik, welcome! 
LT DA 08 I would go with `ALTHERMA(LT_DA_04-08KW).h`! 
Happy installation!

raomin, 2021-03-11:
@rufik, is it ok with you? If so, you can close this issue :)

rufik, 2021-03-12:
I've just collected all stuff needed and I'm gonna install & run this weekend - so let this issue be opened 2-3 days more please.
I'll be back with feedback 😉

rufik, 2021-03-16:
It looks like it's working quite fine, so let's close this issue right now 😁
I'm gonna take a closer look into values...

closed 2021-03-16 by rufik

rufik, 2021-03-19:
Just for the records - all attrs from this EHVH:
```
{
  "*Refrigerant type": "Conv 802 not avail.",
  "Override CMD:0x00 N_INV=1": "Conv 996 not avail.",
  "Override CMD:0x00 N_STD=0": "Conv 996 not avail.",
  "Override CMD:0x00 N_FAN=2": "Conv 996 not avail.",
  "Override CMD:0x00 N_EV=2": "Conv 996 not avail.",
  "Override CMD:0x00 N_20S=1": "Conv 996 not avail.",
  "Override CMD:0x00 N_CH=1": "Conv 996 not avail.",
  "Override CMD:0x00 N_SV=3": "Conv 996 not avail.",
  "Sensor Data Qty": "0",
  "INV compressor Qty": "0",
  "STD compressor Qty": "0",
  "Fan Data Qty": "0",
  "Expansion Valve Data Qty": "0",
  "4 Way Valve Data Qty": "0",
  "Crank Case Heater Qty": "0",
  "Solenoid valve Qty": "0",
  "Max. connectable indoor units": "0",
  "Connected Indoor Unit Qty": "0",
  "O/U MPU ID (xx)": "0",
  "O/U MPU ID (yy)": "0",
  "O/U capacity (kW)": "0",
  "NextDataGrid": "Conv 995 not avail.",
  "In-Out separator": "Conv 998 not avail.",
  "Override All Clear": "Conv 996 not avail.",
  "Operation Mode": "Heating",
  "Thermostat ON/OFF": "ON",
  "Restart standby": "OFF",
  "Startup Control": "OFF",
  "Defrost Operation": "OFF",
  "Oil Return Operation": "OFF",
  "Pressure equalizing operation": "OFF",
  "Demand Signal": "OFF",
  "Low noise control": "OFF",
  "Error type": "Warning",
  "Error Code": "AH",
  "Target Evap. Temp.": "21.0938",
  "Target Cond. Temp.": "0",
  "Discharge Temp. Drop": "OFF",
  "Discharge Temp. Protection Retry Qty": "Conv 310 not avail.",
  "Comp. INV Current Drop": "OFF",
  "Comp. INV Current Protection Retry Qty": "Conv 311 not avail.",
  "HP Drop Control": "OFF",
  "HP Protection Retry Qty": "Conv 310 not avail.",
  "LP Drop Control": "OFF",
  "LP Protection Retry Qty": "Conv 311 not avail.",
  "Fin Temp. Drop Control": "OFF",
  "Fin Temp. Protection Retry Qty": "Conv 310 not avail.",
  "Other Drop Control": "OFF",
  "Not in use": "46080",
  "O/U EEPROM (1st digit)": "Conv 215 not avail.",
  "O/U EEPROM (3rd 4th digit)": "Conv 215 not avail.",
  "O/U EEPROM (5th 6th digit)": "Conv 215 not avail.",
  "O/U EEPROM (7th 8th digit)": "Conv 215 not avail.",
  "O/U EEPROM (10th digit)": "Conv 215 not avail.",
  "O/U EEPROM (11th digit)": "Conv 214 not avail.",
  "Outdoor air temp.(R1T)": "-2.5",
  "O/U Heat Exch. Temp.": "0",
  "Discharge pipe temp.": "48",
  "Suction pipe temp.": "0",
  "Heat exchanger mid-temp.": "-8",
  "Liquid pipe temp.(R6T)": "0",
  "Heat sink temp.": "2.5",
  "Pressure sensor": "0",
  "Pressure sensor(T)": "Conv 405 not avail.",
  "INV primary current (A)": "7.3",
  "INV secondary current (A)": "7.3",
  "Voltage (N-phase) (V)": "230",
[... 90 lines cut ...]
  "[Future] 3 way Valve Mixing 1": "0",
  "[Future] 3 way Valve Mixing 2": "0",
  "I/U EEPROM (3rd digit)": "Conv 215 not avail.",
  "I/U EEPROM (4th 5th digit)": "Conv 215 not avail.",
  "I/U EEPROM (6th 7th digit)": "Conv 215 not avail.",
  "I/U EEPROM (8th 9th digit)": "Conv 215 not avail.",
  "I/U EEPROM (11th digit)": "Conv 215 not avail.",
  "I/U EEPROM (12th digit)(rev.)": "Conv 214 not avail.",
  "Hybrid Op. Mode": "H/P only",
  "Boiler Operation Demand": "OFF",
  "Boiler DHW Demand": "OFF",
  "Bypass Valve Output": "ON",
  "BE_COP": "0",
  "Hybrid Heating Target Temp.": "0",
  "Boiler Heating Target Temp.": "29.9",
  "Add pump": "OFF",
  "Main pump": "OFF",
  "Mixed water temp.": "0",
  "2nd Domestic hot water temperature": "59.1",
  "Target delta T heating": "4",
  "Target delta T cooling": "5"
}
```

Heating is on (undefloor type), weather dependant, Madoka indoor thermostat. A lot of information :)

Renouveaux, 2022-10-13:
HI @rufik,

Do you have connect your ESP on the X9 port ? 
Seem's I have same Heat Pump, but I haven't any X10 port, the X9 was a blue one.

Thank in advance for your replie.

rufik, 2022-10-14:
It's X10A at mine, see pics attached.
[image]
[image]

Renouveaux, 2022-10-14:
Hi,

Thanks for your quick reply.

Seems to be a old PCB board. Mine was more little, so I have to check wich one have RX and TX.

Best regards

[IMG_20210303_181712__01-01]<https://user-images.githubusercontent.com/425558/195770720-ab072c02-ff9a-4a1d-b5b9-80a6eb8e93aa.jpeg>
[IMG_20210316_135325-01]<https://user-images.githubusercontent.com/425558/195770729-6fd460cc-891c-45e6-bb99-d6a35d9ca312.jpeg>

rufik, 2022-10-14:
>  Seems to be a old PCB board. Mine was more little, so I have to check wich one have RX and TX. Best regards 
> https://user-images.githubusercontent.com/7583362/195792996-da5ecde6-c724-4391-9608-f349ecf2998c.png

Isn't it board attached to front case with LCD screen? If yes, it's not that board you're looking for :) The proper board is behind the shield AFAIR, on the main unit.

Renouveaux, 2022-10-15:
Oh, ok, I will open my unit to see what's inside.
Because yes, you are right, I'm looking on the part with LCD screen.

Thanks a lot for your time