raomin/ESPAltherma #32: Wrong Voltage values for EHVX11S18CB3V

FlexxFR, 2021-03-04:
Hi raomin, 
I don't retrieve any good values for voltage.

0x21,4,101,2     Voltage (N-phase) (V), provide a 0
0x21,2,101,2     Voltage (N-phase) (V), provides a value 60-90?, it's something else.

Is there any way to retrieve this value by polling a different register or maybe a way to scan all registers and find the one that represents the voltage?

raomin, 2021-03-11:
Sorry for not replying (here) earlier.
It might be that you don't have a voltage sensor in your heatpump. (model?)
@pengemetal had the same issue on https://github.com/raomin/ESPAltherma/issues/17 

0x21,2,101,2 is most probably Amp x10 (here, so 6-9Amp). 

There are many possible conversions for the registers, so you cannot really imagine going through them all and try to find 230 somewhere. All registers information were deduce from DChecker Daikin application, so I don't think you can expect finding it elsewhere. 
Now don't worry too much about the voltage, in most countries, it's pretty stable. Mine never moved from 230.

FlexxFR, 2021-03-17:
Np, it's not a very big issue but seemed nice to calculate the COP value.
So far the number still don't add up.

I have:
INV primary current (A): 3.3
INV secondary current (A): 5.3
Voltage (N-phase) (V): 53

And I know from several smart plugs / smart meter the voltage level is about 240VDC over here (bit high).
That is indeed stable enough. But what is then the best current value? 
INV primary, secondary or the 0x21,2,101,2 divided by 10 (which is in this example exactly INV secondary....hmmm need to monitor/verify that)

Do the COP calculation only work for normal heating patterns? 
I sometime see it;s in another heating mode for DHW, the leaving water temp are then much higher. Is there a valve or something routing this flow into the tank? I assume when the tank is filled, no water (with that higher temp) is punt into the floor heating so the flow value is for the tank?

KalaNagHTD, 2021-03-18:
Primary Current is from the mains (240V) to the inverter. Secondary is from the inverter to the compressor (higher amps, lower voltage). COP calculation works on DHW too. The heating water is routet by the DHW valve through the tank instead the floor.

KalaNagHTD, 2021-03-18:
DHW position of the valves
[image]

raomin, 2021-03-20:
Hey @FlexxFR, don't 
Just to add to this: 
- don't use 0x21,2,101,2 for your voltage. With your HP, you're getting INV secondary current (A) x10.
So, in the COP formula just replace the voltage with 240 and you should be good.

You'll see more discussion on calculating COP on issue #17.

The leaving water is what you need for the COP, regardless if it goes to your floor or your DHW tank, it's still heat production :)

FlexxFR, 2021-03-21:
ok, thanks again for the support. I will give it another try :)
I'm also trying to retrieve other interesting values to show on a dashboard....there is a lot!

closed 2021-04-01 by raomin

sahni2021, 2021-10-01:
> Primary Current is from the mains (240V) to the inverter. Secondary is from the inverter to the compressor (higher amps, lower voltage). COP calculation works on DHW too. The heating water is routet by the DHW valve through the tank instead the floor.
> 
> [image]

@KalaNagHTD Hi, would you mind sharing the input values you've used to calculate Thermische Leistung und Elektrische Leistung in your nice diagram? What's the secondary inverter current used for? Cheers

KalaNagHTD, 2021-10-01:
Elektrische Leistung = inv. prim. current * 230
Thermische Leistung = Durchfluss [l/h] * (Vorlauf - Rücklauf) * 1,163

sahni2021, 2021-10-02:
> Elektrische Leistung = inv. prim. current * 230 Thermische Leistung = Durchfluss [l/h] * (Vorlauf - Rücklauf) * 1,163 Secondary current is not used for my calculations

Thanks @KalaNagHTD !
I have an Altherma 3 R ECH2O (EHSX08P30DA) and currently use the values from "Outlet Water Heat Exch. Temp. (R1T)" and "Inlet water temp.(R4T)" for my calculations. Is this correct?

KalaNagHTD, 2021-10-02:
That‘s correct!