raomin/ESPAltherma #33: Missing ATTR Value in ioBroker MQTT

aendue, 2021-03-16:
Hi, I have a Altherma 3 R ECH20 (EHSX08P50DA) and used the def/German/ALTHERMA(HPSU6_ULTRA).h include on my M5StickC.
On ioBroker I dont get the ATTR Value with the json. 
See: [https://ibb.co/0Vn5gZ3](url)

Somebody an idea? Is this the right include?

BR

eldi0s, 2021-03-18:
Did you try this? ioBroker expects a different json format.

https://github.com/raomin/ESPAltherma/issues/12

KalaNagHTD, 2021-03-18:
Anyway you should see that:

[image]

raomin, 2021-03-20:
Hello @aendue, did you manage to get this working with the clarifications from @KalaNagHTD?
Can we close this issue?

aendue, 2021-03-22:
Hi, thanks so far. I'm going to try it this week again. Will let you know end of the week. BR

raomin, 2021-04-01:
I believe you found what you were looking for, I'm closing it, feel free to reopen if needed.

closed 2021-04-01 by raomin