raomin/ESPAltherma #34: behaviour when home assistant goes offline

m-reuter, 2021-03-17:
Hi, what is the expected behaviour when home assistant goes offline. For example when I shutdown my raspberry pi, the heat pump stops heating and looses the little icon for the room thermostat. I am usually in automatic mode (but during the day the automation is on). 

- is this the expected behaviour? 
- does it mean the heat pump is working in automated mode (as if the room thermostat is not installed) or rather it means that there is no heating request from room thermostat any longer, which is why it shuts off?
- the ESP is connected via MQTT, so the heat pump should not even know about the Raspi and Homeassistant. It should just continue running?
- or maybe I wired it so that the relay switches off after a short time unless HA tells it not to? Then I should probably swap those cables to the other connection in order to ensure that if my Raspy looses power, it will continue to heat?

Thanks.

KalaNagHTD, 2021-03-17:
What is the „little icon for the room thermostat“? Do you mean the „Ext“ icon?

m-reuter, 2021-03-17:
Yes, EXT and the heat waves next to it, instead I see the icon for automatic 1.

KalaNagHTD, 2021-03-18:
„Ext“ shows up when the thermostat contacts are closed and disappears when they are open.

m-reuter, 2021-03-18:
Thanks, that means the relay opens the contacts when the server goes down .. and Rotex stops heating. 
@raomin is there a way to keep it in the state it was in even if the Raspi goes offline? Because the opposite behaviour is also not wanted (because then it could be that Raspi looses power at night and then Rotex starts heating). 
Thanks

raomin, 2021-03-18:
Hi @m-reuter,
You're raising a good point.
Currently, ESPAltherma is retrying to connect to the MQTT for a 30 secs, then if it still fails it will reboot.

Normally, when ESPAltherma reboots, it restores its previous state. But it does this after connecting to MQTT...!
So in practice, when MQTT is off, it will turn off the relay until MQTT is back.

I still believe that a failure to connect to the MQTT should trigger a reboot at one point (as it may be caused by things that would be cleaned up by a reboot eg, wifi/dhcp issue). Now it could be longer than 30 sec and also the relay state should remain after reboot and while (trying to) connecting to the MQTT.

I'll implement this change.

raomin, 2021-03-18:
Done! 
It should reboot every 500s but restore immediately its state.

Give it a try and tell us!

m-reuter, 2021-03-19:
Thanks. I tried it and it seems to work now. "EXT" symbol stayed active even though the MQTT server was down. 
 However, it looks to me that the m5stick is constantly rebooting (like every couple seconds, not waiting 500s). Is that only the first time ?

raomin, 2021-03-19:
No, it shouldn't reboot. If it is it must be crashing for some reasons I haven't anticipated. I'll try turning off also mqtt.
Did you turn off only the MQTT or also the wifi?

m-reuter, 2021-03-19:
Only the raspberry pi , which is running mqtt and homeassistant. Wifi was still up.

raomin, 2022-11-05:
I guess this was solved by #157

closed 2022-11-05 by raomin