raomin/ESPAltherma #35: Parameters/registries explained

rufik, 2021-03-18:
I've just started to monitor my Altherma 3 LT heat pump (EHVH-D6V) with espaltherma and I'm stunned by a number of parameters/registers read! It's much more better then P1P2 adapter I'm using :) Thank you for your work.

Nonetheless I don't get some of these parameters, so maybe someone else can explain them?

- _INV primary current (A)_ and secondary: is it current draw by Inverter unit? What's _primary_ and _secondary_?
- _Fan 1 (10 rpm)_ and _Fan 2 (step)_: external unit fan rpm? What 1 and 2 does mean here?
- _LW setpoint (main)_ and _(add)_: what's that?
- _STD Compressor 1_ and _2_: what's that? The _2_ values changes but for _1_ it's OFF all the time...

KalaNagHTD, 2021-03-18:
There are a lot of parameters for many different heat pumps. Only a limited number of parameters fits the Altherma 3. Example: Some units have fixed speed compressors (STD), others have inverter controlled compressors (INV). Some units have one fan (Fan 1), others have two (Fan 1, Fan 2). You have to find out what is usefull.

Inv primary current is the current of interest (from mains to Inverter at 240V).

Karl

rufik, 2021-03-18:
OK, I see.
But what does it mean _(10 rpm)_ and _(step)_ for fan? x10 multiplier?
LW is Leaving Water? So what's _setpoint (main)_ and _(add)_?

raomin, 2021-03-18:
Hi @rufik! Thanks for your appreciation! :)

So, here is my understanding:

- (10 rpm) means that you should multiply by 10 to get rpm value. eg 5 -> 50rpm
- LW setpoint (main) would be the target temperature from your _water logic_ (function of the external temp).
- LW setpoint (add) would be the delta you can chose from the _main setpoint_ (on my thermostat it goes from -5 to 5

If added this to the [wiki page on values](https://github.com/raomin/ESPAltherma/wiki/Information-about-Values), fill free to add your understanding there too!

Enjoy!

--
_If this project brings value to you, please consider supporting it. I don't do this for money but it feels good to get some support! Thanks :)_

rufik, 2021-03-19:
Thank a lot!
Is there any attribute/register that indicates DHW tank disinfection operation?

raomin, 2021-03-20:
I'm not aware of such register, no. Sorry. Now you could check, in such case, it is possible that the DHW goes very high.

rufik, 2021-03-21:
Yes, if DHW setpoint is 60 degrees then it's disinfection operation. It's nice workaround :)

There are some "Not is use" parameters but I can see some value returned from them, like "18432". What's that?

raomin, 2021-03-21:
The "not in use" might either be some sensor information that are not documented, or something completely different like a memory area that is just memory junk because reserved for future use....

rufik, 2021-03-24:
Thanks a lot. Is it worth to monitor these "not in use" params (gather in InfluxDB) and take a look into it? Or it's just a waste of time?

I've made some observations within "I/U operation mode" and "Operation Mode" parameters.
It looks like possible values are as follows:

**I/U operation mode:**
```
DHW:Domestic Hot Water	
Heating					
Heating + DHW			
Stop
```
**Operation mode:**
```
Fan Only
Heating
```

As long as I have heating only Altherma, I assume that:
1. _Operation mode = Fan Only_ means heating is completely off?
2. _I/U operation mode = "Heating + DHW" or "DHW:Domestic Hot Water"_ means heating DWH? Why there are different value? Could "DHW:Domestic Hot Water" be shorten to "DHW"?
3. _Defrost Operation_ is superior operation modes, right? If is "ON" then defrosting is in progress?

raomin, 2021-03-24:
> Operation mode = Fan Only means heating is completely off?
Yes
> I/U operation mode = "Heating + DHW" or "DHW:Domestic Hot Water" means heating DWH tank? Why there are different value? Could "DHW:Domestic Hot Water" be shorten to "DHW"?

Good point. I just modified it in the latest commit, You can upgrade your code.

> Defrost Operation is superior operation modes, right? If is "ON" then defrosting is in progress?
Yes :)

rufik, 2021-03-29:
Thanks a lot.

What does _I/U_ abbr stand for? I've noticed that _I/U operation mode == Stop_ means that DHW is completely disabled.

raomin, 2021-03-30:
I/U = indoor unit. So, yes, when it's off, no DHW ;)

raomin, 2021-04-01:
I'll close this issue @rufik. Feel free to reopen it if you're still looking for clarification on the values.

rufik, 2021-04-03:
Sure, no problem :)

1. What's _Startup Control_ parameter?
2. How to know if outdoor unit is working? Checking if _Fan 1/2_ and/or _INV frequency_ param is greater then 0?

raomin, 2021-04-12:
_1. What's Startup Control parameter?_
I have no clue. I believe some parameter that is 1 when the heat pump starts....?

_2. What's Thermostat ON/OFF parameter? Is it DHW State similar to Space heating Operation ON/OFF for house heating?_
This is whether the thermostat is open or closed (ie interior temp is higher or lower the setting). In normal situation, ON should trigger the space heating.

_3. How to know if outdoor unit is working? Checking if Fan 1/2 and/or INV frequency param is greater then 0?_

Check `Operation mode`. If it's Fan Only, it's not heating. If it's `Heating` then check `Indoor Operation mode` that will tell you _what_ it is heating: either `DHW` or `Heating` (ie house heating) or `DHW + Heating` for both.

closed 2021-04-12 by raomin

rufik, 2021-04-13:
3. Operation Mode(s) are not sufficient - it happened once values were `heating` but outdoor unit was inactive, water pump was active, power draw was c.a. 68W, so it was some kind of "sampling" state.
That's why I ask _precisely_ for outdoor unit state parameter. It looks like _Fun 1/2_ or _Inv current_ can be used to determinate O/U state...

habitoti, 2023-08-18:
> * LW setpoint (main) would be the target temperature from your _water logic_ (function of the external temp).
> * LW setpoint (add) would be the delta you can chose from the _main setpoint_ (on my thermostat it goes from -5 to 5

This is quite old, but maybe still relevant: _LW setpoint (main)_ is for the main heating (usually floor-heating), and _LW set point (add)_ is for the secondary heating circle (radiators). This is for bivalent setups.

RS600807, 2023-12-13:
Hi @rufik ,
I also have an Altherma 3 LT ERGA08 (ext unit)+ EHVH08 (int unit). I am about to received my M5stick and get ready to make some tests.
Could you let me know which file you selected among the def repository ? Thanks for your support.

rufik, 2024-01-23:
@RS600807 It was `ALTHERMA(LT_DA_04-08KW).h` file but I can see that it has been renamed into `Altherma(ERGA D EHV-EHB-EHVZ DA series 04-08kW).h`

RS600807, 2024-01-23:
Thanks @rufik !
I finally got the information when reviewing the different posts in the section. I use the defintion file you mentioned, even if it appears that there is a lot of parameters which seem not in use (from what I saw since almost two months of use)