raomin/ESPAltherma #36: Problem with OTA - timeout and ESP is non-funcional

rufik, 2021-03-19:
Today I tired to OTA upload new firmware for my ESP32 devkit but it failed every time :( It looks like a communication problem:
```
esptool.py v3.0
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Uploading .pio\build\esp32\firmware.bin
18:32:39 [DEBUG]: Options: {'esp_ip': 'ESPAltherma.local', 'host_ip': '0.0.0.0', 'esp_port': 3232, 'host_port': 33493, 'auth': '', 
'image': '.pio\\build\\esp32\\firmware.bin', 'spiffs': False, 'debug': True, 'progress': True, 'timeout': 10}
18:32:39 [INFO]: Starting on 0.0.0.0:33493
18:32:39 [INFO]: Upload size: 784320
Sending invitation to ESPAltherma.local 
18:32:39 [INFO]: Waiting for device...
18:32:49 [ERROR]: No response from device
*** [upload] Error 1
=================================================== [FAILED] Took 29.31 seconds ===================================================
Environment    Status    Duration
-------------  --------  ------------
esp32          FAILED    00:00:29.313
============================================== 1 failed, 0 succeeded in 00:00:29.313 ==============================================
The terminal process "C:\Users\rufik\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload', '--environment', 'esp32'" terminated with exit code: 1.
```

- I'm using VS Code, it compiles fine, uploading via USB was fine.
- Windows 10
- I have `espaltherma.local` entry in my hosts file, it resolves fine
- After OTA upload try (unsuccessful) ESP send "Offline" to LWT topic and  stays non-functional
- ESP responds to pings
- after forcing wifi client disconnection, it never comes back (so it stuck I think)

**So what can be wrong?**
I have a problem restarting ESP32 because it's placed in my Altherma unit behind metal cover with 4 screws, so I need to power down&up all heat pump (which is kinda "dirty restart" for me).

I think it would be nice to have some reboot functionality and maybe auto-reboot/auto-reconnect after wifi is lost?

StRobiEst, 2021-03-20:
I guess that your Windows firewall is the problem. esptool sets up a server (in your case "0.0.0.0:33493") and then sends a invite to the ESP32 to download the required file from your computer. But if your firewall blocks the connection, then the ESP hangs and stays waiting for your computer to answer.

rufik, 2021-03-20:
Wow, I thought esptool is pushing firmware to ESP32 on port 3232. But you says that ESP32 is pulling fw image from my PC (esptool server)?
My windows 10 laptop has NOD32 Security controlled by company's administrators. So I need to setup this project on my Linux box - does it work well on Ubuntu/Debian Linux (Code+project+OTA)?

rufik, 2021-03-20:
I've pulled ESP32 out of heat pump, connected via USB to have debug monitoring.
I configured platformio to specify upload flags like host & port:
```
upload_port = ESPAltherma.local
upload_protocol = espota
upload_flags = 
		--timeout=60
		--host_ip=192.168.66.16
		--host_port=32100
```
Still the same error on VSCode:
```
CURRENT: upload_protocol = espota
Uploading .pio\build\esp32\firmware.bin
20:05:55 [DEBUG]: Options: {'esp_ip': 'ESPAltherma.local', 'host_ip': '192.168.66.16', 'esp_port': 3232, 'host_port': 32100, 'auth': '', 'image': '.pio\\build\\esp32\\firmware.bin', 'spiffs': False, 'debug': True, 'progress': True, 'timeout': 60}
20:05:55 [INFO]: Starting on 192.168.66.16:32100
20:05:55 [INFO]: Upload size: 784320
Sending invitation to ESPAltherma.local 
20:05:56 [INFO]: Waiting for device...
20:06:06 [ERROR]: No response from device
*** [upload] Error 1
=========== [FAILED] Took 26.25 seconds ==========
Environment    Status    Duration
-------------  --------  ------------
esp32          FAILED    00:00:26.255
```

But ESP32 serial debug shows this:
```
[E][WiFiClient.cpp:258] connect(): socket error on fd 56, errno: 113, "Software caused connection abort"
[E][Updater.cpp:269] end(): premature end: res:0, pos:0/784320
[E][ArduinoOTA.cpp:370] _runUpdate(): Update ERROR: Aborted
````
So it internet says that it's firewall problem...and it's true! OTA works fine on my PC with firewall disabled.

closed 2021-03-20 by rufik