raomin/ESPAltherma #37: Restart/reboot functionality

rufik, 2021-03-19:
It would be nice to have ESP32 restart functionality :)
- via the dedicated mqtt topic?
- via the http page (portal)?
- something else?

raomin, 2021-03-20:
Hmm... IMO ESPAltherma should be stable enough not to need a restart functionality.
In which condition would you need one?

rufik, 2021-03-21:
I think I should been more precise: restart by user is just "nice to have" feature, but auto-reboot in case of "hung" is very useful. Resetting ESP requires power cut off of heat pump, which is not really good thing.
I don't use ESP32 a lot, but in ESP8266 it's common auto/soft-reboot in case of lost wifi and no re-connection in 1-5min period. The second is case of mine - OTA blocked by firewall, so pulling ESP32 out of heatpump is pain in the ass.

So maybe Watchdog Timer (TWDT?) can be used for such task?

m-reuter, 2021-03-21:
I had connected a usb cable to the esp and have it coming out of the heatpump where the other wires come out. That way I can always update via cable when I want.

raomin, 2021-03-21:
Hey @rufik, thanks for clarifying. ESPAltherma has already this type of 'watchdog' functionality: it monitors MQTT and Wifi connections. If one breaks and cannot recover in few minutes, ESPAltherma will reboot on itself and maintain it's status in the meantime.

rufik, 2021-03-24:
That's very good. Is it possible to apply watchdog to OTA errors? I mean when OTA goes wrong (like `[E][ArduinoOTA.cpp:370] _runUpdate(): Update ERROR: Aborted` in my case) it could reboot ESP as well. Possible?

raomin, 2021-03-24:
In case of an OTA failure, the best is yet to relaunch the OTA.
But still, I added this feature, ESPAltherma in now rebooting on error in OTA.
I also added an MQTT for reboot: send "R" to espaltherma/POWER and it will restart.

closed 2021-03-24 by raomin

rufik, 2021-03-25:
Perfect, thank you :)