raomin/ESPAltherma #38: COP sensor template error (config/logs)

stosoorok, 2021-03-24:
Home Assistant shows 220 erros about COP Sensor template. I think the error came after hass update.

`Logger: homeassistant.components.template.template_entity
Source: components/template/template_entity.py:71
Integration: template (documentation, issues)
First occurred: March 18, 2021, 10:46:59 AM (220 occurrences)
Last logged: 11:22:45 AM

TemplateError('ZeroDivisionError: float division by zero') while processing template 'Template("{{ ((state_attr('sensor.althermasensors','Flow sensor (l/min)')| float * 0.06 * 1.16 * (state_attr('sensor.althermasensors','Leaving water temp. before BUH (R1T)') | float - state_attr('sensor.althermasensors','Inlet water temp.(R4T)')|float) ) / (state_attr('sensor.althermasensors','INV primary current (A)') | float * state_attr('sensor.althermasensors','Voltage (N-phase) (V)')|float / 1000)) |round(2) }}")' for attribute '_state' in entity 'sensor.espaltherma_cop'

I can't figure out what the home assistant don't like about it any more.

raomin, 2021-03-24:
Hmmm do you have a definition like the one on the main page, starting with `{% if is_state_attr('sensor.althermasensors','Operation Mode', 'Heating') and is_state_attr('sensor.althermasensors','Freeze Protection', 'OFF') ` ?

stosoorok, 2021-03-28:
What do you mean by the main page?

I only have in the esp_sensors.yaml:
```
    espaltherma_cop:
      friendly_name: "COP"
      unit_of_measurement: 'COP'
      value_template: "{{ 
        ((state_attr('sensor.althermasensors','Flow sensor (l/min)')| float * 0.06 * 1.16 * (state_attr('sensor.althermasensors','Leaving water temp. before BUH (R1T)') | float - state_attr('sensor.althermasensors','Inlet water temp.(R4T)')|float) )
          /
        (state_attr('sensor.althermasensors','INV primary current (A)') | float * state_attr('sensor.althermasensors','Voltage (N-phase) (V)')|float / 1000))
        |round(2)
      }}"
```

raomin, 2021-03-28:
Have a look here: https://github.com/raomin/ESPAltherma#calculating-cop
The formula will only calculate the COP if the heat pump is running, avoiding the division of by zero.

rufik, 2021-03-29:
Calculating COP is somehow little bit more complicated because of:
- COP below 1 is quite not possible, because of definition: ratio of heat produced to energy used. Simple "heating rod/wire" has COP=1 and that's minimal COP possible by definition. But this formula comes down even below zero when defrosting operation is in progress!
- _Energy used_ should be calculated using not only Voltage and Current values but Power Factor as well. I have SDM630 power meter dedicated to my Altherma only, so I enabled logging of V, A, VA, VAr, PF and I'm going to use it in COP calculation formula in HA. We'll see :)

So what to do with COP < 1? It looks like it should be ignored, right? But defrosting (and idling I think) are factors that lowers down the seasonal COP and should be taken into account...

raomin, 2021-03-29:
Meh... I think it's really a question of personal appreciation. 
A COP below 1 is possible if you are heating outside and cooling inside, which is what happens in defrost, and I like to see how inefficient my heating is when defrosting. Besides it never remains long under 1...

But it will be nice to see if you have a difference between the watt consumption we calculate and the one of the sdm630.
There is obviously a problem as (mine) shows 0 Amp when in standby whereas the pump and UI etc is running...!

stosoorok, 2021-03-29:
> 
> 
> Have a look here: https://github.com/raomin/ESPAltherma#calculating-cop
> The formula will only calculate the COP if the heat pump is running, avoiding the division of by zero.

Yes that line was missing, but now the COP is always 0. No errors sensor is just zero.

stosoorok, 2021-03-31:
Got it working. The problem was in Freeze protection sensor. I just commented it in and made a new sensors and after a restart everything works.

stosoorok, 2021-03-31:
And the Zero Division Error is Back! :(
```

Logger: homeassistant.components.template.template_entity
Source: components/template/template_entity.py:71
Integration: template (documentation, issues)
First occurred: 2:07:49 PM (9 occurrences)
Last logged: 3:52:34 PM

TemplateError('ZeroDivisionError: float division by zero') while processing template 'Template("{% if is_state_attr('sensor.althermasensors','Operation Mode', 'Heating') and is_state_attr('sensor.althermasensors','Freeze Protection', 'OFF') %} {{ ((state_attr('sensor.althermasensors','Flow sensor (l/min)')| float * 0.06 * 1.16 * (state_attr('sensor.althermasensors','Leaving water temp. before BUH (R1T)') | float - state_attr('sensor.althermasensors','Inlet water temp.(R4T)')|float) ) / (state_attr('sensor.althermasensors','INV primary current (A)') | float * state_attr('sensor.althermasensors','Voltage (N-phase) (V)')|float / 1000)) |round(2) }} {% else %} 0 {%endif%}")' for attribute '_state' in entity 'sensor.espaltherma_cop'
```

raomin, 2021-03-31:
Stating the obvious 0 div means that Amp and/or Voltage is 0.
What is the state 'Operation Mode' and/or 'Freeze Protection' when this occurs?

stosoorok, 2021-04-01:
Operation mode is 'Heating' and Freeze Protection is 'OFF'. I don't use Freezing Protection (disabled from the HP menu), it's always off and I don't use other operation modes other than heating.

Voltage is always about 230V and the current is varying 0....20A. When the current is 0 (HP is in idle mode), then the COP is Unavailable not zero.

[image]

[image]

[image]

raomin, 2021-04-01:
Ok, I know understand that you have occasional 'division by 0' error. To avoid them completely, I would suggest to add the following condition in the `if` of the formula.
```
state_attr('sensor.althermasensors','INV primary current (A)') | float != 0  
```
that would make something like:
`{% if is_state_attr('sensor.althermasensors','Operation Mode', 'Heating') and is_state_attr('sensor.althermasensors','Freeze Protection', 'OFF') and state_attr('sensor.althermasensors','INV primary current (A)') | float != 0  %} ` etc.

stosoorok, 2021-04-02:
Thank you again @raomin - now it seems to be fixed!

Current is 0, COP is 0 and HA log shows no division errors! Maybe you should add this to COP sensor example.

raomin, 2021-04-03:
OK I'll update (and simplify) it.

closed 2021-04-03 by raomin

miguelangel-nubla, 2021-06-08:
If anyone wants this to also work while Cooling I use this:
```
{% if
  state_attr('sensor.althermasensors','Operation Mode') in ('Heating', 'Cooling')
  and state_attr('sensor.althermasensors','Freeze Protection') == 'OFF'
  and state_attr('sensor.althermasensors','INV primary current (A)')  | float != 0
%} 
{{ 
  (
    (
      state_attr('sensor.althermasensors','Flow sensor (l/min)') | float
      * 0.06
      * 1.16
      *
        (
          state_attr('sensor.althermasensors','Outlet Water Heat Exch. Temp. (R1T)') | float
          - state_attr('sensor.althermasensors','Inlet water temp.(R4T)') | float
        )
    )
    /
    (
      state_attr('sensor.althermasensors','INV primary current (A)') | float
      * states('sensor.mains_voltage') | float
      / 1000
    )
  )
  | round(2)
  | abs
}}
{% else %}
0
{% endif %}
```

Replace `states('sensor.mains_voltage')` with your own or hardcode your country's voltage eg: `230` for most of Europe or `120` for USA. [Link with list](https://en.wikipedia.org/wiki/Mains_electricity_by_country)

raomin, 2021-06-08:
Great I'll update the readme. Thanks.

zollak, 2023-02-08:
> hi [stosoorok](https://github.com/stosoorok), how did you make the diagram? I would like to create my own heating system too.