raomin/ESPAltherma #40: X10A already used

Freeman656, 2021-03-24:
Hi,
First, thank you for your work, it's impressive...
I'm facing some interesting issue with my installation.
Indeed, the X10A is already used by some other component (I guess it's the bi-zone component)
PFA the pictures of this installation, and the ref of the heat pump : EHVZ04S18CB.
I guess, I'm f****d, but if it's possible to used some sort of Y cable or another port than X10A, please tell me.

[image]
[image]

raomin, 2021-03-24:
Woah, that's super interesting @Freeman656. We really need to know what is going on this port: If your equipment changes some values on your main unit, that could allow us to understand how it is done and turn ESPAltherma into a controller...!

Here is how you could tap into the serial port and record values so we can see what is going on the port: Get one of these USB - TTL module and plug in only the RX port to the TX port of the X10A on your board. Get a serial monitor (eg  RealTerm), set the port to 9600, 8E1 and you should see some bytes going. Let us know if you need help with this.

You could be able to add ESPAltherma on the same port but it has not been made to share the port and might cause issues with what is using it already. Can you follow these 2 wires from the module to see what is connected to it?

Freeman656, 2021-03-24:
I'm glad to here that this could be useful for further enhancements and would be glad to participate.
I have very little knowledge on the electronic part but this seems quite easy to do.
I'll let you know if I managed to dump some data from the X10A port...

Also, you can find in the schematics page 23 of this document https://www.daikin.ch/content/dam/document-library/installation-manuals/heat/air-to-water-heat-pump-low-temperature/ehvz-cb3v/EHVZ-S18CB3V_4PFR401672-1B_IM_Installation%20Manuals_French.pdf
some information about the card which is connected to the X10A.

X10A seems connected to the A6P module (black ISCL-1V3 on the picture).
The A6P is then connected to the bigger board A5P on its X10A port (which has only 2 wires...)

As per documentation, the A5P is the bizone controller, and the A6P is a current loop board (just translated from french, I'm not entirely sure...).

This may be a bit specific to have two zones in the house, one is low temp (ground floor), and one is higher temp (1st floor), so I guess this second control board (A5P) is used for additional components linked to the high temp circuit (sensors, water circuit control, etc).

I'm still a big anxious about the idea of pluging the ESPAltherma at the same time in the same port.... :)

raomin, 2021-03-24:
Let us know if we can be of any help....!

> I'm still a big anxious about the idea of pluging the ESPAltherma at the same time in the same port.... :)

Once we know what there is in parallel we will know how to "share the port" :)

raomin, 2021-03-25:
Ah, reading what I wrote ; I'll try to explain more... You should put your RX port of your USB TTL to the RX pin of the X10A port (pin 3), so we can see what the daughter board is sending to the mother board. 

Note: once you have done this you can also _listen_ to the TX pin of the X10A port to see how it reacts.

Freeman656, 2021-03-26:
Ok I'll try this.

Freeman656, 2021-03-26:
So, I captured around 3-4 minutes of working time.
I completly shutdown the heat pump, and the beginning of this capture is the restart of the heat pump.
Then around 2 mn, I increased the temperature and the heat pump started its stuff...
I hope you can get something with this file. I forgot to save it as HEX but I guess the data are the same anyways.
If you want me to do other recording, just ask, everything is still in place :)

[capture-RX.txt](https://github.com/raomin/ESPAltherma/files/6213434/capture-RX.txt)

raomin, 2021-03-26:
Hi @Freeman656, Thanks a lot for this! :)

Unfortunately, there is nothing much in your file. Only 0xE0 or 0xF0 and a lot of 0x00.
Could you please make sure that the settings are 9600 bps 8 bits and Even parity?
I believe recording in HEX might be needed as the Text/ASCII version might strip out some precious data.

Freeman656, 2021-03-27:
Hi,
Second attempt, still the same values...
Maybe my FTD1232 is not installed correctly, I'll have to check with an arduino or something...
Do you know a program which can be loaded in the M5StickC to record the RX/TX ports and maybe send it via wifi ?

[capture-rx-hex.txt](https://github.com/raomin/ESPAltherma/files/6215917/capture-rx-hex.txt)
[capture-tx-hex.txt](https://github.com/raomin/ESPAltherma/files/6215918/capture-tx-hex.txt)

raomin, 2021-03-27:
Hmm there might just be no communication when you're listening....? Did you chenge your zone temperature setting while capturing?
Also an easy way to test your capture is to connect your RX to the TX of ESPAltherma ;)

Freeman656, 2021-03-29:
Ok, rookie mistake, I forgot to hit the 'Change' button in RealTerm for 8E1 ;)
So here are the file.
One on the TX pin and the other on the RX.
For both, I changed the zone temperature to trigger heating.

[captureTX.txt](https://github.com/raomin/ESPAltherma/files/6223159/captureTX.txt)

[captureRX.txt](https://github.com/raomin/ESPAltherma/files/6223158/captureRX.txt)

raomin, 2021-04-01:
Bonjour @Freeman656!
I started to look at the files. This is pretty interesting, but doesn't look like anything I've seen before...
I'm struggling a bit with the files as there are no timing information: all bytes are in one long stream. So I'm just guessing here what forms a "command" and its response.
I believe you've seen it live. So could you tell me if this assumption of sequence is correct?

Also, can you confirm that the RX and TX are the one of the X10A main board? (not the one of the dual zone)

RX
```
03224199
03224298
0D2261008E0056FF00000000008C
03224199
03224298
0D2261008E0056FF00000000008C
03224199
03224298
0D2261008E0056FF00000000008C
```

TX
```
224110000001C7000000FA005E0164000007
22420D4B4B2A0114000000000000B922617C
224110000001C7000000FA005E0164000007
22420D4B4B2A0114000000000000B922617C
224110000001C6000000FA005E0164000008
22420D4B4B2A0114000000000000B922617C
224110000001C6000000FA005E0164000008
22420D4B4B2A0114000000000000B922617C
224110000001C6000000FA005E0164000008
22420D4B4B2A0114000000000000B922617C
224110000001C5000000FA005E0164000009
```

raomin, 2021-04-01:
What could help, if you have a second USB TTL is to have the logs in parallel and timed...

I do now remember the serial monitoring tool I used it's YAT https://sourceforge.net/projects/y-a-terminal/
It does break lines after a short timeout and can add timestamp.

If you dont, then I could test a bit or even prepare a project for a serial monitor on ESP32...

Freeman656, 2021-04-02:
Ok I found another USB-to-TTL module in my mess...
And I used YAT, I don't know which file in RX and TX, but I guess you will found out quite easily !

The X10A I use is the one on the main module of the heat pump.

[YAT-Log-20210402-081601.log](https://github.com/raomin/ESPAltherma/files/6248001/YAT-Log-20210402-081601.log)
[YAT-Log-20210402-081609.log](https://github.com/raomin/ESPAltherma/files/6248003/YAT-Log-20210402-081609.log)

raomin, 2021-04-02:
Merci @Freeman656 
Can you redo it with this option without this option: 
[image]

This would make the line cut better.

raomin, 2021-04-02:
... and maybe set the timeout to 200ms instead of 500. Thanks!

kaskii, 2021-05-07:
I'm in the same position. I have Daikin Altherma 3 R F (EHVZ08S23DA9W7) and the X10A is already in use. I have tried now 2 different USB to TTL modules. RX pin of the module is only pin I connect and I connect it to pin 3 of X10A. Every other pin is untouched. I have tried to read serial data with putty, some other program and Open Serial Monitor with settings posted above ( 9600 bps 8 bits, Even parity and 1 bit stop) and I get nothing. I tried to change some settings on Daikin and rebooted it but the serial seems to be quiet. Am I doing something wrong? Maybe this unit uses different settings, is there a way to "analyze" the communication and determine the correct configuration based on that?

raomin, 2021-05-07:
Hi @kaskii,
What are you trying to do? 

- If you are trying to plug ESPAltherma despite the used port, you can try. This might have a slight chance to work and there is a slight amount of chance that it will not disturb your setting. So here you'll need all pins plugged in addition.

- If you are trying to provide some information on the data exchanged between the 2 boards then you also need to connect the GND of the USB TTL to the GND of the board and it should be good to get the data.
I was waiting after @Freeman656 to provide some test results but he seems very busy ;)

kaskii, 2021-05-08:
Thanks. Connecting GND did the trick. Here are some files, do they make any sense? I'm not sure what settings I should be using.  

I bought LAN adapter just before I saw this project. I managed to reverse engineer the adapters websocket protocol but it doesn't provide much data. My ultimate goal is to just get all available data and display it in my custom dashboard. I just need to log these sensor data somewhere and your project seems promising start.

[dump 1.txt](https://github.com/raomin/ESPAltherma/files/6445398/dump.1.txt)
[dump 2.txt](https://github.com/raomin/ESPAltherma/files/6445399/dump.2.txt)

raomin, 2021-05-09:
Hey @kaskii, I had a look at your dumps.
They are similar to the one of @Freeman656 but provides interesting timing data.
It appears that the communication is constant every 1 sec. I believe you did the 2 dumps on the same line, so I only see one end of the communication (most probably RX). It would be great if you could do both, ideally in parallel with 2 usb-ttl ; so I could try to make some sense of this protocol that could eventually possibly allow us to control the heat pump over serial....
Note the second channel (TX) data should look like this: `22420D4B4B2A0114000000000000B922617C`, `224110905F01C2006C00FA005E01640000B1`

Now, for your objective of plugging in ESPAlterma in addition to your setup: the communication is too frequent (every sec) to just plug ESPAlterma in addition to the current setup. It's unlikely to break anything (every message has a CRC) but it is also unlikely that you'll get all the expected values.

One option would be that ESPAltherma 'listens' to the messages returned by the main board and send its query right after a message, but it is really unsure that the other module would not also try to talk at the same time. 

What could work in your case is to use a 2 channel relay to _switch_ the serial port to ESPAltherma when it should talk. Basically it would unplug periodically your 'dual zone module', switch the serial port to ESPAltherma to allow for the communication and switch back the the dual zone module when completed. This would mean a 2 channel relay and a little code adjustment to trigger the switch for the communication.

Now, there is also a CAN bus in our heat pumps (the P1P2 port). It also can query and set values and it's the one used by the LAN module but I haven't looked at it too much for the moment and I'm not sure you'll have access to the same values. But you can have a look at https://github.com/Arnold-n/P1P2Serial it's an arduino implementation for this bus.

kaskii, 2021-05-10:
Thank you @raomin for taking time in this, I appreciate it. Sorry for taking time. I had problems with other USB-to-TTL to output anything but I found my old Arduino and used that instead. Here are the files. I'm not sure what settings I should be running with YAT for your convenience, hope these are ok.

[pin2.txt](https://github.com/raomin/ESPAltherma/files/6454002/pin2.txt)
[pin3.txt](https://github.com/raomin/ESPAltherma/files/6454003/pin3.txt)

As for 2 channel relay, would something like [this](https://www.ebay.de/itm/271513574279) work or do I need some special relays? I don't want to cut wires so I will also buy T-splitters or just connectors to extend current connector - if we can make ESP to work ofcourse.

raomin, 2021-05-10:
Yes @kaskii, I was thinking about something like this, you would bridge the signal so they trigger together.
Pin3.txt is coherent with what you and @Freeman656 sent thanks. 
But I have doubt on Pin2.txt, the content makes little sense and is not coherent with what @Freeman656 gave.
Are you sure the serial settings are correct? make sure you set the parity bit to Even.
If you're using YAT, use the following settings:
[image]

Thanks!

raomin, 2021-05-10:
Ah also, I forgot one thing, while logging, if you could change the temp on one zone and note the time when you do, that would be great.

kaskii, 2021-05-10:
[2 - pin2.txt](https://github.com/raomin/ESPAltherma/files/6455117/2.-.pin2.txt)
[2 - pin3.txt](https://github.com/raomin/ESPAltherma/files/6455118/2.-.pin3.txt)
[1 - pin3.txt](https://github.com/raomin/ESPAltherma/files/6455119/1.-.pin3.txt)
[1 - pin2.txt](https://github.com/raomin/ESPAltherma/files/6455120/1.-.pin2.txt)

I forgot how I wired up last time so here are both cases. 

I actually might have changed the temp to 30.

kaskii, 2021-05-10:
[3 - pin2.txt](https://github.com/raomin/ESPAltherma/files/6455216/3.-.pin2.txt)
[3 - pin3.txt](https://github.com/raomin/ESPAltherma/files/6455217/3.-.pin3.txt)

Here is new set of files. So around 00.48.50 there was following values in the sensors menu:
- Outdoor temp: 14C
- Tank temperature: 60C
- Leaving water: 64C
- Leaving water (PHE): 61C
- Leaving water (main): 34C
- Inlet water temp.: 61C
- Refrigerant temp.: 40C
- Flow rate: 17.4 l/min
- Water pressure: 1.3 bar

The UI "front page" had values:
- Tank 60C
- Main 26C
- Additional 28C
- Outside 14C

I changed the language to english and that seemed to mess with the UI. But at least following should be found:
00.50.26 Main zone temp increase
00.50.52 Additional zone to 40C
00:51:05 Additional zone to 28C

I don't see any decimals in the UI for temps.

raomin, 2021-05-10:
1 - pin2.txt is now the proper data I was missing.
2 - pin2.txt is completely different, I would not trust the converter you're using (or the Arduino...?)

so, my first attempt to understand it makes no sense anymore as the value would now be to... 64 deg.
I'll keep digging. Thanks!

Ah also, I thought that you don't need a relay now to test. Just see how your system behaves when you temporarily disconnect the X10A.

kaskii, 2021-05-21:
Ok, I updated the serial settings to SERIAL_8E1 and now I think [the output](https://gist.github.com/kaskii/2c1ee3ecad52950ad363fcae81b4621b) makes more sense. I tried to raise main zone temp at 31-36 sec and lower it at 62-67 sec mark. I didn't write down the temps but I think it was 33 -> 38 -> 33. 

If the output makes sense now, what would be the next step towards using this project on my pump?

raomin, 2021-05-21:
Thanks @kaskii, good job for using the 2 serials on esp32 :). I'll have a look at it over the weekend.
Understanding this new protocol opens the possibility for ESPAltherma to control in details the Heat Pump instead of just reading values.

Now your problem remains that the port is used by another board. And that I don't know what would it make if ESPAltherma talks at the same time. So you need a proper solution for this.
That solution can be as explained earlier: you make a switch (relay) that will "unplug" your board and connect it to espAltherma so it can query and read the registries and once done, it would reconnect your board.

You can test this manually:
1. Disconnect the connector
2. Plug in ESPAltherma let it read values
3. Unplug ESPAltherma
4. Re-plug your board.

It will take you some time to do it but it could tell if this is a valid solution. Then, the esp32 and 2 relays (one for RX one for TX) would do it much faster.

I hope this clarifies it :)

kaskii, 2021-05-21:
I tried to disconnect X10A and in 5-10 seconds I got an error `UA-16: Extension/hydro communication problem`. Reconnecting got rid of the error message and everything seems to be working just fine so I guess it is ok if ESPAltherma wouldn't be fast enough to read data in that timeframe. I'll learn how to design PCBs and order one with needed parts (relays and such) so I don't have to touch original wires and plugs (in case of warranty issues).

I'll try to hook ESPAltherma into my unit this weekend. Any recommendations which definition file I should be using for testing against Altherma 3 R F (EHVZ08S23DA9W7)?

raomin, 2021-05-21:
That sounds good. ESPAltherma is very conservative in term of timing. You can definitely speedup the query process by reducing the waiting time [here](https://github.com/raomin/ESPAltherma/blob/4b38f592dad00e94216b029930ad1f523c2c7ac5/src/main.cpp#L232). Seeing your timing, I believe 30ms would be good enough.

As for the relays, I don't get why you want to deign a PCB for this. You would just need some PH 2mm connectors (eg [this](https://www.aliexpress.com/item/32955251479.html)) to connect the PCB to the dual relay module, and then to go from the relay module to the small connector PCB. You leave the existing cable in place, unpluged.

As for the definition file, have a look here https://github.com/raomin/ESPAltherma/wiki/Choosing-the-right-definition-file, but this one looks good: ALTHERMA(LT_DA_04-08KW).h

If you want to limit the interruption duration, try to limit the number of different registers that you select (the first hex value in the definition lines).

You can also limit the frequency of polling here: https://github.com/raomin/ESPAltherma/blob/4b38f592dad00e94216b029930ad1f523c2c7ac5/src/setup.h#L11

Have fun!

haz-69, 2022-02-02:
Any solution for this yet ? I have same problem with X10A occupied,,,,,

raomin, 2022-02-03:
Hi @haz-69,
So if I were facing this issue, I would keep the connection to the BIZONE module, and ESPAltherma would _switch_ the connection to itself, do the query, get the result and connect back the X10A to the BIZONE module.

This could either be done directly by chaining ESPAltherma with BIZONE (harder but software only)
Or add a bilateral switch (eg CD4066) and ESPAltherma would switch X10A from BIZONE to itself.
Nothing very straightforward, but nothing impossible...

qris, 2022-03-08:
@haz-69 another option is [P1P2Serial](https://github.com/Arnold-n/P1P2Serial).

commoswiss, 2022-04-22:
Hi @raomin ,
Since our last conversation, have you heard about someone using the TS3A5018 ?
I might try this solution, but not really sure about the consequences on the BIZONE components, will the bizone be shutdown during the commutation ? Is there any recovery mode ? what will happend if the heatpump sends a message during espaltherma connection ?
Is there any code already implemented in ESPAltherma project which could connect/disconnect the bizone as needed during the espaltherma does it's thing...

Now, a final word about another way to handle this issue.
Would it be possible for ESPAltherma to act like a sort of proxy which would send and receive all the messages from the heatpump by default, and sometimes use some down time to send it's own messages ?

Freeman656, 2022-04-22:
Hi @raomin ,
Since our last conversation, have you heard about someone using the TS3A5018 ?
I might try this solution, but not really sure about the consequences on the BIZONE components, will the bizone be shutdown during the commutation ? Is there any recovery mode ? what will happend if the heatpump sends a message during espaltherma connection ?
Is there any code already implemented in ESPAltherma project which could connect/disconnect the bizone as needed during the espaltherma does it's thing...

Now, a final word about another way to handle this issue.
Would it be possible for ESPAltherma to act like a sort of proxy which would send and receive all the messages from the heatpump by default, and sometimes use some down time to send it's own messages ?

raomin, 2022-04-22:
HI @Freeman656
I don't know anyone having implemented the bilateral switch method. And as I don't have this setup, I cannot test and reply to your questions. But I don't think you can break anything. 
I expect the bizone protocol to be resistant to interruption, as there is CRC in all protocol I have seen.

What I would do to handle this at time where ESPAltherma would need to pool data:
1. switch the TS3A5018
2. wait a bit for a timeout in case we are in the middle of a message (eg 500ms)
3. ESPAltherma query and response
4. reflip the TS3A5018

I really think it's the best way to do it. Be conservative at first, only pull data once every 5 min.

The proxy way could work but: 
1. it would require an additional serial port we don't have (SoftwareSerial could work though) and
2. with our limited knowledge of this bizone protocol, it might be difficult to know exactly when the message is complete on one end and when to send it to the other side, lastly
3. if the esp32 is down, your bizone is isolated.

Freeman656, 2022-04-22:
Good points !
How long does ESPAltherma takes to pull all the needed data from the main board ?
And how often do we need to pull data ? ( every 10mn ? 1h ?)

raomin, 2022-04-22:
> How long does ESPAltherma takes to pull all the needed data from the main board ?
It depends on how many registries you are querying. I haven't measured it, but I would say something like 100ms per registry.
(one registry holds several values).
> And how often do we need to pull data ? ( every 10mn ? 1h ?)
That's really up to you, data do not change very often. Default is 30s, but you can got for 5m or more...
https://github.com/raomin/ESPAltherma/blob/8feeb7218414fc21edce614726f34c2e9880e340/src/setup.h#L11

TonKro, 2022-05-23:
I also do have a BiZone. Did someone already manage a solution? 
Is the data already known to manage a nice solution with the ESP32 only. Is the X10A a kind of bus protocol? Then you know which data belongs to the readings and which not which you then can pass?

glosair, 2022-09-04:
Just had a system installed, can X12A on AP4 be used instead of X10A on AP1? 
[image]

raomin, 2022-09-04:
Wha @glosair, that's a nice finding.
Where did you find this?

Looks like yes, the X10a is propageted to this other one...
Definitely worth trying!

glosair, 2022-09-04:
Found it in the "Installer reference guide" 

https://www.daikin.pt/content/dam/document-library/Installer-reference-guide/as/EWAQ006-008BAVP,%20EWYQ006-008BAVP,%20EKCB07CAV3,%20EK2CB07CAV3,%20EKMBUHCA3V3,%20EKMBUHCA9W1_4PEN492901-1_2017_06_Installer%20reference%20guide_English.pdf

Freeman656, 2022-09-09:
Great news !
Is the wiring/layout the same ?
This seems like some kind of daisy chaining of the X10A over multiple modules...

Freeman656, 2022-09-09:
I tried the same, and I confirm this is working fine !

@raomin Finally !! It's working !

haz-69, 2022-09-09:
Does the x12 exist on Altherma EHVZ08S18DA9W
?

glosair, 2022-09-09:
> Does the x12 connectit exists on Altherma EHVZ08S18DA9W ? [image]

Having a look at the installation manual I would say it's the blue connection on A5P but don't hold me to it ;) 

[image]

https://www.manualslib.com/download/1360382/Daikin-Ehvz04s18da6v.html

glosair, 2022-09-09:
> Great news ! Is the wiring/layout the same ? This seems like some kind of daisy chaining of the X10A over multiple modules...

If you look at the green connector above the blue and white ones it appears to be a current loop interface connected to the serial interface (X10A) on the heat pump controller.

glosair, 2022-09-09:
Spafrost who was your installer?

TonKro, 2022-09-09:
On the board next to the board with X10a there is indeed another board with X12A
[image]

glosair, 2022-09-09:
Yup

TonKro, 2022-09-10:
I can confirm as well it works via X12A. Equipment  EHVZ08S23E6V bizone using ALTHERMA(BIZONE_CB_04-08KW).h.

raomin, 2022-09-10:
That's great news everyone!
Thanks for reporting success, I'll update the readme page!

closed 2022-10-18 by raomin

haz-69, 2022-10-19:
> I can confirm as well it works via X12A. Equipment EHVZ08S23E6V bizone using ALTHERMA(BIZONE_CB_04-08KW).h.

Can some show a picture of the X12A pin configuration ?

glosair, 2022-10-19:
It's th same pinout as in the main instructions

glosair, 2022-10-19:
The connector is keyd but top should be pin1 as marked on the PCB

haz-69, 2022-10-19:
[image]
You mean this is pin1?
[image]

glosair, 2022-10-19:
Yes, that's the badger, will match up with this
[image: image.png]

glosair, 2022-10-19:
[image]

tobylorenz, 2025-03-20:
Hi raomin,

I just found your traced that you posted on April 1st 2021.

Pretty helpful. It's basically a sequence of request and responses:
```
> 03224199
<   224110000001C7000000FA005E0164000007

> 03224298
<   22420D4B4B2A0114000000000000B9

> 0D2261008E0056FF00000000008C
<   22617C
```

This continues endlessly. Removing the framing, so length prefix and checksum suffix, this shows two requests without further arguments, so likely are status request:
- Request `22 41` is followed by a 15-byte response `10000001C7000000FA005E01640000`.
- Request `22 42` is followed by a 12-byte response `0D4B4B2A0114000000000000`.

The next one transmits many parameters and is just shortly acknowledged, so likely is a controlling request:
- Request `22 61` with args `008E0056FF0000000000` is followed by 0-byte response, so acknowledge only.

I run the first two status requests on my heat pump EGSAH10D9W:
```
> 03 22 41 99
< 22 41 10 00 00 01 ab 00 00 00 2c 01 2c 01 3c 00 00 4a

> 03 22 42 98
< 22 42 0d 4b 4b 41 00 14 00 00 00 00 00 00 a3
```

I haven't tried the `22 61` command yet. I first try to understand what the parameters mean, not to break anything.

Bye
Tobias

tobylorenz, 2025-03-20:
I found that all the `22 4x` commands are status requests. My heat pump responses to these:
```
> 03 22 40 9a
<    22 40 04 00 05 94
> 03 22 41 99
<    22 41 10 00 00 01 ab 00 00 00 31 01 31 01 3c 00 00 40
> 03 22 42 98
<    22 42 0d 4b 4b 41 00 14 00 00 00 00 00 00 a3
> 03 22 43 97
<    22 43 06 00 00 00 00 94
> 03 22 44 96
<    22 44 06 00 50 00 80 c3
```
The third byte of the response is a fixed length information.

All `22 6x` commands are for control. I have `22 60`, `22 61`, `22 62`, `22 63`, `22 64`, `22 65` in my firmware.