raomin/ESPAltherma #41: Problem uploading to ESP32

Oystein-github, 2021-03-24:
I have a Daikin Altherma heatpump from 2008, EKHB007AGT1 and ERYQ006ACV3. PCB DB-G57-101. I have installed and maintained my heatpump for many years. I have also done some modification for relay to the water pump, after fault code AA, relay KPR was defect and I mounted a new relay. Then mounted a new relay unit outside the PCB to omit new fault for KPR relay in the future. The X10 connection have 8 pins.
I'm new to platformio and started yesterday. I have an ESP32 devkit v1, from e-bay. Testet with some hello wold and test internal blink led. But with loading your code to ESP32 I got an error message:

> Executing task: C:\.platformio\penv\Scripts\platformio.exe run --target upload <

Processing esp32 (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (3.1.1) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10005.210308 (1.0.5)
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - tool-mkspiffs 2.230.0 (2.30)
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <PubSubClient> 2.8.0
|-- <ArduinoOTA> 1.0
|   |-- <Update> 1.0
|   |-- <WiFi> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|-- <WiFi> 1.0
|-- <EEPROM> 1.0.3
Building in release mode
Retrieving maximum program size .pio\build\esp32\firmware.elf
Checking size .pio\build\esp32\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  14.5% (used 47364 bytes from 327680 bytes)
Flash: [======    ]  59.0% (used 772677 bytes from 1310720 bytes)
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa      
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM6
Uploading .pio\build\esp32\firmware.bin
esptool.py v3.0
Serial port COM6
Connecting........_
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 3c:61:05:2a:6d:74
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 215200
Changed.
Configuring flash size...

A fatal error occurred: Timed out waiting for packet header
*** [upload] Error 2
==================================================================== [FAILED] Took 9.17 seconds ====================================================================

Environment    Status    Duration
-------------  --------  ------------
esp32          FAILED    00:00:09.175
=============================================================== 1 failed, 0 succeeded in 00:00:09.175 ===============================================================
The terminal process "C:\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

raomin, 2021-03-24:
Hello @Oystein-github and Welcome!
`A fatal error occurred: Timed out waiting for packet header` this is your error.
1. Make sure that nothing is connected on the board when you upload.
2. Try to lower your upload speed `upload_speed = 115200` in your platformio.ini
3. Press the reset button and release just before the upload starts.

Oystein-github, 2021-03-25:
Thank you, it works with # 2, change the upload_speed to 115200. Then I can go forward to test it.

raomin, 2021-03-25:
Glad to hear this :)

closed 2021-03-25 by raomin