raomin/ESPAltherma #42: JSON and numeric values as string

rufik, 2021-03-29:
Right now all values are sent to mqtt broker as string (`espaltherma/ATTR` topic). It's not a problem with HA but it becomes a little bit problematic when storing in some DB like InfluxDB for example. Possible I can transform/cast values to numeric but it requires additional work.
I'd like to **propose enhancement** - send numeric values (these with numeric converter function) as-is, not as string.

raomin, 2021-03-31:
Good point. I'll add it.

KalaNagHTD, 2021-03-31:
I solved the problem with javascript in ioBroker. As rufik said: "additional work" ;-)

[https://forum.iobroker.net/topic/41289/gel%C3%B6st-json-tabelle-in-datenobjekte-aufl%C3%B6sen-javascript/7?_=1617211024919](url)

raomin, 2021-04-24:
Hey great news! No more need for "additional work", ESPAltherma now knows how to recognize a number 😊
I just committed it, it now returns number without quotes in json,
`..."DHW setpoint":55,"LW setpoint (main)":35.5,"Water flow switch":"ON"...`

closed 2021-04-24 by raomin