raomin/ESPAltherma #45: Heat pump selection

vojtechalbl, 2021-04-12:
Is it possible to use ESPAltherma with ROTEX HPSU Compact 508 manufactured in 2014?
Thanks in advance.

KalaNagHTD, 2021-04-12:
Sure! 😀

raomin, 2021-04-12:
Hi @vojtechalbl, welcome!
I believe `ALTHERMA(HPSU6_ULTRA).h` should be a good file definition for your heat pump.
Test and let us know if you get what you need!

vojtechalbl, 2021-04-18:
So, I am using M5StickC and ALTHERMA(HPSU6_ULTRA).h. My board has 8pin X10A port. I have used pins according to your instructions.

Unfortunately, I am not receiving any data:-(.

Any suggestions?

Here is my mqtt output in homeassistant:

Message 787 received on espaltherma/ATTR at 4:53 PM:
{
    "M5VIN": "0V",
    "M5AmpIn": "0mA",
    "M5BatV": "3.92V",
    "M5BatCur": "86.5mA",
    "M5BatPwr": "0mW"
}
QoS: 0 - Retain: false
Message 786 received on espaltherma/log at 4:53 PM:
0x03 0x40 0x62 0x5a 
QoS: 0 - Retain: false
Message 785 received on espaltherma/log at 4:53 PM:
ERR: Time out on register 0x62! got 4/98 bytes
QoS: 0 - Retain: false
Message 784 received on espaltherma/log at 4:53 PM:
Querying register 0x62... 
QoS: 0 - Retain: false
Message 783 received on espaltherma/log at 4:53 PM:

QoS: 0 - Retain: false
Message 782 received on espaltherma/log at 4:53 PM:
Retrying...
QoS: 0 - Retain: false
Message 781 received on espaltherma/log at 4:53 PM:
0x03 0x40 0x62 0x5a 
QoS: 0 - Retain: false
Message 780 received on espaltherma/log at 4:53 PM:
ERR: Time out on register 0x62! got 4/98 bytes
QoS: 0 - Retain: false
Message 779 received on espaltherma/log at 4:53 PM:
Querying register 0x62... 
QoS: 0 - Retain: false
Message 778 received on espaltherma/log at 4:53 PM:

QoS: 0 - Retain: false
Message 777 received on espaltherma/log at 4:53 PM:
Retrying...
QoS: 0 - Retain: false
Message 776 received on espaltherma/log at 4:53 PM:
0x03 0x40 0x62 0x5a 
QoS: 0 - Retain: false
Message 775 received on espaltherma/log at 4:53 PM:
ERR: Time out on register 0x62! got 4/98 bytes

raomin, 2021-04-19:
Hi @vojtechalbl,
Seems that your m5 has no input power and runs on battery. Make sure that the ground is properly connected to the ground of the X10A port and that if needed the M5stickc is powered with external USB charger.

vojtechalbl, 2021-04-19:
Thank you very much for your advise. 

You were right. I forgot to connect M5stickc to the GND. It is powered with external USB charger. 

Now the output is full of data.

Message 247 received on espaltherma/ATTR at 5:13 PM:
{
    "Operation Mode": "Fan Only",
    "Thermostat ON/OFF": "OFF",
    "Defrost Operation": "OFF",
    "Outdoor air temp.": "10",
    "INV primary current (A)": "0.5",
    "INV secondary current (A)": "0",
    "INV frequency (rps)": "0",
    "Freeze Protection": "OFF",
    "DHW setpoint": "49",
    "LW setpoint (main)": "15",
    "Outlet Water Heat Exch. Temp. (R1T)": "27.9",
    "Outlet Water BUH Temp. (R2T)": "25",
    "Refrig. Temp. liquid side (R3T)": "27.5",
    "Inlet water temp.(R4T)": "24.2",
    "DHW tank temp. (R5T)": "47.1",
    "RT Temp.": "0",
    "M5VIN": "0V",
    "M5AmpIn": "0mA",
    "M5BatV": "4.1V",
    "M5BatCur": "0mA",
    "M5BatPwr": "0mW"
}

thanks again

raomin, 2021-04-24:
Hey @vojtechalbl, glad you got it working!
Remember to post what you do with this on the community post!

Happy ESPAltherma :)

_If this project has value for you, please consider [supporting it](https://www.buymeacoffee.com/raomin). I don't do this for money but it feels good to get some support! Thanks :)_

closed 2021-04-24 by raomin