raomin/ESPAltherma #47: After connectin m5stickc, register reading fails, keeps sending error message "check connection"

ocsele, 2021-05-26:
Dear Raomin,

Please help me with this issue if you can. After turning off the heatpump (EHVX16S26CB9W, tried with ALTHERMA(LT_CA_CB_11-16KW.h did some translations but the file looks okay to me compared to the original), I connected the ESP with dupont cables (made by me from UTP cable), correct pin layout checked a lot of times, they connect to the correct I/Os:
[image]

The stick connects to my network, sends correctly MQTT messages, but only reports values regarding the stick itself, nothing about the heatpump itself, and the display continuously throws "timeout reading register, check connection":
[image]

Can you please give me any advise on troubleshooting the issue? Already tried with another set of dupont cables (factory made, not me), to exclude wiring issues, but the result is the same unfortunately.. :( 

Thanks in advance!

ocsele, 2021-05-26:
WOW! I'm not sure what I did, pulled off, then replugged the cables (even tested them with a multimeter, they were fine), but now it works!!!!!! Now I just have to figure out what are the useful registers for me :) Beer(s) will be sent to you Raomin for your wonderful job done here, and I'll document the setup in the other thread to show off :)
[image]

Thanks a lot Raomin!!!!!

closed 2021-05-26 by ocsele

raomin, 2021-05-26:
Great! Glad it works. I would have asked you to recheck your cablings anyway ;)
Thanks for your support! Enjoy!