raomin/ESPAltherma #48: Problems with Receiving Data

Tirocinante01, 2021-06-16:
Hi everyone
I got a M5StickC on my ROTEX heating system: RHYHBH05AAV3
the Stick seems to be working but for some reason I receive no data back. The cables are attached correctly and I'm pretty confident about it but the problem persists
does anyone know what might be the problem?

Tirocinante01, 2021-06-16:
[image]
[image]
[image]
[image]

here are some images of the situation for better understanding

raomin, 2021-06-16:
Check the cables, it's always the cables. 😉
No. Really it can only be the cables/connections. Change them if needed.

raomin, 2021-06-23:
So @Tirocinante01, did you manage to find and fix your issue?

raomin, 2021-06-26:
I'll close this one, feel free to reopen it if your problem is not resolved.

closed 2021-06-26 by raomin