raomin/ESPAltherma #49: No entities in home assistant

Jogiarea, 2021-07-03:
Hello everybody,

I joined the great project and got down to work to implement it.
My setup:
- Rotex HPSU Compact 2014
- NodeMCU esp32
- Home server with home assistant supervisor in Docker

I successfully flashed the esp32 and also have both sensors (1x switch / 1x sensor) in the home assistant 

[image]

If I now search for the entities under developer, I get nothing displayed 

[image]

But when I call up the MQTT Info under Devices in the ESPAltherma, I get all data displayed as received 
[image]

I somehow get stuck here 

I Also tried to insert some sensors manually, but without any succes
[image]
[image]

here are my settings of the firmware :

`; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[platformio]
default_envs = esp32

[env:esp32]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
monitor_speed = 115200
upload_speed = 115200
; upload_port = COM5
; Uncomment this line to allow for remote upgrade. If name resolution does not work for you, replace with the IP of ESPAltherma
; upload_port = ESPAltherma.local

lib_deps = 
	PubSubClient

[env:m5stickc]
platform = espressif32
board = m5stick-c
framework = arduino
monitor_speed = 115200
upload_speed = 215200
; Uncomment this line to allow for remote upgrade. If name resolution does not work for you, replace with the IP of ESPAltherma
; upload_port = ESPAltherma.local

lib_deps = 
	M5StickC
	PubSubClient

[env:m5stickcplus]
platform = espressif32
board = m5stick-c
framework = arduino
[... 274 lines cut ...]
//{0x63,4,215,1,-1,"I/U EEPROM (6. 7. Ziffer)"},
//{0x63,5,215,1,-1,"I/U EEPROM (8. 9. Ziffer)"},
//{0x63,6,215,1,-1,"I/U EEPROM (11. Ziffer)"},
//{0x63,7,214,1,-1,"I/U EEPROM (12. Ziffer)(rev.)"},
//{0x64,0,307,1,-1,"Daten Aktivieren/Deaktivieren"},
//{0x64,1,152,1,-1,"Adresse des Innengeräts"},
//{0x64,2,316,1,-1,"Hybridbetrieb Modus"},
//{0x64,2,303,1,-1,"Boiler Betrieb Anforderung"},
//{0x64,2,302,1,-1,"Boiler DHW Anforderung"},
//{0x64,2,301,1,-1,"Bypassventil Ausgang"},
//{0x64,2,300,1,-1,"Hybrid Trinkwassererwärmung"},
//{0x64,3,105,2,-1,"BE_COP"},
//{0x64,5,105,2,1,"Hybrid Heizung Solltemp."},
//{0x64,7,105,2,1,"Boiler Heizung Solltemp."},
//{0x64,9,302,1,-1,"Add pump"},
//{0x64,9,301,1,-1,"Main pump"},
{0x64,10,105,2,1,"Mischwassertemperatur"},
//{0x64,12,105,2,1,"2nd Domestic hot water temperature"},
//{0x64,14,152,1,1,"Target delta T heating"},
//{0x64,15,152,1,1,"Target delta T cooling"}
};`

I probably just forgot a little thing, but I can't figure it out

Thank you in advance for your help

raomin, 2021-07-07:
Hello @Jogiarea .
I understand that the 2 entities were properly auto discovered by home assistant.
As the mqtt message is properly coming to Home Assistant, you should be able to get the attributes of the AlthermaSensor in the developer tool>states.  As it is not coming up there might be a pbm with the interpretation of the message, but I never heard of such issue.
Definitely, you should have a look in the logs, there should be an information there, if needed, raise the log level of the mqtt module.

Jogiarea, 2021-07-07:
Hi Roamin,

Thanks for the fast answer, ok iv´e looked at tho logs
[image]

Incorrect JSON:
I think that what goes wrong when creating the JSON message but can't find anything
pbm ?? I do not understand, sorry 

it may be that the JSON cannot cope with ä ö ü, because I use the German setting

I have to admit that I'm not that deep into programming

raomin, 2021-07-07:
hmm I would more go for an extra " or : 
Can you paste the json message here?

Jogiarea, 2021-07-07:
how can I see this? is that possible in Homeassistant

raomin, 2021-07-07:
Just copy paste here the message from the logs

Jogiarea, 2021-07-07:
Here the Logs

    Erroneous JSON: [{"Aktuelle Betriebsart":"Fan Only","Thermostat EIN/AUS":"OFF","Wiederanlaufverzögerung":"OFF","Abtauung":"OFF","Schallreduzierter Betrieb":"OFF","Betrieb / Störung":"Normal","Störcode":" 0","Außenlufttemp.":27,"O/U Wärmetauscher Temp.":0,"Ablassleitungstemp.":22.5,"Ansaugleitungstemperatur":0,"O/U Wärmetauscher Mitteltemp.":22,"Druck":14.5,"Druck(T)":"Conv 405 not avail.","Stromaufnahme INV-Verdichter (A)":0,"Spannung (V)":"-28160","Drehzahl Verdichter (U/Sek)":0,"Ventilator 1 (10 rpm)":0,"Thermostat EIN/AUS":"OFF","Betrieb / Störung":,"Sollwert Brauchwasser":36,"Sollwert Heizungsvorlauf":15,"Wasserflussschalter":"OFF","Wärmeschutz (Q1L) BUH":"OFF","Wärmeschutz BSH":"ON","Vorteil kWh Raten Stromversorgung":"OFF","Bivalenter Betrieb":"OFF","BSH":"OFF","Heizstab Stufe 1":"OFF","Heizstab Stufe 2":"OFF","BPH":"OFF","Wasserpumpenbetrieb":"OFF","Auslass Wasser Wärmetauscher Temp. (R1T)":31.6,"Auslass Wasser BUH Temp. (R2T)":30.4,"R3T-Kältemittel Flüssigkeitstemperatur":29.4,"R4T-Wasser Rücklauftemp. vor dem Plattenwärmetauscher":28.5,"R5T-Brauchwassertemperatur im Speicher":31.2,"RT Temp.":0,"Außen Umgebungs- oder Ext-Sensor":22.6}]

Jogiarea, 2021-07-07:
And in jsonlint it looks good

[image]

raomin, 2021-07-07:
[image]

```
	"Betrieb / Störung": ,
```
There seems to be an issue with this one which returns nothing.... remove it from your ESPAltherma config file and you should be ok.

Jogiarea, 2021-07-07:
ok many thanks

i just saw it too 

I ll  try it and give a feedpack

Jogiarea, 2021-07-07:
many, many thanks, the beer will come
it works fine 

[image]

raomin, 2021-07-07:
Happy dashboarding :)
Thanks!

closed 2021-07-07 by raomin