raomin/ESPAltherma #51: ESP working on PC USB Power but not on USB charger

andreapier, 2021-09-21:
Hello, I am facing a weird issue.
My ESP32 running ESPAltherma is working fine when powered through PC USB but it times out when powered through an USB charger.  
I tried many of them, from different manufacturers (huawei, samsung, lg). ESP does not work with any of them.
**I understand this has nothing to do with ESPAltherma**, but I am curious if anyone else is facing the same issue.
I found [this forum thread](https://esp32.com/viewtopic.php?t=17718) that is more or less relevant. Unfortunately I did not get any answer yet.
By "times out" I mean that I get timeout errors in the log, like if my ESP is not able to read from the serial port.
**Pins connections are fine** as I tried many times to plug my ESP32 in my PC and it reads values correctly.
Does someone have any idea about what to try to at least get some more information about this problem?
Thanks!

raomin, 2021-09-21:
Hi @andreapier! 
You might want to try changing the dupont wire. Especially the on connecting to the ground as it looks like a ground issue (that could be temporarily solved with the ground of your PC being the same -or close- to your HP's). I've had dupont wires that looked fully intact on the outside but were actually cut.
Did you try to connect to power your ESPAltherma directly from the  X10A?

andreapier, 2021-09-23:
Thanks @raomin!
It was indeed a ground issue! Solved using good quality wires.

closed 2021-09-23 by andreapier