raomin/ESPAltherma #56: Pin assignment for the 8-pin X10 connector

abutscher, 2021-10-31:
I have a Rotex HPSU compact where the X10 connector is configured with 8 pins. The README mentions this configuration but it's not completely clear to me:

1. The corresponding picture shows only the assignment of RX, TX and GND. But the pin which shall provide the voltage supply is not mentioned. Does someone know the full pin assignment of the 8 pin X10 connector? (Are there maybe even specs available?)
2. Is above question technically related to the fact that the supply from the connector is not enough for the ESP32. In other words: Is it correct to say that it just does not matter where the voltage supply pin is because the ESP32 has to be supplied externally with USB-C anyway?

Is anybody even around who attached the ESP32 to a ROTEX HPSU Compact model? It's one of those where the PCB with the X10 is behind the front one. It's not trivial to get access to it. Would be really interesting to hear about how others managed to do it.

Anyway big kudos to the community and especially raomin on all the efforts and achievements.

raomin, 2021-11-07:
Hello @abutscher and welcome!
Sorry for the late reply (and thanks @m-reuter for the first reply!).

I don't have a Rotex, but I understood that the power of the Rotex port was indeed not enough to power the ESP32.
As you suggest, you would have to use an external USB power supply to power your ESP32 and please remember to still connect the GND of the X10A port with the one of your ESP32.

As to how to access the main PCB, it should be behind a metal plate behind the black plastic upper part. But, again, I don't own one, so better check with a Rotex owner like @KalaNagHTD.

Have fun!

closed 2021-11-07 by raomin

m-reuter, 2021-11-08:
The Rotex HPSU compact ultra can power the M5 stickCplus.