raomin/ESPAltherma #57: Problem with visibility of variables in HA

TomaszCwik, 2021-11-05:
Hello,
I have problem with integration ESP with HA
in this moment when i giving ip address of HA in MQTT explorer  i see 3 record LWT,LOG,ATTR 
[image]
but in HA i see only 
[image]
i use ESP32 with external Power suplly 
in setup.h i uncomment #include "def/DEFAULT.h"

TomaszCwik, 2021-11-07:
in MQTT info in HA i see:
[image]
in ATTR i always have "}"
in setup.h:
[image]
[image]
in default.h
[image]
IN HA i have information about wrong JSON

raomin, 2021-11-07:
Can you show us what pops under espaltherma/LOG or show what is coming up on the Serial monitor?

Are you sure the grounds of your Esp32 is linked with the ground of the heat pump?

TomaszCwik, 2021-11-08:
That's it ...
i didn't connect GND with "-"
i`m sorry : P

raomin, 2021-11-09:
Good to hear it's now working for you.
I'll emphasise on this in the readme.

closed 2021-11-09 by raomin