raomin/ESPAltherma #59: Reading from some specific registers times out. But some work

abutscher, 2021-11-15:
Hi,

Another issue on reading timeouts. I know, it's already been handled in some before and the "classic" fix is checking/replacing the GND cable. But I think that my issue might be different (although it's rather similar to https://github.com/raomin/ESPAltherma/issues/44).

First, I am using a Rotex HPSU compact (which is more or less the same as in issue #44, I think) and I am using the Altherma HPSU6 Ultra definition header file.

In fact, I am querying registers 0x00, 0x10, 0x20, 0x21, 0x30, 0x60, 0x61, 0x62 and 0x64. In each query cycle, reading registers 0x00, 0x62 and 0x64 fails. Always the same pattern. For each of those reads, I get this log
`Querying register 0x62... 
ERR: Time out on register 0x62! got 2/10 bytes

0x15 0xea 
Retrying...`

This rises some questions:

1. Is this really a "GND cable issue"? Obviously, 2 bytes can be read...
2. It always reads the same 2 bytes: 0x15 and 0xea. Does anybody know its meaning? According to Raomin's documentation on the communication protocol, the response should always begin with 0x40 (protocol identifier) followed by the register ID (which obviously should not be 0xea in that case).
3. Could that be some error handling from the heat pump? Considering the fact that my model (Rotex compact HPSU) does not fully comply to the selected definition header file (Altherma HPSU6 Ultra), could the response mean something like "error, register not present/not implemented/..."?

Thanks in advance for any help.
I really start to like digging into all this :-).

abutscher, 2021-11-15:
Just another thought on why I think it's not an issue with the GND cable.
I think, it's not the actual read command on the serial line that times out (if it was, I guess it would more likely an issue with GND). I suspect, that the heat pump is just not sending more than those 2 bytes because it is an error case. But the ESPAltherma SW expects to read at least 10 bytes (or is it 12?) and therefore reports an timeout.

raomin, 2021-11-16:
Hi @abutscher,
That's weird, never saw this before. Can you confirm you have this issue with a code untouched?

`0x15 0xea` tells that the HP did not understand the query. But here there is little to understand, as ESPAltherma basically tells the HP "give me register 0x00"...

You could try to increase the timeout value [here](https://github.com/raomin/ESPAltherma/blob/c4d6691a89b9aec34e2d97c918bb1360b808a104/include/comm.h#L5)  but I doubt it would do something as the first register 0x00 also fails...

If it's always always these 3 registers, then it's unlikely it's a GND issue, but still I would try to power the ESP directly from the HP and put an additional GND cable somewhere...

abutscher, 2021-11-16:
Hi,

So, I did a test with the code from origin/main. I also increased the timeout to 10000 msec. But the result is the same...
OK!

Adding registry 0x 0 to be queried.

ESPAltherma started!
Querying register 0x00... 
Time out! Check connection

Retrying...

Querying register 0x00... 
ERR: Time out on register 0x00! got 2/10 bytes

0x15 0xea 
Retrying...

Querying register 0x00... 
Time out! Check connection

Retrying...

Querying register 0x00... 
ERR: Time out on register 0x00! got 2/10 bytes

0x15 0xea 
Done. Waiting 30 sec...

abutscher, 2021-11-16:
@raomin the last sentence is interesting but I don't quite understand.
Powering the ESP32 from the HP basically means to use PIN 1 from X10A connector and connect it with the 5V Input of the M5 Stick C Plus?
How should I put an additional GND cable somewhere? Sorry, don't understand that :-). Where should I connect cables?

Anyway, really appreciating your help.

raomin, 2021-11-16:
> Powering the ESP32 from the HP basically means to use PIN 1 from X10A connector and connect it with the 5V Input of the M5 Stick C Plus? No external USB-C power then?
Correct. Just like the schema on the README:

[image]

GND of esp32 is connected to PIN5 (the GND of the HP).

You can also find the ground of the HP in many places of the HP, like.. the casing for instance...

What is critical is that the ground of the esp32 is common with the one of the HP. If you power the esp32 with a usb charger and that you do not connect the 2 GND together you will have unexpected results.

abutscher, 2021-11-16:
I connected Pin 1 of X10 with 5V according to above picture. USB-C Power was removed. This did not help regarding my actual issue. Additionally, it drained the battery power. I saw "M5VIN" values varying from 4,3 Volt down to 2,5 Volt.
As I do have a Rotex HP, I guess this complies to what other users reported. That the power provided by X10 is not enough...

abutscher, 2021-11-17:
Hi Raomin,

please have a look at https://github.com/raomin/ESPAltherma/pull/63. I think, I found the reason why register 0 can not be queried. I was a bit curious and sniffed the TX path with a logic analyzer. Only two bytes were visible on the line...
With the fix from #63 I can query register 0. But I am wondering, if anybody could ever query register 0 before...?

Still there is the issue with accessing registers 0x62 and 0x64. They are still not accessible...

abutscher, 2021-11-17:
I double checked the packets that are sent to query register 0x62 and 0x64 with the logic analyzer.
For register 0x62, I can see this byte sequence sent to the heat pump `0x03 0x40 0x62 0x5a 0x00`. For register 0x64 it's `0x03 0x40 0x62 0x5a 0x00`. Both seem to be ok, I think.
The heat pump replies for all queries of above register **always** with `0x15 0xea`. Compared to the issue with register 0x00, this is different. The heat pump replied only for two of the four retry queries with `0x15 0xea`.

abutscher, 2021-11-17:
One question on CRC calculation. The corresponding function is declared to return char, so the CRC can be at most 1 byte. Is this intended? Nevertheless, the query packet is declared to be 5 bytes of size. The calculated CRC is put at the second to last index. So the last byte is always 0.
Considering a register ID 0xff. Then, the CRC would be 2 bytes: 0x3+0x40+0xff = ~0x142.

raomin, 2021-11-17:
> I double checked the packets that are sent to query register 0x62 and 0x64 with the logic analyzer. For register 0x62, I can see this byte sequence sent to the heat pump `0x03 0x40 0x62 0x5a 0x00`. [...]

With the current code, `0x03 0x40 0x62 0x5a 0x00` is properly sent as `0x03 0x40 0x62 0x5a` as `write(prep)` is internally  `write(prep,strlen(prep))`... With your PR #63 this becomes `0x03 0x40 0x62 0x5a 0x00`
I'm not sure the controller of the HP actually cares as it reads 0x03 bytes after 0x03.

raomin, 2021-11-17:
> One question on CRC calculation. The corresponding function is declared to return char, so the CRC can be at most 1 byte. Is this intended? Nevertheless, the query packet is declared to be 5 bytes [...]

Yes, the CRC is always one byte. It's a SumAndInvert CRC.
I sized `prep` as 5 to have null termination and send it to `write()` which, by default, sends strlen() bytes, hence 4. Only is some cases it will fail, but your PR will solve this! Thanks.

raomin, 2021-11-17:
So with #63 merged, I think we can close this one! Thanks @abutscher.

abutscher, 2021-11-17:
Nothing to thank for ;-)

abutscher, 2021-11-17:
Unfortunately my issue is only partly solved. I still cannot read registers 0x62 and 0x63
https://github.com/raomin/ESPAltherma/issues/59#issuecomment-971774518

abutscher, 2021-11-17:
Just wondering. In https://github.com/raomin/ESPAltherma/issues/59#issuecomment-970097744 you mentioned that the sequence 0x15 0xea basically means that the heat pump does not understand the query. Can that include the case that the queried register is not implemented on the heat pump?

raomin, 2021-11-17:
yes, Best is to ask Rotex users @m-reuter @giuluc @KalaNagHTD ...

DieterTHeck, 2022-07-16:
@abutscher did you ever resolve the issues with 0x62 and 0x64?

xalex75, 2022-07-21:
Hi. I'm having exactly the same issue on 0x62 and 0x64.
ESP32-CAM connected to Daikin Altherma HPSU Compact and currently using HPSU6 Ultra definition.

abutscher, 2022-07-23:
Hi,
@DieterTHeck @xalex75 Unfortunately, I could not resolve that issue. I even tried to probe all registers from 0x00 to 0xFF to see if there is anything to read from. To me, it somehow seems that the Compact models have some limited set of registers. It looks like as if 0x62 and 0x64 are not implemented. But still strange because at least some of the features/values that can be retrieved via 0x62 and 0x64 are really existing in the heat pump device. E.g. water pressure or the flow sensor are technically existing. Those values can be seen at the front panel of the pump.

abutscher, 2022-07-23:
Maybe this issue https://github.com/raomin/ESPAltherma/issues/74 could help. I have not read it in depth but it least it sounds like as if there may be an additional protocol available. Finally, it seems that probing the heat pump with the DChecker (also mentinoned in #74) might help to find "hidden" registers. I plan to give that a try when I find the time for it.

xalex75, 2022-07-25:
@abutscher thank you.
the values inside those registers are useful and need for COP calculation, but I think we already have the most important values from other registries.
So reading from those addresses would be very good, but I think I can live without them.

@raomin already did something incredibile, I thought I would never have been to have heatpump values in HA, and now I have the most important of them. :-)

DieterTHeck, 2022-07-25:
Hi all,

@abutscher I think it is not very realistic that for most registers one protocol is used, and for x62/x64 another.

Nevertheless, I just tried the following:
A) cloned comm.h into comm2.h with changes like in #74 and included that one in main.cpp instead of comm.h
Maybe somebody wants to check that I have not forgotten anything.
```
  //preparing command:
  char prep[] = {0x02, regID, 0x00}; //{0x03, 0x40, regID, 0x00};
  prep[2] = getCRC(prep, 2); //prep[3] = getCRC(prep, 3);

  //Sending command to serial
  MySerial.flush(); //Prevent possible pending info on the read
  MySerial.write(prep, 3); //  MySerial.write(prep, 4);
  ulong start = millis();
```
B) created a test.h which queries these registers:
```
{0x10,0,217,1,-1,"Operation Mode"},
{0x62,9,105,2,-1,"Durchflussmenge (l/min)"},
{0x64,2,302,1,-1,"Boiler DHW Anforderung"},
```
C) compiled & uploaded

RESULT 
```
ERR: Time out on register 0x10! got 2/10 bytes
0x15 0xea 

ERR: Time out on register 0x62! got 2/10 bytes
0x15 0xea 

ERR: Time out on register 0x64! got 2/10 bytes
0x15 0xea 
```

DieterTHeck, 2022-07-25:
It could still be another, third protocol to query them.
I am afreid I cannot use this d-checker tool here, but I can run a few tests like the above of somebody tells me.

raomin, 2022-07-25:
Really unlikely to be another protocol, just to access some values... Might be that this information is simply not available.
Do you know if you can see some values from 0x62 and 0x64 in the user interface of the Rotex?

DieterTHeck, 2022-07-26:
Flow in l/h is for instance shown in the UI. And water pressure as well. I guess they are meant to sit here.
```
//{0x62,9,105,2,-1,"Durchflussmenge (l/min)"},
//{0x62,11,105,1,2,"Wasserdruck"},
```

I just checked that x62/ x64 also do not work when I set my Rotex from summer to normal operation. 
Unfortunately that is the case. :-(

I also wondered what these entries mean ... 
`//{0x61,0,307,1,-1,"Data Enable/Disable"},`
but they are also not conclusive. For x60 they are OFF, x61 ON, but I get values from both registers.
So apparently it is also not about sending a ON value there to enable registers.

I think I give up on x62/x64 here... 
Still much more info than one month ago :-)

xalex75, 2022-07-27:
Thanks both very much for your effort.
We will not be able to calculate COP, but we already have enough useful information, more that what I ever thought ;-)

d3zit, 2022-11-17:
https://lamiacasaelettrica.com/daikin-hpsu-compact-hack-seconda-parte/ 
Don't know if this can help, but they managed to extract the "missing" data from the CAN-Bus from the HPSU Compact

This is the github: https://github.com/zanac/pyHPSU

robertoP-git, 2022-12-28:
> Thanks both very much for your effort. We will not be able to calculate COP, but we already have enough useful information, more that what I ever thought ;-)

Hi xalex75! 
i have your same problem.
Have you found a solution? thank you

d3zit, 2022-12-29:
Same issue here. I have the usb serial adapter and I am willing to test it with the daikin apps, but would appreciate some help.

xalex75, 2023-01-01:
> https://lamiacasaelettrica.com/daikin-hpsu-compact-hack-seconda-parte/ Don't know if this can help, but they managed to extract the "missing" data from the CAN-Bus of the HPSU Compact
> 
> This is the github: https://github.com/zanac/pyHPSU

This would be fantastic!
But I'm not able to do that myself... maybe somebody here can help?

xalex75, 2023-01-01:
> > Thanks both very much for your effort. We will not be able to calculate COP, but we already have enough useful information, more that what I ever thought ;-)
> 
> Hi xalex75! i have your same problem. Have you found a solution? thank you

Hi.
No solution till now, and the issue has been marked as closed, so I don't know if we will ever have a solution :-(

raomin, 2023-01-01:
No problem to reopen it.

robertoP-git, 2023-01-09:
> This would be fantastic! But I'm not able to do that myself... maybe somebody here can help?

I'm trying to understand this project... but it's not simple.
It would be nice to fix the problem for everyone

seb-koch, 2023-10-22:
I got the same issue with the 0x62 registers (especially Flow sensor) with my Rotex HSPU compact. I am using a m5stickC. I hope one of the experts figure out a solution. Without the Flow info there is no COP calculation afaik.

raomin, 2026-10-23:
It's been a while, but I'd like to close the old ones with the 2.0 release. ESPAltherma 2 surveys the registries first and only polls the ones that answer, and it detects the S/I protocol by itself, which covers the HPSU compact case here. If it still times out on v2, open a new issue with the Diagnostics survey. Thanks for the report back then!

closed 2026-10-23 by raomin