raomin/ESPAltherma #64: Logic level converter?

Gollam, 2021-11-19:
We have an Altherma 3, and I'd like to include a logic level converter in my setup, because it's very cheap, so why not. I am just wondering, is this the correct setup if one would like to use it?

[image]

raomin, 2021-11-19:
Hi @Gollam, I think no one ever implemented it. Now, if you really want to do it, yes that should be the way to go.

Gollam, 2021-11-19:
@raomin Thanks!

closed 2021-11-19 by Gollam

Gollam, 2021-11-19:
@raomin BTW, how can it be determined what levels a given interface uses? I haven't found any documentation about X10A and I'm just curious.

raomin, 2021-11-19:
Usually, it uses the VCC you can find in the port ;)
So here +5V. If you want to be sure, you need a oscilloscope.