raomin/ESPAltherma #66: Wrong COP calculation

stomko11, 2021-11-24:
I am using following formula to calculate COP:

>    espaltherma_cop:
        friendly_name: "Altherma COP"
        unit_of_measurement: "COP"
        value_template: "{% if is_state_attr('sensor.althermasensors','Operation Mode', 'Heating') and is_state_attr('sensor.althermasensors','Freeze Protection', 'OFF')  %} 
        {{ 
          ((state_attr('sensor.althermasensors','Flow sensor (l/min)')| float * 0.06 * 1.16 * (state_attr('sensor.althermasensors','Leaving water temp. before BUH (R1T)') | float - state_attr('sensor.althermasensors','Inlet water temp.(R4T)')|float) )
            /
          (state_attr('sensor.althermasensors','INV primary current (A)') | float * 233 / 1000))
          |round(2)
        }}
        {% else %} 0 {%endif%}"

My heatpump is not returning voltage, so I have replaced voltage with static value of  "233". However I am getting calulations that are way off, with COP around 10 and more. I can't figure out what am I doing wrong.

[image]

Here are also values for my primary current:

[image]

raomin, 2021-11-24:
Your graph seems odd: It seems you have sometimes values every min or so and sometimes nothing for half an hour... Is everything ok with your data collection?

Besides, your primary current is low... 4A max that's not even 1kw electric, something is weird here.
Can you graph the other values in the COP definition?

raomin, 2021-11-24:
Looking at your picture, it seems to be a 3-phase installation, right? This might be then that the `INV primary current (A)` is the one of one phase and that it should be multiply by 3...? 
Is your compressor also 3-phase?

stomko11, 2021-11-25:
I do have meter for the whole house, so I can't isolate just HP. But here is graph for last 12 hours, blue is the HP, other 3 are current values (A) on each phase.

[image]

And also watt usage (W) on each phase + total

[image]

Here are other values that I use in calculation - past 12 hours.

[image]

As for the data collection, they are represented OK in HA, just have issues sometimes in influxDB/grafana where consecutive same values are not displayed correctly.

Gollam, 2021-11-29:
Sorry for hijacking your thread, but my issue is kind of similar. This is what I receive as voltage: "Voltage (N-phase) (V)": 25, I guess this is also incorrect, but something is being read as it is not 0, it is 25. So is this case considered as a bug in decoding the voltage value?

stomko11, 2021-11-29:
What value do you query for that?

Gollam, 2021-11-29:
{0x21,4,101,2,-1,"Voltage (N-phase) (V)"}, I found this in 90% of the header files.

stomko11, 2021-11-29:
OK, for me it returns always 0. Even my Altherma does not show consumed energy, just produced, so I suspect it somehow does not read that value.
In some threads I found 0x21,2,101, however that returns usually just 1/10th of secondary current.

Gollam, 2021-11-29:
FTR, ours is a EHVH08S18EJ6V, and this voltage value varies between 17 and 50 as I saw so far, so that's quite a huge range :D

raomin, 2022-11-22:
I'll close this old one, COP calculation has been clarified in other 'issues' and discussion.

closed 2022-11-22 by raomin