raomin/ESPAltherma #67: Get data for external thermostat for main zone

stomko11, 2021-11-24:
I am using external thermostat on my heatpump. None of the 2 values below reflect that state:
{0x10,1,307,1,-1,"Thermostat ON/OFF"},
{0x60,2,303,1,-1,"Thermostat ON/OFF"},

I checked wiki, there is a comment for one of them that this is basically ANY thermostat (DHW, LW, main zone, additional zone). Is there a way to capture only main thermostat? I would like to know when there is a request for heating from my external regulation.

Additionally, is there a way to tell when compressor is actually running? (that is displayed on heat pump as symbol for DHW or LW with circling arrows around it).

raomin, 2021-11-24:
I don't think you can isolate the thermostat information... Personally I replaced the thermostat with several indoor esp with temp sensors and an additional relay on the HP as _external thermostat_. So I can control when to heat on HomeAssistant.

For the operation mode:

`Operation mode` will tell you whether it's doing nothing (`Fan only`) or `Heating`: that's for the compressor
`Indoor Operation mode` or `I/U Operation mode` will tell you for what the compressor is running: `DHW` (default) or `Heating` or both `Heating + DHW`

stomko11, 2021-11-25:
Even the "Operation mode" - it is almost always in "heating", yet it is not heating constantly.
"Indoor operation mode" does reflect when there is DHW heating, then it switches to Heating by default, doesn't mean it actually heats water. Most of the time, it is only water pump turning on, cycling heated water, turning off, until water temperature is insufficient and then it starts heating.

As for external thermostat, I basically do have the same, except for logic not being tied to Home Assistant, as it is standalone system (Danfoss Link). Although it is bit OT, but can I ask you how do you manage your heating? I read quite a few articles, various thermostats (virtual, PWM, etc.), but majority is suitable for one thermostat / one actor.
What I need is one thermostat, possibly few actors (as one room with one thermostat can have several valves that need to be opened/closed), which I guess could be managed by group of valves and simply let Thermostat operate groups instead of standalone valves.
But on the top of it, I need some kind of logic to say if there is at least one circuit opened, and if there is, turn on heating, and turn it off once there is no more heat needed.

dncex, 2021-12-23:
I am sorry for the newvie question. I've connected the relay in order to control the general ON/OFF heating. I know that I need to "RT needs to be switched ON in the heatpump Connection menu". In order to do this I need to enter the config with the installer auth level and I can't rember how to do this (my installer told me but I cant remeber). It should be something like press X for 5 secs...

stomko11, 2021-12-23:
On my altherma, it is entering the menu button (bottom right button), then holding top left (i) button for few seconds until menu expands, then got to information - user level access (not sure about precise translation) where you see "advanced user" and then hold again menu for few seconds until you get to technician.
Technician mode gets disabled automatically after some time (or reboot). 

This is how my control panel looks like:

[image]

dncex, 2021-12-23:
many thanks stomko11!

dncex, 2021-12-24:
Now I succeded in controlling the heating ON and OFF with the relay. What is strange is that sometimes (especially during the night, so when the outside temp is low, here around zero) the heat pump is starting by itself even if the heating is off (I have an hybrid hpu, so gas boiler +heat pump). Do you know which settings I should look at?

stomko11, 2021-12-24:
Maybe it is just doing its things - defrost or whatever needs to be done at that time. Could be irrelevant to thermostat being ON or OFF.

KalaNagHTD, 2021-12-24:
It is due to freeze protection. Set the freeze protection temperature to -12°C!

dncex, 2021-12-24:
many thanks KalaNagHTD! The protection make sense for the external portion of the tubes connecting the extenal unit. Anyhow do you know where to check that parameter and eventually how to modify it?

KalaNagHTD, 2021-12-24:
If you have a split unit with refrigerant in the tubes there is no risk of freezing.

KalaNagHTD, 2021-12-24:
I know how to set the freeze protection temperature for the HPSU/ECH2O internal unit only.

closed 2022-11-22 by raomin