raomin/ESPAltherma #69: Power usage during defrost operation

niekveldhuizen, 2021-11-30:
Hi,

I have build my ESPAltherma with an M5StickCPlus, everything is working as it should. But now it is getting colder and the unit starts more defrost cycles during heating.

When the defrost start the primary inv current drops to 0, but in my P1 meter i can see that is peaks from 4Kw up to 9kw. Defrost cycle takes about 5 min.  But it is not counted in my total for the day for the heatpump. 
How can i achive this?

KalaNagHTD, 2021-11-30:
These are the data my unit logs during defrost:

[image]

niekveldhuizen, 2021-11-30:
I use Home Assistant to log everything. I used my P1 data to measure the kwh spent during the 4/5 minutes of defrost and i averages about 0.4kwh for a defrost cycle. But the primary inv (A) does not reflect this so my logging for the usage of the heatpump total kwh for the day is missing the defrost cycle

KalaNagHTD, 2021-11-30:
So something is wrong with your primary current data…

niekveldhuizen, 2021-11-30:
Yeah i dont know. the normal heating proces is registerd just fine. It looks like the defrost is is not using the primary inv. But somtething like the heating element in the DHW tank (that is 9kw) but i do not se a register for that value

raomin, 2022-11-22:
I'll close this one. Open a discussion if still not clear

closed 2022-11-22 by raomin