raomin/ESPAltherma #70: Can I use this with EKHHP300AC2V3?

joseal, 2021-12-06:
Hi I've an EKHHP300AC2V3 heat pump and I'm looking for a way to connect it with my home assistant.

[image]

With this board:

[image]

Could you please confirm if anyone has this code working with this model of pump?

KalaNagHTD, 2021-12-06:
It‘s a DHW heatpump w/o space heating function. As long as it provides the Daikin X10A service connector there is a good chance you can draw lots of useful data. Give it a try!

Karl

joseal, 2021-12-06:
Thanks @KalaNagHTD. I have a few Esp32 and I am going to try. Do you know where are the 5v on this 8pins connector?

robsonke, 2021-12-06:
https://github.com/raomin/ESPAltherma#8-pin-x10a-connection

KalaNagHTD, 2021-12-06:
[image]

joseal, 2021-12-10:
Hi @KalaNagHTD,

I've compiled the code for board esp32doit-devkit-v1 (as I'm using the ESP32 Wroom) and using both default.h and def/ALTHERMA(LT_MULTI_DHWHP).h, but I'm not able to comunitcate with unit.

This is what I see in console:

[image]

The cable is ok (tested the connection to the board) and the ESP32 is sending data on serial 2 (tested with an RS232 adaptor setup for 9600 8E1) and I've already swapped the TxD and RxD (just in case).

So, I'm not sure what I'm missing.... Any ideas?

KalaNagHTD, 2021-12-10:
The Daikin diagnostic software D-Checker (Windows) uses the same connector and data. I would give it a try!
[D-Checker](https://www.thermalsupplyinc.com/daikin-service-checkers/)

andreapier, 2021-12-10:
I had a similar issue (#51) and it turned out to be a grounding problem. Check you ground connection and make sure you are following @raomin advise in README.md:
> Whatever you do, make sure you keep a wire connecting the GND of the ESP32 to the GND pin of the X10A (even if you power your ESP32 with a USB charger)!!

raomin, 2021-12-10:
This one is pretty weird (too). The ESPAltherma is reading what it is sending...
Sending 0x03 0x40 0x61 0x5b is a query for register 0x61, but it reads back exactly the same... As if RX was linked to TX or if there was an echo somewhere...
Please check once more your wiring and remember that your RX on the ESP32 must be linked to TX. Same for TX to RX...

joseal, 2021-12-10:
Hi All,
I've already review and tested the wires multiple times and I'm not finding any issue...
Also, I'm getting this response, even when I swap the TxD with RxD...
To be honest, I'm a bit without ideas, but I'm going to follow @KalaNagHTD 's suggestion and see if D-Checker shows something.

thank very much All for your support so far!

joseal, 2021-12-11:
@raomin, I just noticed this on the mqtt side:

[image]

Do you know what it means this 0x15 0xea?

joseal, 2021-12-11:
In the meantime, I've found a comment on another forum, saying that the 0x15 0xea is the typical answer (from the pump), when the last query was not understood.

If this is the case, I believe we can assume that the wires are good (as we have the right answer from the pump).

Also, having a clear 0x15 0xea answer, it probably means that the com port speed and setup should be good, right?

raomin, 2021-12-11:
0x15 is indeed an error code.
But you might still read properly an error response if your parity is wrong. You can check (as I was also suggesting in this other forum) to change your parity, maybe this unit had a different one...

joseal, 2021-12-11:
> 

Just tried the following combinations, without success:

  MySerial.begin(9600, SERIAL_8E1, RX_PIN, TX_PIN);
  MySerial.begin(9600, SERIAL_8O1, RX_PIN, TX_PIN);
  MySerial.begin(9600, SERIAL_8N1, RX_PIN, TX_PIN);

raomin, 2021-12-20:
Really @joseal, I'm out of idea. Maybe last chance would be to try with DChecker as @KalaNagHTD suggested. If it work there, then there might be some protocol specificities. Then if you have a serial monitor you can see what changes...

dncex, 2021-12-22:
just a possible hint: how long are the cables? If you receive what you transmit, it might be a crosstalk problem. Double check the ground connection too

joseal, 2021-12-22:
Thank you both, I'm going to build a new cable, just to eliminate any chances to be the cable. @dncex could you please elaborate a bit more about that issue?

joseal, 2021-12-22:
Ok, I've tested it with a new cable (with coaxial ground, just to make sure it wasn't noise), but no changes:

[image]

I've also tried the service checker type 4 tool, but it also fails doing the maps....

Do you know about anyone with the same model as me, that was able to communicate with the unit?

thank you very much for your support!

ruislv, 2022-03-08:
Hi @joseal,
i'm facing a similar issue with 500L version. By any change you came to a conclusion or what other findings you got so far?
tkx
r.

joseal, 2022-03-20:
Hi Rui,
unfortunately, I was not able to find why the x10 port is not working as expected in my heat pump.
Maybe this model has a different board/protocol...
I'm trying to find someone that has the original technical interface, in order to test the port.
It was nice to know if someone else with these models were able to make it work...
thank you,
Jose

maps2002, 2022-04-15:
Hi there.

I have the same heat pump model, and the results are the same.
I also have a USB to TTL adapter that I can try to use with DChecker to capture the protocol.

But before I do something wrong, can someone (@raomin ?) advise me/confirm how the connections should be like?
 - TX/RX (X10A) ->  RX/TX (TTL)
 - GND (X10A) -> GND (TTL) (because of the common ground?)
 - No other pin connection is necessary (TTL receives power from the computer)

Thanks in advance

raomin, 2022-04-22:
Yes @maps2002, your connection is correct.
GND-GND is essential.

maps2002, 2022-04-25:
Thanks for the feedback.
Tried with service checker 4, but had the same result as @joseal .

So it seems that until the "right" service check is found, it won't be possible to reverse the protocol and support this model :|

raomin, 2022-11-22:
@maps2002, the protocol on this old one is most probably the S-Protocol that has been included in ESPAltherma last month.
I added some info on README. Give it a try!

closed 2022-11-22 by raomin

joseal, 2022-12-01:
@maps2002 did you had chance to try the S-Protocol on yours?

maps2002, 2022-12-04:
Hi @joseal and @raomin, tried with the S-Protocol. Definitely works!
With PROTOCOL_S_ROTEX definition file (not with PROTOCOL_S), the "DHW tank temp.(C)" and "Refrig. Temp. liquid side(C)" seems to be read properly. All the other values don't seem to contain any valid data that I can relate with anything.

This DHW only heat-pump is from this year. Not-old, but it seems it uses PROTOCOL S none the less.

Maybe more registers can be queried, but at this point not sure what to try/test.

joseal, 2022-12-04:
Hi @maps2002,
thank you very much for the update. Fantastic news! I'll refresh the repo and give it a try later today or tomorrow to see if I can somehow help you find more registers. Let's keep in touch.

andrepcg, 2023-08-23:
I also have a EKHHP300AC2V3 and it works with `PROTOCOL_S_ROTEX`.

```
Circulation pump: OFF
External heater?: OFF
Priority to domestic water: OFF
Burner inhibit from solaris: OFF
Refrig. Temp. liquid side(C): 32.75
Inlet water temp.(C): 212.953
Outlet Water Temp.(C): 212.953
D(C): 212.953
DHW tank temp.(C): 48.5
F(C): 0.00390625
Delta-Tr(deg): 0
R/C Setpoint(C): 0
Operation Mode: Fan Only
Error Code:  0
Thermo Off Error:  0
Warning Code:  0
Caution Code:  0
????: -69.5547
WifiRSSI: -58dBm
FreeMem: 24216
friendly_name: ESPAltherma AlthermaSensors
```

`DHW tank` and `Refrig. Temp. liquid side(C)` are temperatures for what the manual says it has sensors for, although the labels are a bit off:
[image]

`External heater?` turned ON when the booster-heater (BSH) was activated. Would be cool to figure out what other values and labels we have available for this unit