raomin/ESPAltherma #73: Freeze/crash issue?

Gollam, 2021-12-19:
I have had my m5stickcplus running for a month without issues, then suddenly 2 days ago it started going offline, and it has been like that since a day now. I'll take it out and reset it in the afternoon, but do you have any idea what should I check?

[image]

stomko11, 2021-12-19:
Happened to me twice. Like you said, month or so without single issue, then one day 9:00 am sharp, esp (m5stickc) was dead. Pulled 5v input, put it back, All worked. Hour later, same thing. Did that procedure again, and all is working now since then (that was about a week ago)

raomin, 2021-12-19:
Hi! 
That's weird, I don't see what could happen. 
Never had any issue, mine has been running for a year now (!). I've had more general issues with my WiFi or dhcp overlapping with ESPAltherma but that's all.
Best would be to log the ~/Log topic somewhere and see if something happens...

Gollam, 2021-12-21:
Now I took it apart. The screen of the stick said: Restoring previous state: Off Setting up Wifi... Connecting to XXX and nothing else was happening.

raomin, 2021-12-21:
Then probably the WiFi is unreachable or it's dhcp not responding... Check in a different location closer to the access point.

andreapier, 2021-12-21:
This happens to me too from time to time but I have never been able to understand what is going on exactly. I do not think it is a wifi issue as my AP is next to ESP

stomko11, 2021-12-21:
In my case, when that happened, there was nothing on the display. No reaction to any buttons. Pulling 5V and connecting it again immediately resolved it. Unfortunately I do not have any logs from that time to help narrow down the issue.

Gollam, 2022-01-03:
Happy New Year! I discovered that if I leave it running long enough, it will eventually return to normal state. Maybe sometimes the wifi drops it, in that case router restart triggers the restore. And other times maybe HA restart solved it. I'm still trying to determine the exact cause.

raomin, 2022-02-17:
This issue gave me an idea, I've added the wifi signal strength as a sensor value.
This might help identifying wifi issues...

Gollam, 2022-02-18:
I installed a wifi repeater nearby the device, and so far, I had no large outages. Now I only have very small, max 30 minute ones. So I think that helped.

closed 2022-02-18 by raomin