raomin/ESPAltherma #75: Altherma not send the current value for

mpietrusa, 2022-01-12:
Hi All, 
I am owner of Altherma LT  ERLQ014CAW1 + EHVH16S26CB9W + daikin lan adapter BRP069A62 + 
ESPAltherma on board Wemos D1 Mini ESP32.
I have struggle with problem wrong value for some attributes. 
Currently I pull below attributes every 30s:
` "Operation Mode": "Heating",
  "Thermostat ON/OFF": "ON",
  "Defrost Operation": "ON",
  "Liquid pipe temp.(R6T)": 36.4,
  "Outdoor air temp.(R1T)": "-8.7",
  "INV frequency (rps)": 103,
  "I/U operation mode": "DHW",
  "BUH Step1": "OFF",
  "BUH Step2": "OFF",
  "Water pump operation": "ON",
  "Leaving water temp. before BUH (R1T)": 43.2,
  "Leaving water temp. after BUH (R2T)": 42.2,
  "Inlet water temp.(R4T)": 37.6,
  "DHW tank temp. (R5T)": 28.4,
  "Flow sensor (l/min)": 29.6`
Unfortunately from time to time some above attributes are not true it seems like like the value freeze. 
In above case "Defrost Operation" is ON from few hours
[image]
The same situation happen very often for "Outdoor air temp.(R1T)", "Thermostat ON/OFF", rarely for "INV frequency (rps)". Problem occoures even pump is working so it should not be the problem mentioned in issue #28 

If someone have any idea what could I try i will do :)

MadJack-maker, 2022-01-13:
I have the same problem. In may case Outdoor air temp.(R1T) is one of the attributes that freezes. Same with the attribute defrost.

KalaNagHTD, 2022-01-13:
When you have an optional external temp. sensor installed and the ext. unit is in standby the data from the external unit are no longer transmitted to the internal unit. This is a power saving feature.

mpietrusa, 2022-01-13:
I not use optional external temp sensor, so my problem is not related with power saving feature.

raomin, 2022-01-16:
Hmm strange.
Can you have a look at the logs when you witness this behaviour?

Logs are posted on MQTT topic: espalterma/log

You can view them with HA:
Config -> Integration -> MQTT -> Config -> Listen to a topic
This is what you should see:
[image]

mpietrusa, 2022-01-17:
Hi Raomi,
I use mqtt on separate server (not addon in Home Assistant), however I analyzed logs from espaltherma and I have not noticed any difference. Logs looks like the same for working and not working case. 
I will try to check it again when issue occure and I upload log.

mpietrusa, 2022-01-18:
Hi Raomi
Now I seen the case when INV frequency (rps) freeze. 
[image]
and logs not show enything unsual
[image]

raomin, 2022-01-18:
Registers are properly queried, no issue with the checksum.
But that's strange that the attribute is not updated! (4 hours ago).
Can you listen to "espaltherma/#" and see the content of the message with the all the attributes...?

mpietrusa, 2022-01-18:
[image]
`{"Operation Mode":"Heating","Thermostat ON/OFF":"ON","Defrost Operation":"OFF","Outdoor air temp.(R1T)":"-2.6","INV frequency (rps)":31,"I/U operation mode":"Heating","BUH Step1":"OFF","BUH Step2":"OFF","Water pump operation":"ON","Leaving water temp. before BUH (R1T)":29.3,"Leaving water temp. after BUH (R2T)":28.8,"Inlet water temp.(R4T)":22.9,"DHW tank temp. (R5T)":39.3,"Flow sensor (l/min)":30.3}`

raomin, 2022-01-19:
It's weird. ESPAltherma is getting these stuck values from the HP. There is no way that the value is kept from one cycle to the other. 
Could you check on the HP interface to see if you have stuck values too?

mpietrusa, 2022-01-19:
I will check it however I think that in HP panel it did not hangs but we will see.

mpietrusa, 2022-01-23:
During last two days I saw problem with defrost information and Outdoor air temp.(R1T) and both are displayed correctly on HP panel.

mpietrusa, 2022-02-04:
Today, 
I have noticed problem with "Outdoor air temp.(R1T)"
[image]
[image]
{
  "Operation Mode": "Heating",
  "Thermostat ON/OFF": "ON",
  "Defrost Operation": "OFF",
  "Outdoor air temp.(R1T)": 0.8,
  "INV frequency (rps)": 24,
  "I/U operation mode": "Heating",
  "BUH Step1": "OFF",
  "BUH Step2": "OFF",
  "Water pump operation": "ON",
  "Leaving water temp. before BUH (R1T)": 32,
  "Leaving water temp. after BUH (R2T)": 31.7,
  "Inlet water temp.(R4T)": 26.7,
  "DHW tank temp. (R5T)": 42.4,
  "Flow sensor (l/min)": 12.9
}

stomko11, 2022-03-01:
OK, same just happened to me with outdoor temperature. Querying of other values is still OK. Temperature on the Altherma's display was displayed correctly (around -5 in the morning).
I do read values from Altherma every 30 seconds, what I noticed in the logs is that values are probably updated only when there is a change, so I don't get values every 30 seconds, but since around 8 pm yesterday, there is gap until 7 AM today.

[image]

I did unplug altherma from 5V (I power it via Altherma) and let battery discharge for about hour (I found no other way how to reset it?), without any difference. I also tried to upload code to it, again, no change. 
In MQTT info under espaltherma device in HA, I see MQTT info, but temperature reading is just 0.

The only thing that actually helped me was to cut the power to the Altherma, and restore it back. That helped and I am now getting correct readings. I guess that would mean the issue is not in ESP Altherma, but the serial output from heatpump?

stomko11, 2022-03-12:
I don't want to mix up things, but today I found out that I was getting no data at all through my M5stickC. I looked at the display and there was no output. Shutting down my daikin and powering it back on resurrected the stick. Pulling 5V pin would probably do the same. Anyone has same experience? Could this be issue with code, M5 or daikin?

raomin, 2022-03-12:
Really, the log from ESPAltherma says it's getting this from the Altherma. So I would more go for a bug on the Altherma, or maybe a communication bug between the outdoor unit and the inner one, although, the displayed information should be frozen also...
There are firmeware updates available maybe they fix it, @fmode updated his, maybe he can tell a bit more on the process.

stomko11, 2022-03-12:
Thanks. However what I have experienced today is actually M5StickC being completely turned off (nothing on display, no reaction to buttons) while Altherma was operating fine.
Right now, I disconnected 5V from Daikin and hooked it up on USB-C charger, to see if that could fix those occasional strange behavior.

raomin, 2022-03-23:
Hi @stomko11, did you have some success with the external charger?

stomko11, 2022-03-23:
I have experienced 2 types of issues:
1) found out that ESP was dead, nothing on the display, while everything works normally on HeatPump. Of course, no values from ESP in HA, as it was dead. Pulling 5V and reinserting it back resurrects M5.
2) found out that just external temperature was not updating, all other sensors were updated. M5 seems to operate normally. Disconnecting M5 from 5V, letting it to discharge, and powering it back on from 5V does not bring the value back. I restarted indoor unit and things started to work instantly.

These 2 issues were observed by me ~11 days ago, day or so apart from each other.
I can't see how issue number 2 could be related to M5 and looks like issue in the heatpump (although control unit shows correct outdoor temperature). Issue nr.1 could be potentially related to M5 stick, and is also easy to spot as all values I use in HA dashboard just show "unknown"
Since then, I have decided to disconnect it from F5 and am powering it now with USB-C charger. So far, so good. But truth is that I am usually running it OK for months. So ask me in a year, I might be able to tell you :)

mpietrusa, 2022-04-06:
Hi,
I have updated my firmware in Daikin Unit and unfortunately the issue is still exists. Some parmaeters are still freeze in espaltherma. I have also checked that even i restart espaltherma the value which Altherma send to espaltherma are incorret.

raomin, 2022-04-24:
@stomko11, just re-thinking at your temp stuck at 0 when it was -5... are you sure your receiving part can read negative values properly?

@mpietrusa Do you still have this issue? Did you manage to get more info on this?

stomko11, 2022-04-24:
@raomin I read even negative values at other times without any issues.

raomin, 2022-04-24:
@stomko11, ok, just found surprising it would stop at 0. BTW, do you still have frozen values with your external charger?

stomko11, 2022-04-25:
I haven't had issues since then (so far)

mpietrusa, 2022-04-29:
@raomin 
Unfrotunately yes. 
[image]

raomin, 2022-04-29:
Did you try to use an external power supply like @stomko11?

Gollam, 2022-05-04:
Hi,
I just wanted to mention, same issue happening here with outdoor temp.

ERGA08EAV3 + EHVH08S18EJ6V

[image]

amaduain, 2022-05-25:
Same issue here, using an Altherma EHVX16S26CB9W.
The MQQT is reporting the same value for hours, and I can see the logs that the ESP is reading the registries without problem.

[image]

Maybe the bus needs some kind of reset from time to time.

Temperature also shows flat readings (Left is daikin, right is external sensor independent from Daikin unit):
[image]

stomko11, 2022-10-17:
To those that may or may not experience issues with M5StickC like me. It happened to me ocasionally that I would not get any data from heatpump, discovering M5StickC totally off, blank screen, no reaction to any button. Pulling 5V and reinserting it back restored it. This was the case regardless if I used power from HP or dedicated charger. For me, it started to happen more and more often, even exchanging for another M5StickC did not resolve it. 
I have recently switched to other ESP32 board and so far, I have not had any outages. I hope it stays that way.

raomin, 2022-11-22:
Might be a loss of wifi, then the M5 would look dead from the inside trying to connect back. I has been solved since then with automated reboot in case of loss wifi. Other esp32 have probably better wifi sensitivity.
@Gollam @amaduain have you solve your issue?
I also added the WifiRSSI to the reported values that should give you an idea of the Wifi signal quality.

Gollam, 2022-11-23:
Probably in my case it is a temporary weakness of wifi signal that is causing it. The problem is that the keepalive is too strict I think. If only one PINGREQ is left unanswered, the connection is determined broken immediately.

Here is a ticket open for pubsubclient for similar reasons:

https://github.com/knolleary/pubsubclient/issues/795

raomin, 2022-11-24:
I think we're confusing 2 issues here that might not be related:
- One is the loss of connectivity
- The other one is the stalled values
I'll update this issue's title and create a new issue for connectivity loss

amaduain, 2022-11-24:
> 

The issue is still happening randomly, in my case it is not a connectivity issue, bc the ESP32 is close to the AP. Also the data is being pushed to the mqtt topic, so wifi is working seamlessly.

minagerges, 2023-03-21:
I was having disconnect "AlthermaSensors" shows offline randomly, but several times a day.
Resolved by increasing MQTT keepalive.

`client.setKeepAlive(60);`

nicx, 2024-02-06:
@minagerges I ran in the same issue for the outside temperature sensor. Which file do I have to edit? :)

raomin, 2026-10-23:
It's been a while! ESPAltherma 2 rebuilt the whole MQTT side (a streamed discovery, a sized readings message, recommended values instead of everything). Closing with the 2.0 release; if values still stall on v2, a new issue with the Diagnostics page is welcome.

closed 2026-10-23 by raomin