raomin/ESPAltherma #76: Daikin Daikin EGSQH10S18AA9W

newlund, 2022-01-14:
Hi!

This is not an issue. More a question...

Anyone tried ESPAltherma with Daikin EGSQH10S18AA9W?
Which #include line should be used?

I have not purchased the ESP32 yet. Just wanted to know before hitting the buy button :)

Thanks!
Nicklas

raomin, 2022-01-16:
Hi @newlund,

As explained in [the wiki page](https://github.com/raomin/ESPAltherma/wiki/Choosing-the-right-definition-file), the definition files are mostly identical. Only few names will differ regarding specific feature of an heat pump (DWH or not, hybrid or not, etc.).
I would start with `ALTHERMA(LT_CA_CB_11-16KW).h` you should have most of the data correct.

Let us know!

newlund, 2022-01-17:
Thanks! I will try.
The ESP32 is on the way... :)

onix62, 2022-02-01:
I have also a question.
Anyone tried ESPAltherma with Daikin Altherma 3 GEO EGSAX06DA9W?
Which #include line should be used?

Thank you.

mihu-ov, 2023-11-07:
Daikin Altherma 3 GEO EGSAX06DA9W works fine with ALTHERMA(GSHP2).h and inverter current definition copied from ALTHERMA(GSHP).h