raomin/ESPAltherma #77: Connector for X10A

andreapier, 2022-01-19:
Hello, can someone tell the female connector that should be used in X10A port?
I know I can use normal dupont wires, but I am experiencing some random read errors and would like to try do the wiring from scratch (maybe also make things cleaner).
Is it a 5P JST XH 2.5mm like [this kit](https://www.amazon.it/ARCELI-connettore-Femmina-Connettore-terminale/dp/B07RP6876C/ref=sr_1_9?__mk_it_IT=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=75N1HEZC9WXY&keywords=jst+2.54mm&qid=1642615036&sprefix=jst+2.54mm%2Caps%2C121&sr=8-9)?

KalaNagHTD, 2022-01-19:
[image]

closed 2022-01-20 by andreapier

andreapier, 2022-01-21:
Thanks, @KalaNagHTD!
Looks like those connectors are really rare... I can only find them on aliexpress.
Do you know if other JST family connectors are compatible?

KalaNagHTD, 2022-01-21:
Sorry I have no idea…

RodEzIO, 2022-01-22:
nice question, the mating connector should be a JST EHR-5 
https://www.digikey.com/en/products/detail/jst-sales-america-inc/EHR-5/527227
as usual nowadays any electronic component is out of stock and impossible to buy.

andreapier, 2022-01-22:
Yeah, thanks. At the end I decided to build my own cable so I purchased [some connectors](https://www.aliexpress.com/item/1005003388263149.html?spm=a2g0o.order_list.0.0.21ef1802b9YGK7)
I hope they fit. I'll let you know

andreapier, 2022-02-12:
@KalaNagHTD I can confirm the connector is JST EH with 2.5mm pitch as you suggested.
@raomin Would you like me to make a pull request to improve the current README.md? It currently states that the connector is a JST PH with 2.0 mm pitch.