raomin/ESPAltherma #78: Anyone got a success with an ETVH16S18D6V?

Clust3r, 2022-01-20:
Hi, 

First of all "Merci et bravo" for this program

I have just finalized the installation of an ETVH16S18D6V, I connected the ESP to the X10A but unfortunately, even when trying several profiles, I still have no feedback.

Regards

```
espaltherma/log Querying register 0x64...
espaltherma/log Time out! Check connection

espaltherma/ATTR {"*Refrigerant type":"Conv 802 not avail.","Sensor Data Qty":0,"INV compressor Qty":0,"STD compressor Qty":0,"Fan Data Qty":0,"Expansion Valve Data Qty":0,"4 Way Valve Data Qty":0,"Crank Case Heater Qty":0,"Solenoid valve Qty":0,"Max. connectable indoor units":0,"Connected Indoor Unit Qty":0,"O/U MPU ID (xx)":0,"O/U MPU ID (yy)":0,"O/U capacity (kW)":0,"NextDataGrid":"Conv 995 not avail.","NextDataGrid":"Conv 995 not avail.","In-Out separator":"Conv 998 not avail.","M5VIN":"4.97V","M5AmpIn":"35mA","M5BatV":"4.12V","M5BatCur":"0mA","M5Bat Pwr":"0mW"}
```

raomin, 2022-01-21:
Hi @Clust3r 
_Time out! Check connection_
You probably have an issue with your connection for now. Check your GND connection especially: If you are powering your ESP with an external 5VCC charger, please make sure you also bridge the GND from the HP with the GND from the ESP...

Also, uncomment more (meaningful) values.

Clust3r, 2022-01-22:
Yeah... After five cable replacement, data are retrieved correctly !  Sorry for the mistake :)