raomin/ESPAltherma #81: Smart Grid Pins locations

vekexasia, 2022-02-15:
Hello, 

I just discovered this project and read almost all issues, In the main readme there is a mention on J8 connector to leverage smartgrid functionalities.

Since i'm a rookie i couldn't find any reference to the location of such connector on my machine. I took a picture of it:

[image]

PS: my model number is: Daikin Altherma 3 R ECH20 - EHSXB08P50DA which does not have a perfect match on the include files (according the models listed in the wiki pages). But i believe is ALTHERMA(HPSU6_ULTRA).h. 

Another question. Assuming I can find the j8 i would also need the 2 relays to use. any suggestion on good ones? PS: I'd be using the m5stack atom lite esp32 dev board https://shop.m5stack.com/products/atom-lite-esp32-development-kit

KalaNagHTD, 2022-02-15:
[image]

KalaNagHTD, 2022-02-15:
[image]

vekexasia, 2022-02-15:
<3

andreapier, 2022-02-15:
I recently purchased [this 2 channels relay module](https://www.aliexpress.com/item/1005001725711302.html?spm=a2g0o.order_list.0.0.21ef1802jb7TBb).
It works fine, but I will wait a bit more time before commenting on how good (or bad) it is.
About the board you want to use, just make sure it has a GND GPIO or you might experience weird issues reading values from your heat pump.

raomin, 2022-02-17:
Just keep in mind that the voltage of ESP32 GPIOs is 3.3v, usually this is enough to power the opto couplers on these module, but I would recommend to get a 3.3v model.
I'm closing this issue as the answers have been provided by @KalaNagHTD and @andreapier 
but feel free to comment on it with your progress.

closed 2022-02-17 by raomin

ocsi01, 2025-01-26:
@vekexasia Did you managed to run this on the `m5stack atom lite`? I just got the ´[AtomS3 Lite](https://shop.m5stack.com/products/atoms3-lite-esp32s3-dev-kit)´, but facing some issues deploy it.